ghdl/ghdl #1563: libghdl generating unessesary midff

LAK132, 2020-12-30:
**Description**
libghdl is generating unnecessary (for synthesis) midffs when code is split across multiple `if rising_edge(clock) then` blocks.

**Expected behaviour**
After https://github.com/LAK132/mega65-core/commit/a79329934bb30df26d521c3a3a8ca9920223fa96 libghdl only generates a single midff that doesn't seem to make it to ghdl-yosys-plugin and is hence not synthesized. I would expect this behaviour regardless of the code layout when attempting to use ghdl for synthesis.

**How to reproduce?**

Generates 2 Midffs (1 winds up in ghdl-yosys-plugin which can't be synthesized):
```vhdl
if rising_edge(clock) then
  thing <= something;
end if;
if rising_edge(clock) then
  thing <= somethingelse;
end if;
```

Generates 1 Midff:
```vhdl
if rising_edge(clock) then
  thing <= something;
  thing <= somethingelse;
end if;
```

https://github.com/LAK132/mega65-core/commit/a79329934bb30df26d521c3a3a8ca9920223fa96 runs ghdl with: `ghdl --std=93c -v -frelaxed -frelaxed-rules --mb-comments $$($1_VHDL) -e gs4510`

**Context**

- OS: Windows 10 (Ubuntu 20.04 WSL)
- Origin:
  - [ ] Built from sources: `fbe43ad4dc8451fecba5318af97541283207a0e8`

tgingold, 2020-12-31:
Do you have a standalone reproducer ?

LAK132, 2020-12-31:
Produces midff in yosys: 
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity bug is
port (
    clock : in std_logic;
    output : out std_logic
);
end bug;

architecture bug_arch OF bug is
begin
    process (clock)
    begin
        if rising_edge(clock) then
            output <= '1';
        end if;
        if rising_edge(clock) then
            output <= '0';
        end if;
    end process;
end bug_arch;
```

Doesn't produce midff in yosys: 
```vhdl
library ieee;
use ieee.std_logic_1164.all;

entity bug is
port (
    clock : in std_logic;
    output : out std_logic
);
end bug;

architecture bug_arch OF bug is
begin
    process (clock)
    begin
        if rising_edge(clock) then
            output <= '1';
            output <= '0';
        end if;
    end process;
end bug_arch;
```

I'm also currently putting together a full toolchain repro if that helps.

LAK132, 2020-12-31:
Running `make ghdl-error` and/or `make ghdl-ok` off [this commit](https://github.com/LAK132/mega65-core-toolchain/tree/72d949ee77f178f50adcd256e115fa67501f1d91) should automatically build the toolchain I'm using and reproduce the issue with the above code snippets

commit e84f35ed (2021-01-01): netlists-inference: handle multiple dff with the same clock.  Fix #1563
closed 2021-01-01 by tgingold