ghdl/ghdl #1570: libghdl.link contains hardcoded paths from the build environment

umarcor, 2020-12-31:
Yesterday, Yosys was added to MSYS2's MINGW-packages repository. As seen in the PKGBUILD recipe (https://github.com/msys2/MINGW-packages/blob/master/mingw-w64-yosys/PKGBUILD), it depends on GHDL and ghdl-yosys-plugin is built in.

While doing so, the Yosys makefile patch was upstreamed from ghdl-yosys-plugin to Yosys (see YosysHQ/yosys#2514). A fix needs to be merged yet (YosysHQ/yosys#2515), but that's independent from this issue.

GHDL, Yosys and nextpnr (with GUI), all of them are successfully built in MSYS2's CI (GitHub Actions). Any user can download the temporary packages from https://github.com/msys2/msys2-autobuild/releases/tag/staging-mingw and use any or all of them.

However, when a user tried to build Yosys locally using the PKGBUILD recipe from MINGW-packages, an annoying problem arised. The paths in `libghdl.link` are hardcoded and contain the location of `libgnat.a` at built time. That was unnoticed on MSYS2's CI system because the packages are built in the same location. Different jobs and different installations are used, but the environment is setup in the same location. As a result, building Yosys alone locallly fails because the location of `libgnat.a` does not match the location in CI (precisely, in CI MSYS2 is _not_ located in `C:\msys64`).

It seems that the problem is not easy to fix:

https://github.com/ghdl/ghdl/blob/7ca54117b8f757396ba5ef04c83ff1228ca94384/Makefile.in#L423-L424

`grep` and `sed` are used for filtering and modifying a previously generated file (`libghdl.bind`). It's challenging to make that generic enough for different platforms.

I think I can handle it in PKGBUILD recipes as shown in the following patch: https://github.com/umarcor/MINGW-packages/commit/b1eac65b612340cba1f9141078511f3079ccbf62. That is, I remove `C:/....` from the absolute path and it is left as `/mingw...`. That is still absolute, but it's an MSYS2's Unix style absolute path. That approach is tested in https://github.com/umarcor/MINGW-packages/actions/runs/454943832, where Yosys is forced to rebuild.

I am unsure about proposing a fix in this repo, because other users build GHDL and Yosys using MINGW32/MINGW64 resources, but they don't use PKGBUILD recipes or bash shells. On the one hand, it might be difficult to cover those cases. On the other hand, those might need to define the path to `libgnat.a` manually anyway.

Closing remarks:

- This issue does not exist if users build both GHDL and Yosys (with built-in ghdl-yosys-plugin) on the same environment.
- I don't know whether packagers on GNU/Linux distros are affected by this.

/cc @edbordin @juanmard

umarcor, 2021-01-28:
FTR, this is also a problem when using `--list-link` on MSYS2: https://github.com/ghdl/ghdl-cosim/blob/1beee2f7d3c8dd30786112291310ef61c287c1d1/vhpidirect/quickstart/linking/bind/run.sh#L37-L38

kimstik, 2023-03-09:
UCRT64 fresh revision v3.0 have hardcoded path
"D:/a/_temp/msys64/ucrt64/lib/gcc/x86_64-w64-mingw32/12.2.0/adalib/libgnat.a"

umarcor, 2023-03-09:
Maybe applying https://github.com/msys2/MINGW-packages/pull/16035 to the recipes in this repo will fix it.