ghdl/ghdl #1572: Crash with error `unhandled call to ieee function "to_integer"`

MJoergen, 2021-01-01:
**Description**
During synthesis (and using PSL) ghdl crashes with the error `unhandled call to ieee function "to_integer"`.

The work-around is to use the function `conv_integer` instead and use the command line option `-fsynopsys`.

**Expected behaviour**
No crash. The error is (probably?) correct, but ghdl should not crash.

**How to reproduce?**
```vhd :file: ent.vhd
-- ent.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;
   
entity ent is
   port (
      clk_i  : in  std_logic;
      done_o : out std_logic
   ); 
end entity ent;
   
architecture synthesis of ent is
   signal u0 : std_logic_vector(2 downto 0) := "101";
begin
   done_o <= '0';
end architecture synthesis;
```

```vhd :file: ent.psl
-- ent.psl
vunit i_ent(ent(synthesis))
{
   default clock is rising_edge(clk_i);

   process (all)
      variable num_colors : integer_vector(0 to 7);
   begin
      num_colors := (others => 0);
      num_colors(to_integer(u0)) := num_colors(to_integer(u0)) + 1;
   end process;
}
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl --synth --std=08 ent.vhd ent.psl -e ent
```

**Context**
Please, provide the following information:

- OS: Linux Mint 20
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `commit #83dfd494` (Date:   Tue Dec 8 06:26:20 2020 +0100)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1105-g83dfd494) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/mike/tmp/
Command line:
ghdl --synth --std=08 ent.vhd ent.psl -e ent
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : synth-values.adb:252 access check failed
Call stack traceback locations:
0x564e857e6204 0x564e858284db 0x564e85802e4e 0x564e858044aa 0x564e858061bd 0x564e858111a7 0x564e858119cd 0x564e85813160 0x564e85813bb1 0x564e85801c19 0x564e85801fcb 0x564e858020ea 0x564e857fb32f 0x564e85865f38 0x564e8578a890 0x564e85914570 0x564e85544a51 0x7f79f829e0b1 0x564e855431cc 0xfffffffffffffffe
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2021-01-01:
I think the workaround is to use numeric_std instead of numeric_std_unsigned.

MJoergen, 2021-01-01:
> I think the workaround is to use numeric_std instead of numeric_std_unsigned.

If I use `numeric_std` then I get the error `no overloaded function found matching 'to_integer'` (but no crash).

tgingold, 2021-01-01:
Yes as you have to convert the parameter to `unsigned`.

MJoergen, 2021-01-01:
Yes, you are right. Replacing `conv_integer(u0)` with `to_integer(unsigned(u0))` works (together with `numeric_std`).

tgingold, 2021-01-01:
I am adding the support for it.

commit cbeafcc2 (2021-01-01): synth-oper: handle ieee.numeric_std_unsigned.to_integer.  Fix #1572
closed 2021-01-01 by tgingold