ghdl/ghdl #1574: Assertion failure when using a case inside a function

LAK132, 2021-01-02:
**Description**
When a case statement in a function has more than one return statement it causes ghdl to crash

**Expected behaviour**
ghdl shouldn't crash

**How to reproduce?**

With more than one return statement it crashes:
```vhd :file: ghdl-error.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity bug is
    port (
        clock : in std_logic;
        output : out std_logic_vector(7 downto 0)
    );
end bug;

architecture bug_arch OF bug is
    type my_enum is (
        thing1,
        thing2,
        thing3
    );

    signal input : my_enum := thing1;

    impure function my_enum_func return std_logic_vector is
    begin
        case input is
            when thing1 => return x"00";
            when others => return x"A0";
        end case;
    end my_enum_func;
begin
    process (clock)
    begin
        if rising_edge(clock) then
            output <= my_enum_func;
        end if;
    end process;
end bug_arch;
```
`ghdl ghdl-error.vhdl -e bug`

With only one return statement it works fine:
```vhd :file: ghdl-ok.vhdl
library ieee;
use ieee.std_logic_1164.all;

entity bug is
    port (
        clock : in std_logic;
        output : out std_logic_vector(7 downto 0)
    );
end bug;

architecture bug_arch OF bug is
    type my_enum is (
        thing1,
        thing2,
        thing3
    );

    signal input : my_enum := thing1;

    impure function my_enum_func return std_logic_vector is
    begin
        case input is
            when others => return x"A0";
        end case;
    end my_enum_func;
begin
    process (clock)
    begin
        if rising_edge(clock) then
            output <= my_enum_func;
        end if;
    end process;
end bug_arch;
```
`ghdl ghdl-ok.vhdl -e bug`

**Context**
Please, provide the following information:

- OS: Windows 10 (WSL Ubuntu 20.04)
- Origin:
  - [ ] Built from sources: `b423d311`

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev () [Dunoon edition]
Compiled with unknown compiler version
Target: x86_64-linux-gnu
/mnt/d/Repos/mega65-core-toolchain/
Command line:

Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : synth-stmts.adb:2055
******************************************************************
```

**Additional context**
[Full toolchain repro branch](https://github.com/LAK132/mega65-core-toolchain/tree/ghdl-bug-multi-return-assert) (`make ghdl-error` and `make ghdl-ok`)

tgingold, 2021-01-02:
It is not the number of return statements but the fact the function is impure.  Impure functions are not well handled.

LAK132, 2021-01-02:
Still crashes if it's marked as `pure` instead of `impure`

tgingold, 2021-01-02:
Just make it really pure: do not reference `input` but pass it as a parameter.

LAK132, 2021-01-02:
Ah ok, that works. Thanks.

dalex78, 2026-08-19:
# Issue #1574 — Crash: function with multiple `return` statements in a `case`

Analysis by an AI agent. Recommend (optionally merging the test and) closing the ticket.

## High-level summary

A function whose `case` statement has more than one `return` statement (one per branch, rather than a single `return` at the end) crashed `ghdl --synth` with an internal `ASSERT_FAILURE` at `synth-stmts.adb:2055`. A version with only one `return` (in the `others` branch) worked fine.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl --synth ghdl-error.vhdl -e bug` now produces a correct netlist (the two-branch `case`/`return` function inlines into a `with ... select` concurrent signal assignment) instead of crashing. This must have been fixed as part of general improvements to multi-return function inlining in `src/synth` since this was reported (originally against a 2020 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1574/` (the exact repro from the report, renamed `bug.vhdl` for the test directory), whose `testsuite.sh` synthesizes the design and re-analyzes the resulting netlist. This locks in the current, correct behavior as a regression guard.

closed 2026-08-20 by tgingold

dalex78, 2026-08-20:
# Issue #1574 — Crash synthesizing a function that reads a signal

Address the second part of https://github.com/ghdl/ghdl/issues/1574 that was missed previously.

I made a mistake on some of the tickets for which I opened PRs. When letting the AI agent work on it, I didn't realize that it was not looking at the conversations several people has in the thread. Therefore, I asked him to go back and re-analyze the tickets.

Analysis by an AI agent.

## High-level summary

A function called from a clocked process crashed `ghdl --synth` with an internal `ASSERT_FAILURE`. The report attributed it to the `case` statement having more than one `return` statement, but that is not the cause. The maintainer identified it on the thread:

> It is not the number of return statements but the fact the function is impure. Impure functions are not well handled.

The function reads the signal `input` declared in the enclosing architecture, which is what makes it impure.

## Current status

The design as reported — the function declared `impure` — synthesizes on current `master`, so that half is fixed.

The follow-up on the thread is not. Asked whether the purity declaration mattered, the reporter answered:

> Still crashes if it's marked as `pure` instead of `impure`

That is still true today, on all three backends:

```
raised ADA.ASSERTIONS.ASSERTION_ERROR : synth-vhdl_stmts.adb:3049
```

The workaround the maintainer suggested — *"Just make it really pure: do not reference `input` but pass it as a parameter"* — sidesteps it, but writing `pure` on a function that reads a signal should not crash the compiler.

## What is actually wrong in GHDL's implementation

A function that reads a signal from an enclosing declarative part breaks the pure rule. `Sem_Check_Pure` (`src/vhdl/vhdl-sem_expr.adb`) detects it and reports it, but for a function it returns without recording anything:

```ada
--  Function.
if Get_Kind (Subprg) = Iir_Kind_Function_Declaration then
   Error_Pure (Subprg, Obj);
   return;
end if;
```

The procedure case a few lines below does record it, with `Set_Purity_State (Subprg, Impure)`. The diagnostic is relaxed — a warning by default (`-Wpure`) — so analysis continues with `Pure_Flag` still set on a function that is demonstrably not pure.

Synthesis then trusts that flag. `Synth_Subprogram_Call` (`src/synth/synth-vhdl_stmts.adb`) decides how to synthesize a call from it:

```ada
--  If the subprogram is not pure, clear the const flag.
if Ctxt /= null and then not Get_Pure_Flag (Imp) then
   Set_Instance_Const (Sub_Inst, False);
end if;
...
if Get_Instance_Const (Sub_Inst) then
   Res := Synth_Static_Subprogram_Call (...);
else
   Res := Synth_Dynamic_Subprogram_Call (...);
end if;
```

With the flag still set, the call takes the static path, which requires the result to be computable at elaboration:

```ada
pragma Assert (Is_Static (C.Ret_Value.Val));
```

The function reads a signal, so its result is not static and the assertion fails. Spelling the same function `impure` clears the flag, takes the dynamic path, and works — which is exactly why only the `pure` spelling crashes.

## What the fix does

Records the violation, mirroring what the procedure case already does: `Set_Pure_Flag (Subprg, False)` next to the existing diagnostic. The function is not pure whatever it was declared, the rule is only relaxed so the rest of the compiler carries on with this subprogram, and nothing downstream should assume its result can be computed statically.

Both spellings then behave identically, which is the point of the thread.

## Testing

`testsuite/synth/issue1574/bug.vhdl` is the report's design. `bug_pure.vhdl` is the follow-up from the thread — the same function declared `pure` while still reading the signal. Both are synthesized and the resulting netlist re-analyzed; the second crashes on master without this fix.

Full `sanity gna vests synth` pass on mcode, llvm and gcc.