ghdl/ghdl #1579: Incorrect handling of conditional instructions during the elaboration

wzab, 2021-01-04:
**Description**
During the work at the VHDL code generator, I have discovered the fact that the current version of GHDL, during the elaboration, executes the instructions in the process that are located inside the "if" with false condition.

**Expected behaviour**
The instructions in "if" with false condition should be skipped both during the elaboration and during the simulation.

**How to reproduce?**
The `test1.vhd` resembles the original, generated code that allowed me to discover the problem. You may test it for various values of the `nobjs` constant. For `nobjs=4` it works correctly. For lower values of `nobjs` it fails during the elaboration with:
```
./test1:error: bound check failure at test1.vhd:35
in process .test1(beh1).p1
./test1:error: error during elaboration
```
During the analysis phase the warning is displayed:
```
test1.vhd:35:15:warning: static expression violates bounds [-Wruntime-error]
test1.vhd:35:28:warning: static expression violates bounds [-Wruntime-error]
```
but the failing line should never be executed (due to if).

```vhd :file test1.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test1 is
  
end test1;

architecture beh1 of test1 is

  constant nobjs : integer := 3;
  
  signal clk : std_logic := '0';

  type my_array is array (integer range <>) of unsigned(7 downto 0);
  signal a_test : my_array(0 to nobjs-1) := (others => (others => '0'));
  
begin  -- beh1

  clk <= not clk after 10 ns;
  
  p1: process (clk) is
  begin  -- process p1
    if clk'event and clk = '1' then     -- rising clock edge
      if 0 < nobjs then
	a_test(0) <= a_test(0) + 1;
      end if;
      if 1 < nobjs then
	a_test(1) <= a_test(1) + 1;
      end if;
      if 2 < nobjs then
	a_test(2) <= a_test(2) + 1;
      end if;
      if 3 < nobjs then
	a_test(3) <= a_test(3) + 1;
      end if;
    end if;
  end process p1;
  
end beh1;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a test1.vhd
ghdl -e test1
./test1 --stop-time=300ns
```

**Context**
Please, provide the following information:

- OS: Linux Debian/testing
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `340fc792bba2ffdb4f930bc427a39ea3a1b659b2`

tgingold, 2021-01-04:
no, GHDL is correct here.  The method to compute drivers during elaboration is defined by the LRM and the conditions of `if` statements are not considered.

You either have to write more generic code or to use generate statements.

wzab, 2021-01-04:
Well, it looks like my MWE was a little too simplified (it fails even with old versions of GHDL). In the original code the "ifs" were controlled by the generic. I provide the modified MWE below. 

The first set of files uses generic to set the size of the object and if in the process:

```vhd :file test1.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test1 is
  generic (
    nobjs : integer := 4
    );
  port (
    clk : in std_logic
    );

end test1;

architecture beh1 of test1 is

  type my_array is array (integer range <>) of unsigned(7 downto 0);
  signal a_test : my_array(0 to nobjs-1) := (others => (others => '0'));

begin  -- beh1

  p1 : process (clk) is
  begin  -- process p1
    if clk'event and clk = '1' then     -- rising clock edge
      if 0 < nobjs then
        a_test(0) <= a_test(0) + 1;
      end if;
      if 1 < nobjs then
        a_test(1) <= a_test(1) + 1;
      end if;
      if 2 < nobjs then
        a_test(2) <= a_test(2) + 1;
      end if;
      if 3 < nobjs then
        a_test(3) <= a_test(3) + 1;
      end if;
    end if;
  end process p1;

end beh1;
```

```vhd :file test1_tb.vhd
library ieee;
use ieee.std_logic_1164.all;

-------------------------------------------------------------------------------

entity test1_tb is

end entity test1_tb;

-------------------------------------------------------------------------------

architecture test of test1_tb is

  -- component generics
  constant nobjs : integer := 3;

  -- component ports
  signal clk : std_logic := '0';

  -- clock

begin  -- architecture test

  -- component instantiation
  DUT: entity work.test1
    generic map (
      nobjs => 4)
    port map (
      clk => clk);

  -- clock generation
  clk <= not clk after 10 ns;

  -- waveform generation
  WaveGen_Proc: process
  begin
    -- insert signal assignments here

    wait until Clk = '1';
  end process WaveGen_Proc;

  

end architecture test;

-------------------------------------------------------------------------------

configuration test1_tb_test_cfg of test1_tb is
  for test
  end for;
end test1_tb_test_cfg;
```

```sh: image ghdl/ghdl:buster-mcode
ghdl -a test1.vhd test1_tb.vhd
ghdl -e test1_tb
./test1_tb --stop-time=300ns
```
That code simulates perfectly.

The second code also uses generic to pass the size of the vector.
However, it additionally uses `case` to check certain additional condition (like my original code generated by AGWB did).

```vhd :file test2.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test2 is
  generic (
    nobjs : integer := 4
    );
  port (
    clk : in std_logic;
    x	:    integer range 0 to 2
    );

end test2;

architecture beh1 of test2 is

  type my_array is array (integer range <>) of unsigned(7 downto 0);
  signal a_test : my_array(0 to nobjs-1) := (others => (others => '0'));

begin  -- beh1

  p1 : process (clk) is
  begin	 -- process p1
    if clk'event and clk = '1' then	-- rising clock edge
      case x is
	when 0 =>
	  if 0 < nobjs then
	    a_test(0) <= a_test(0) + 1;
	  end if;
	  if 1 < nobjs then
	    a_test(1) <= a_test(1) + 1;
	  end if;
	when 1 =>
	  if 2 < nobjs then
	    a_test(2) <= a_test(2) + 1;
	  end if;
	  if 3 < nobjs then
	    a_test(3) <= a_test(3) + 1;
	  end if;
	when others =>
	  null;
      end case;
    end if;

  end process p1;

end beh1;
```

```vhd :file test2_tb.vhd
library ieee;
use ieee.std_logic_1164.all;

-------------------------------------------------------------------------------

entity test2_tb is

end entity test2_tb;

-------------------------------------------------------------------------------

architecture test of test2_tb is

  -- component generics
  constant nobjs : integer := 3;

  -- component ports
  signal clk : std_logic	    := '0';
  signal x   : integer range 0 to 2 := 0;
  -- clock

begin  -- architecture test

  -- component instantiation
  DUT : entity work.test2
    generic map (
      nobjs => 3)
    port map (
      clk => clk,
      x	  => x);

  -- clock generation
  clk <= not clk after 10 ns;

  p1 : process (clk) is
  begin	 -- process p1
    if clk'event and clk = '1' then	-- rising clock edge
      if x < 2 then
	x <= x+1;
      else
	x <= 0;
      end if;
    end if;
  end process p1;

  -- waveform generation
  WaveGen_Proc : process
  begin
    -- insert signal assignments here

    wait until Clk = '1';
  end process WaveGen_Proc;

end architecture test;

-------------------------------------------------------------------------------

configuration test2_tb_test_cfg of test2_tb is
  for test
  end for;
end test2_tb_test_cfg;
```

```sh: image ghdl/ghdl:buster-mcode
ghdl -a test2.vhd test2_tb.vhd
ghdl -e test2_tb
./test2_tb --stop-time=300ns
```

That code fails with:
```
./test2_tb:error: index (3) out of bounds (0 to 2) at test2.vhd:39
in process .test2_tb(test).dut@test2(beh1).p1
./test2_tb:error: error during elaboration
```

So it looks that the problem occurs only if I put a generic-controlled if into the `case-when` statement.

wzab, 2021-01-04:
I have tried to replace the `case-when` with `if`:

```vhd :file test3.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test3 is
  generic (
    nobjs : integer := 4
    );
  port (
    clk : in std_logic;
    x	:    integer range 0 to 2
    );

end test3;

architecture beh1 of test3 is

  type my_array is array (integer range <>) of unsigned(7 downto 0);
  signal a_test : my_array(0 to nobjs-1) := (others => (others => '0'));

begin  -- beh1

  p1 : process (clk) is
  begin	 -- process p1
    if clk'event and clk = '1' then	-- rising clock edge
      if x = 0 then
	if 0 < nobjs then
	  a_test(0) <= a_test(0) + 1;
	end if;
	if 1 < nobjs then
	  a_test(1) <= a_test(1) + 1;
	end if;
      end if;
      if x = 2 then
	if 2 < nobjs then
	  a_test(2) <= a_test(2) + 1;
	end if;
	if 3 < nobjs then
	  a_test(3) <= a_test(3) + 1;
	end if;
      end if;
    end if;

  end process p1;

end beh1;
```

```vhd :file test3_tb.vhd
library ieee;
use ieee.std_logic_1164.all;

-------------------------------------------------------------------------------

entity test3_tb is

end entity test3_tb;

-------------------------------------------------------------------------------

architecture test of test3_tb is

  -- component generics
  constant nobjs : integer := 3;

  -- component ports
  signal clk : std_logic	    := '0';
  signal x   : integer range 0 to 2 := 0;
  -- clock

begin  -- architecture test

  -- component instantiation
  DUT : entity work.test3
    generic map (
      nobjs => 3)
    port map (
      clk => clk,
      x	  => x);

  -- clock generation
  clk <= not clk after 10 ns;

  p1 : process (clk) is
  begin	 -- process p1
    if clk'event and clk = '1' then	-- rising clock edge
      if x < 2 then
	x <= x+1;
      else
	x <= 0;
      end if;
    end if;
  end process p1;

  -- waveform generation
  WaveGen_Proc : process
  begin
    -- insert signal assignments here

    wait until Clk = '1';
  end process WaveGen_Proc;

end architecture test;

-------------------------------------------------------------------------------

configuration test3_tb_test_cfg of test3_tb is
  for test
  end for;
end test3_tb_test_cfg;
```

```sh: image ghdl/ghdl:buster-mcode
ghdl -a test3.vhd test3_tb.vhd
ghdl -e test3_tb
./test1_tb --stop-time=300ns
```

However the result is still the same:
```
./test3_tb:error: index (3) out of bounds (0 to 2) at test3.vhd:39
in process .test3_tb(test).dut@test3(beh1).p1
./test3_tb:error: error during elaboration
```

wzab, 2021-01-04:
I thought that maybe a problem is related to the  nesting of ifs, so I have combined conditions:
```vhd :file test4.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test4 is
  generic (
    nobjs : integer := 4
    );
  port (
    clk : in std_logic;
    x	:    integer range 0 to 2
    );

end test4;

architecture beh1 of test4 is

  type my_array is array (integer range <>) of unsigned(7 downto 0);
  signal a_test : my_array(0 to nobjs-1) := (others => (others => '0'));

begin  -- beh1

  p1 : process (clk) is
  begin	 -- process p1
    if clk'event and clk = '1' then	-- rising clock edge
      if (x = 0) and (0 < nobjs) then
	a_test(0) <= a_test(0) + 1;
      end if;
      if (x = 0) and (1 < nobjs) then
	a_test(1) <= a_test(1) + 1;
      end if;
      if (x = 2) and (2 < nobjs) then
	a_test(2) <= a_test(2) + 1;
      end if;
      if (x = 2) and (3 < nobjs) then
	a_test(3) <= a_test(3) + 1;
      end if;
    end if;

  end process p1;

end beh1;
```

```vhd :file test4_tb.vhd

library ieee;
use ieee.std_logic_1164.all;

-------------------------------------------------------------------------------

entity test4_tb is

end entity test4_tb;

-------------------------------------------------------------------------------

architecture test of test4_tb is

  -- component generics
  constant nobjs : integer := 3;

  -- component ports
  signal clk : std_logic	    := '0';
  signal x   : integer range 0 to 2 := 0;
  -- clock

begin  -- architecture test

  -- component instantiation
  DUT : entity work.test4
    generic map (
      nobjs => 3)
    port map (
      clk => clk,
      x	  => x);

  -- clock generation
  clk <= not clk after 10 ns;

  p1 : process (clk) is
  begin	 -- process p1
    if clk'event and clk = '1' then	-- rising clock edge
      if x < 2 then
	x <= x+1;
      else
	x <= 0;
      end if;
    end if;
  end process p1;

  -- waveform generation
  WaveGen_Proc : process
  begin
    -- insert signal assignments here

    wait until Clk = '1';
  end process WaveGen_Proc;

end architecture test;

-------------------------------------------------------------------------------

configuration test4_tb_test_cfg of test4_tb is
  for test
  end for;
end test4_tb_test_cfg;
```

```sh: image ghdl/ghdl:buster-mcode
ghdl -a test4.vhd test4_tb.vhd
ghdl -e test4_tb
./test1_tb --stop-time=300ns
```

The output is still the same:
```
./test4_tb:error: index (3) out of bounds (0 to 2) at test4.vhd:36
in process .test4_tb(test).dut@test4(beh1).p1
./test4_tb:error: error during elaboration
```

wzab, 2021-01-04:
I have changed the order of nested ifs (see the test5.vhd in the attached archive). Still fails.
I'm really surprised that test1.vhd works, while all other versions do not work.

The ultimate solution appeared to be usage of the `for-loop` instead of `if` as in the `test6.vhd` code shown below:
```vhd :file test6.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity test6 is
  generic (
    nobjs : integer := 4
    );
  port (
    clk : in std_logic;
    x	:    integer range 0 to 2
    );

end test6;

architecture beh1 of test6 is

  type my_array is array (integer range <>) of unsigned(7 downto 0);
  signal a_test : my_array(0 to nobjs-1) := (others => (others => '0'));

begin  -- beh1

  p1 : process (clk) is
  begin	 -- process p1
    if clk'event and clk = '1' then	-- rising clock edge
      for i in 0 to nobjs-1 loop
	if x mod 3 = i then
	  a_test(i) <= a_test(i) + 1;
	end if;
      end loop;
    end if;
  end process p1;

end beh1;
```

```vhd :file test6_tb.vhd
library ieee;
use ieee.std_logic_1164.all;

-------------------------------------------------------------------------------

entity test6_tb is

end entity test6_tb;

-------------------------------------------------------------------------------

architecture test of test6_tb is

  -- component generics
  constant nobjs : integer := 3;

  -- component ports
  signal clk : std_logic	    := '0';
  signal x   : integer range 0 to 2 := 0;
  -- clock

begin  -- architecture test

  -- component instantiation
  DUT : entity work.test6
    generic map (
      nobjs => 3)
    port map (
      clk => clk,
      x	  => x);

  -- clock generation
  clk <= not clk after 10 ns;

  p1 : process (clk) is
  begin	 -- process p1
    if clk'event and clk = '1' then	-- rising clock edge
      if x < 2 then
	x <= x+1;
      else
	x <= 0;
      end if;
    end if;
  end process p1;

  -- waveform generation
  WaveGen_Proc : process
  begin
    -- insert signal assignments here

    wait until Clk = '1';
  end process WaveGen_Proc;

end architecture test;

-------------------------------------------------------------------------------

configuration test6_tb_test_cfg of test6_tb is
  for test
  end for;
end test6_tb_test_cfg;
```

```sh: image ghdl/ghdl:buster-mcode
ghdl -a test6.vhd test6_tb.vhd
ghdl -e test6_tb
./test1_tb --stop-time=300ns
```

Now the simulation is successful:
```
./test6_tb:info: simulation stopped by --stop-time @300ns
```

Unfortunately, that requires significant rework of my AGWB VHDL code generator...
I attach the archive
[testfiles.zip](https://github.com/ghdl/ghdl/files/5764901/testfiles.zip)
with test files used in that issue.

closed 2021-01-04 by wzab

tgingold, 2021-01-04:
test1.vhd works because nobjs is set to 4.

Changing the if/case statements has no effect.  The issue is due to the signal assignments.

wzab, 2021-01-04:
Yes, I have noticed that. That's why I have cleaned-up the issues from all my further experiments...
Finally I have implemented the code generator in a cleaner and safer way. It is described in https://github.com/wzab/agwb/issues/46 .
Thanks for support!