ghdl/ghdl #1581: Sed error on configure on Ubuntu 20.04

svenssonjoel, 2021-01-04:
**Description**

$ ./configure --with-llvm-config --prefix=/usr/local
Use full IEEE library
Build machine is: x86_64-linux-gnu
Debugging is enabled with llvm 10.0.0
Creating ghdl.gpr
sed: -e uttryck #23, tecken 15: missing filename in r/R/w/W commands
Creating Makefile
sed: -e uttryck #23, tecken 15: missing filename in r/R/w/W commands
./configure: cannot execute config.status

Sorry about the mixed swedish output there. It says "expression #23, character 15" 

Sed version: GNU sed 4.7

I pulled GHDL from github today (Jan 4) master branch.  Is this supposed to work or should I have checked out one of the versioned branches? 

Thanks and best regards

tgingold, 2021-01-04:
There is something very weird as the r/R/w/W commands are not used in the sed scripts.

Can you spot the executed command ?  (maybe use `sh -x ./configure ...`)

svenssonjoel, 2021-01-05:
Thanks. 

You are right, something must have been very weird! I cannot reproduce this issue this today. 

Now I get this, which is something that is much more understandable. 
--- 
./configure --with-llvm-config --prefix=/usr/local
Use full IEEE library
Build machine is: x86_64-linux-gnu
Unhandled version llvm 10.0.0
--- 

Thank you and have a nice day

svenssonjoel, 2021-01-05:
Thank you. 

These are the last lines of output when using 'sh -x' 

```
+ echo done
+ chmod +x ./config.status
+ sh ./config.status
Creating ghdl.gpr
sed: -e uttryck #23, tecken 15: missing filename in r/R/w/W commands
Creating Makefile
sed: -e uttryck #23, tecken 15: missing filename in r/R/w/W commands
+ echo ./configure: cannot execute config.status
./configure: cannot execute config.status
+ exit 1
```
If I checkout ghdl-0.35 and run the same command, this problem does not appear. There I get the much more understandable message that my version of LLVM is not supported. 

Thank you and have a nice day!

eine, 2021-01-05:
I think this is the point where it's failing:

https://github.com/ghdl/ghdl/blob/1fb6cb173ecda149deb1f3afd7f04d52f7468f93/configure#L344-L347

Using ghdl-0.35, the error is not shown because the equivalent part is not executed:

https://github.com/ghdl/ghdl/blob/eb5499b1859680f2b94f89c5eabf544946af4ba3/configure#L283-L286

It fails before:

https://github.com/ghdl/ghdl/blob/eb5499b1859680f2b94f89c5eabf544946af4ba3/configure#L224

tgingold, 2021-01-05:
What is the content of `config.status` ?  This is the file that executes the `sed` command.
You can also try yo execute it manually like `sh -x config.status`.

vgtw, 2021-03-15:
I also had this error, it seems to be some problem with the Swedish locale. Switching `LANG` from `sv_SE.UTF-8` to `en_GB.UTF-8` before ./configure worked as a workaround (running Debian bullseye/testing, compiling v1.0.0/2fb2384d).

tgingold, 2021-03-18:
I was not able to reproduce the issue.
Can you file your config.status with and without setting LANG ?

vgtw, 2021-03-18:
From what I can see, the Swedish version has these flags
` -e "s%@enable_@%$enable_%g"w -e "s%@error@%$error%g" `
instead of this flag present in the English one
`-e "s%@enable_werror@%$enable_werror%g"`

This is my config.status with sv_SE

```sh
#! /bin/sh
# Generated by configure.
# Run this file to recreate the current configuration.

if [ x"$1" = x"--reconfigure" ]; then
   ./configure
  exit
fi

subst_vars="CC CXX GNATMAKE MAKE CFLAGS LDFLAGS build srcdir abs_srcdir prefix backend libdirsuffix libdirreverse gcc_src_dir llvm_config llvm_be backtrace_lib build_mode EXEEXT SOEXT PIC_FLAGS default_pic enable_werror enable_checks enable_gplcompat enable_libghdl libghdl_version ghdl_version"
CC="/usr/bin/clang"
CXX="/usr/bin/clang++"
GNATMAKE="gnatmake"
MAKE="make"
CFLAGS="-g"
LDFLAGS=""
build="x86_64-pc-linux-gnu"
srcdir="."
abs_srcdir="/home/vgtw/build/ghdl/."
prefix="/usr/local"
backend="mcode"
libdirsuffix="lib/ghdl"
libdirreverse="../.."
gcc_src_dir=""
llvm_config=""
llvm_be="llvm6"
backtrace_lib=""
build_mode=""
EXEEXT=""
SOEXT=".so"
PIC_FLAGS="-fPIC"
default_pic="true"
enable_werror="true"
enable_checks="true"
enable_gplcompat="false"
enable_libghdl="true"
libghdl_version="1_0_0"
ghdl_version="1.0.0"
for f in ghdl.gpr Makefile; do
  echo "Creating $f"
  sed  -e "s%@CC@%$CC%g"  -e "s%@CXX@%$CXX%g"  -e "s%@GNATMAKE@%$GNATMAKE%g"  -e "s%@MAKE@%$MAKE%g"  -e "s%@CFLAGS@%$CFLAGS%g"  -e "s%@LDFLAGS@%$LDFLAGS%g"  -e "s%@build@%$build%g"  -e "s%@srcdir@%$srcdir%g"  -e "s%@abs_srcdir@%$abs_srcdir%g"  -e "s%@prefix@%$prefix%g"  -e "s%@backend@%$backend%g"  -e "s%@libdirsuffix@%$libdirsuffix%g"  -e "s%@libdirreverse@%$libdirreverse%g"  -e "s%@gcc_src_dir@%$gcc_src_dir%g"  -e "s%@llvm_config@%$llvm_config%g"  -e "s%@llvm_be@%$llvm_be%g"  -e "s%@backtrace_lib@%$backtrace_lib%g"  -e "s%@build_mode@%$build_mode%g"  -e "s%@EXEEXT@%$EXEEXT%g"  -e "s%@SOEXT@%$SOEXT%g"  -e "s%@PIC_FLAGS@%$PIC_FLAGS%g"  -e "s%@default_pic@%$default_pic%g"  -e "s%@enable_@%$enable_%g"w -e "s%@error@%$error%g"  -e "s%@enable_checks@%$enable_checks%g"  -e "s%@enable_gplcompat@%$enable_gplcompat%g"  -e "s%@enable_libghdl@%$enable_libghdl%g"  -e "s%@libghdl_version@%$libghdl_version%g"  -e "s%@ghdl_version@%$ghdl_version%g" < $srcdir/${f}.in > $f
done
```

tgingold, 2021-03-18:
Ah thank you very much.  Apparently this is an issue with recent glibc, where `w` does not belong to `[a-z]`
See https://sourceware.org/bugzilla/show_bug.cgi?id=23447

I will commit a workaround.

commit c4256d93 (2021-03-18): configure: work-around unexpected regexp range with sv_SE locale.  Fix #1581
closed 2021-03-18 by tgingold