ghdl/ghdl #1584: How to restructure documentation when growing further

Paebbels, 2021-01-05:
GHDL's documentation has evolved a lot since the last years. It started with compile instructions and a tiny example and how we have a the content of a book. With #1580 and similar PRs we added Python APIs and there documentation. We also can now autogenerate CLI option lists and documentation for Python scripts using argparse. (Sphinx extension AutoProgram).

I would like to discuss here a general structure für the Navigation bar in GHDLs documentation.

## Overall Concept

Currently we see 4 levels of a documentation that GHDL is providing:
1. **New** and **Interested Users**:
   * what is GHDL
   * first attempt to use GHDL
   * quick start guides
   * project overview
   * where to ask questions
   * how to install from package/zip
2. **Normal Users**:
   * using GHDL für simulation
   * using synthesis
   * mainly the CLI interface of the ghdl binary
   * how to start language server and run in IDE
3. **Developers** who develop on top of GHDL own tools:
   * GHDLs APIs
     * Python oder
     * libghdl interfaces
4. **GHDL Developers** helping GHDL itself
   * Internals for development in GHDL
   * Auto generated docs
     * gnatdoc
     * low-level python API
5. APPENDIX

**Goals:**
* Reflect different levels of engagement with GHDL: test out, use extensively, develop with, contribute
* Reduce number of level 0 and level 1 navigation bar items
* Chose chapter names (level 1) that can grow with GHDLs future
* Reflect changes in the GHDL project since documentation was started
* Loose no content, just rearrange

## Proposed vs. Current

| New Navigation Bar                         | Current Navigation Bar           | 
| ------------------------------------------ | -------------------------------- |
| **Introduction**                           | **Introduction**                 |
| &emsp;&emsp; About GHDL                    | &emsp;&emsp; About GHDL          | 
| &emsp;&emsp; Quickstart Guides             |                                                | 
| &emsp;&emsp;&emsp;&emsp;  Simulation       |                                                |
| &emsp;&emsp;&emsp;&emsp;  Synthesis        |                                                |
| &emsp;&emsp;&emsp;&emsp;  Lang. Server     |                                                |
| &emsp;&emsp;&emsp;&emsp;  Python API       |                                                |
| &emsp;&emsp;&emsp;&emsp;  Co-Simul.        |                                                |
| &emsp;&emsp; Installation                  | &emsp;&emsp; Contributing                      | 
| &emsp;&emsp;&emsp;&emsp;  From Package     | &emsp;&emsp; Copyrights \| License             |
| &emsp;&emsp;&emsp;&emsp;  From Container   | **Getting GHDL**                               |
| &emsp;&emsp;&emsp;&emsp;  From Archiv      | &emsp;&emsp; Releases and sources              |
| &emsp;&emsp;&emsp;&emsp;  From Source      | &emsp;&emsp; Building GHDL                     |
| &emsp;&emsp; Getting Help                  |                                                |
| &emsp;&emsp;&emsp;&emsp;  Issue            |                                                |
| &emsp;&emsp;&emsp;&emsp;  Gitter           |                                                |
| &emsp;&emsp;&emsp;&emsp;  StackoverFlow    |                                                |
| &emsp;&emsp; Copyrights \| License         | &emsp;&emsp; Precompile Vendor Primitives      |
| **GHDL Usage**                             | **GHDL USAGE**                                 |
| &emsp;&emsp; Invoking GHDL                 | &emsp;&emsp; Quick Start Guide                 |
| &emsp;&emsp; Simulation (runtime)          | &emsp;&emsp; Invoking GHDL                     |
| &emsp;&emsp; Synthesis                     | &emsp;&emsp; Simulation (runtime)              |
| &emsp;&emsp; Co-simulation                 | &emsp;&emsp; Synthesis                         |
| &emsp;&emsp; Precompile Vendor Primitives  | &emsp;&emsp; Additional Command Reference      |
| &emsp;&emsp; Misc. Topics                  | &emsp;&emsp; Co-simulation                     |
| **For Developers**                         | &emsp;&emsp; Implementation of VHDL            |
| &emsp;&emsp; GHDL Command Reference        | &emsp;&emsp; Implementation of VITAL           |
| &emsp;&emsp; Building GHDL from Sources    |                                                |
| &emsp;&emsp; libghdl                       |                                                |
| &emsp;&emsp; Python API (hand written)     |                                                |
| **For Contributors**                       | **DEVELOPMENT**                                |
| &emsp;&emsp; Overview                      | &emsp;&emsp; Debugging                         |
| &emsp;&emsp; Data structures               | &emsp;&emsp; Coding Style                      |
| &emsp;&emsp;&emsp;&emsp;  Front-end        | &emsp;&emsp; gnatdoc                           |
| &emsp;&emsp;&emsp;&emsp;  AST              | &emsp;&emsp; Python Interfaces                 |
| &emsp;&emsp;&emsp;&emsp;  RTI              | &emsp;&emsp; Scripts and Applications          |
| &emsp;&emsp; GNAT-Doc                      | **INTERNALS**                                  |
| &emsp;&emsp; Python APIs                   | &emsp;&emsp; Overview                          |
| &emsp;&emsp; Scripts                       | &emsp;&emsp; Front-end                         |
| &emsp;&emsp; &emsp;&emsp; Makefile ?       | &emsp;&emsp; AST                               |
| &emsp;&emsp; &emsp;&emsp; pnodes           | &emsp;&emsp; Run Time Information (RTI)        |
| &emsp;&emsp; Continuous Integration        | &emsp;&emsp;                                   |
| &emsp;&emsp; &emsp;&emsp; Testing GHDL     | &emsp;&emsp;                                   |
| &emsp;&emsp; &emsp;&emsp; Packaging GHDL   |                                                |
| &emsp;&emsp; Debugging                     |                                                |
| &emsp;&emsp; Contributing                  |                                                |
| &emsp;&emsp;&emsp;&emsp;  Pull Request     |                                                |
| &emsp;&emsp;&emsp;&emsp;  Coding Style     |                                                |
| **Appendix**                               | **INDEX**                                      |
| &emsp;&emsp; Glossary                      | &emsp;&emsp;                                   |
| &emsp;&emsp; Index                         | &emsp;&emsp; Index                             |
| &emsp;&emsp; Python Module Index           | &emsp;&emsp; Python Module Index               |

## Changes Explained
### Quickstart Guides
GHDL has now more main feature like synthesis and co-simulation, so each main feature should be represented with a Qickstart Guide, so users who want to start with GHDL can easily follow a simple guide and can get in touch with the tool.

Topics:
* Simulation - might have 2..4 examples
* Synthesis - might have 1..3 examples
* Language Server - how to start and use it
* Python API - A simple to use Python API e.g. DOM
* Co-Simulation - how to use it
* Yosys - is this quickstart or more advanced?

### Installation
As we started the documentation writing some years ago, it was mostly about how to compile GHDL on each machine and platform, because this was the only way to get up-to-date installations. Now we have pre-compiled archives, binary packages and some platforms (e.g. MinGW) have GHDL integrated in their package management system.

So I consider compilation task more like an advanced topics for GHDL power users, but not for beginners. That's why an installation chapter should focus on pre-packages/pre-built files.

umarcor, 2021-01-06:
FTR, there is WIP in https://github.com/umarcor/ghdl/tree/rework-doc (see https://umarcor.github.io/ghdl/), but it's based on #1583. I'll open a PR once #1583 is merged.

umarcor, 2021-01-06:
Maybe this section of the README can be moved as the index of the Quick Start Guide: https://github.com/umarcor/ghdl/tree/rework-doc#project-structure.

I would also remove https://github.com/umarcor/ghdl/tree/rework-doc#building-ghdl from the README, and place it in the docs only.