ghdl/ghdl #1588: Scope of component instantiation does not find the entity.

sckoarn, 2021-01-08:
**Description**
The definition in the code, of the component, is stated in the test architecture.  This hides the definition of the entity, in work, but it should be visible based on LRM 2000 5.2.2 b)

**Expected behaviour**
The component instantiation is successful.

**How to reproduce?**
See the code below :
```library ieee;
use ieee.std_logic_1164.all;
use work.all;

entity libs12 is
end entity;

architecture tb of libs12 is

  component tent
    port(
      rst_n : in std_logic;
      clk   : in std_logic;
      period : in integer;
      clk_o : out std_logic;
      op1 : out integer;
      op2 : out integer
      );
  end component;

  signal tclk : std_logic := '0';
  signal trst : std_logic := '0';
  signal tper : integer := 1;
  signal oclk : std_logic;
  signal oop1 : integer;
  signal oop2 : integer;
  signal oclk1 : std_logic;
  signal oop11 : integer;
  signal oop21 : integer;
  
begin

  process
    variable v_cnt : integer := 0;
  begin
    trst  <= '0';
    wait for 1 ns;
    trst  <= '1';
    wait for 1 ns;
    while v_cnt < 50 loop
      --report "clk";
      tclk <= not tclk;
      wait for 1 ns;
      v_cnt := v_cnt + 1;
    end loop;
  end process;

u1: entity tent(rtl)
  port map(
    rst_n  => trst,
    clk    => tclk,
    period => tper,
    clk_o  => oclk,
    op1    => oop1,
    op2    => oop2
    );

u2: entity tent(bhv)
  port map(
    rst_n  => trst,
    clk    => tclk,
    period => tper,
    clk_o  => oclk1,
    op1    => oop11,
    op2    => oop21
    );

  mon: process(oclk)
  begin
    if oclk'event and oclk = '1' then
      report "Rising edge oclk with oop1: " & integer'image(oop1) &
      " and oop2: " & integer'image(oop2);
      report "Rising edge oclk with oop11: " & integer'image(oop11) &
      " and oop12: " & integer'image(oop21);
    end if;
  end process mon;

end tb;
```

```library ieee;
use ieee.std_logic_1164.all;

entity tent is
  port(
    rst_n : in std_logic;
    clk   : in std_logic;
    period : in integer;
    clk_o : out std_logic;
    op1 : out integer;
    op2 : out integer
    );
    
begin

  clk_mon: process(clk)
    variable tl : time := 0 ns;
    variable td : time;
    variable lp : time;
  begin
    if(clk'event) then
      --report "" & time'image(now);
      if clk = '1' then
        td := now - tl;
        tl := now;
        if lp /= td then
          report "clk period is: " & time'image(td);
          lp := td;
        end if;
      end if;
      
    end if;
  end process;
  
end entity tent;
```

```architecture rtl of tent is

  signal bv : bit_vector(15 downto 0) := (others => '1');
  signal stdv : std_logic_vector(15 downto 0);
  signal clk1  : std_logic := '1';
  
  signal cnt : integer := 0;
  signal cnt1 : integer := 0;
  
  signal operiod : time;
  
begin

  --operiod <= period * 1 ns;

  clock : process(clk)
    variable clk_v : std_logic := '0';
  begin
    if clk'event and clk = '1' then
      clk_v := not clk_v;
      clk_o <= clk_v;
    end if;
    if cnt > 20 then
      report "END  SIM ..." severity failure;
    end if;
  end process clock;

  process(clk)
  begin
    if clk'event and clk = '1' then
      cnt <= cnt + 1;
      --report "clk event ...";
      op1 <= cnt;
      clk1 <= not clk1;
    end if;
  end process;

  process(clk1)
  begin
    if(clk1'event and clk1 = '1') then
      cnt1 <= 2 ** cnt;
      --report "clk1 event ...";
      op2 <= cnt1;
    end if;
  end process;

end rtl;

architecture bhv of tent is

  signal bv : bit_vector(15 downto 0) := (others => '1');
  signal stdv : std_logic_vector(15 downto 0);
  signal clk1  : std_logic := '1';
  
  signal cnt : integer := 0;
  signal cnt1 : integer := 0;
  
  signal operiod : time;
  
begin

  process
  begin
    report "BHV ...";
    wait;
  end process;

  clock : process(clk)
    variable clk_v : std_logic := '0';
  begin
    if clk'event and clk = '1' then
      clk_v := not clk_v;
      clk_o <= clk_v;
    end if;
    if cnt > 20 then
      report "END  SIM ..." severity failure;
    end if;
  end process clock;

  process(clk)
  begin
    if clk'event and clk = '1' then
      cnt <= cnt + 1;
      --report "clk event ...";
      op1 <= cnt;
      clk1 <= not clk1;
    end if;
  end process;

  process(clk1)
  begin
    if(clk1'event and clk1 = '1') then
      cnt1 <= 2 ** cnt;
      --report "clk1 event ...";
      op2 <= cnt1;
    end if;
  end process;

end bhv;
```

```rm -fr work-obj93.cf
ghdl -a --std=93 dut.vhdl archs.vhdl
ghdl -a --std=93 test.vhdl
ghdl -r --std=93 libs12
```

**Context**
Please, provide the following information:

- OS:  Linux, and  EDAPlayground
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

GHDL 0.37 (v0.37) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 mcode code generator
Written by Tristan Gingold.

If a `GHDL Bug occurred` block is shown in the log, please paste it here:
test.vhdl:62:12:error: entity name expected, found component "tent"
test.vhdl:72:12:error: entity name expected, found component "tent"

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2021-01-08:
LRM 5.2.2 is about default binding indication, not about the visibility of names.

sckoarn, 2021-01-08:
But the default binding depends on visibility.  The way we see it is that the component defined in the top arch, "hides" the definition in work of the entity with the same name.  But 5.2.2 says that we should be able to see "through" the component to find the entity of the same name in work.

I have tested this on EDA PG, with the other guys, and they agree with what I expected,  the entity is found.

tgingold, 2021-01-08:
> But the default binding depends on visibility. The way we see it is that the component defined in the top arch, "hides" the definition in work of the entity with the same name. But 5.2.2 says that we should be able to see "through" the component to find the entity of the same name in work.

As you are using direct instantiation, there is no component involved, so no configuration.

The direction instantiation expect an entity name.  But `tent` denotes a component.  Hence the error.

> I have tested this on EDA PG, with the other guys, and they agree with what I expected, the entity is found.

For me it doesn't make sense.

eine, 2021-01-08:
Maybe `u1: entity work.tent(rtl)`? Fromm LRM 2008 11.7.3.

tgingold, 2021-01-08:
> Maybe u1: entity work.tent(rtl)? From LRM 2008 11.7.3.

Yes, that would work.

eine, 2021-01-08:
I thought that specifying the library was required when instantiating an entity directly, but not for components. However, I could not find that defined in the LRM.

closed 2021-01-14 by tgingold