ghdl/ghdl #1589: Support for constants of unconstrained types lacks inferring sizes from default assignment

suoto, 2021-01-08:
**Description**
Creating a constant of a type that has unconstrained array of an unconstrained array causes stack dump.

**Expected behaviour**
Boundaries can be inferred from the default assignment.

**How to reproduce?**
```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

entity ent is
end ent;

architecture ent of ent is

  type integer_array_t is array (natural range <>) of integer;
  type integer_2d_array_t is array (natural range <>) of integer_array_t;
  type std_logic_array_t is array (natural range <>) of std_logic_vector;

  constant test0 : integer_array_t := (0, 1, 2, 3);                                        -- OK

  -- array of arrays fail if type declaration has no size whatsoever
  constant test1 : integer_2d_array_t := ((0, 1, 2, 3), (0, 1, 2, 3));                     -- Fails
  constant test2 : std_logic_array_t := ( x"10", x"11");                                   -- Fails

  -- Constraining the std_logic_vector but not the array length also fails
  subtype byte is std_logic_vector(7 downto 0);
  constant test3 : std_logic_array_t := ( byte'(x"10"), byte'(x"10"));                     -- Fails

  -- Constraining the std_logic_vector via subtype AND the array length also fails
  constant test4 : std_logic_array_t(0 to 1) := ( byte'(x"10"), byte'(x"10"));             -- Fails

  -- If everything in constrained at the type declaration it works:
  constant test5 : std_logic_array_t(0 to 1)(7 downto 0) := ( byte'(x"10"), byte'(x"10")); -- Works
  constant test6 : std_logic_array_t(0 to 1)(byte'range) := ( byte'(x"10"), byte'(x"10")); -- Works
  constant test7 : std_logic_array_t(0 to 1)(byte'range) := ( x"10", x"10");               -- Works

begin
end ent;

```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 ent.vhd
ghdl --elab-run --std=08 ent
```

**Context**

- OS: Linux (Elementary OS 5.1.7)
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `b01cf47e`

**Additional context**
This is part of an open source project, let me know if you need more code

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (0.37.0.r1280.gb01cf47e) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/home/souto/dev/ghdl_issue/
Command line:
/home/souto/dev/ghdl/build/ghdl1-llvm --std=08 -P/home/souto/dev/ghdl/build/lib/ghdl/ieee/v08/ -P/home/souto/dev/ghdl/build/lib/ghdl/ -c -fpic -o ent
.o ent.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1674
Call stack traceback locations:
0x4a7a00 0x62aa58 0x62b6ca 0x62dde3 0x62c380 0x62c65e 0x5ea74c 0x5eab6d 0x5eaae6 0x5ecfd8 0x5ed094 0x5cedbf 0x5cf19f 0x5cf29e 0x5d7ec9 0x5c4424 0x63c
47a 0x58c85d 0x642a36 0x644396 0x645a00 0x40ab0b 0x7f73748c5bf5 0x409df8 0xfffffffffffffffe
******************************************************************
```

tgingold, 2021-01-09:
Yes, I think this is one of the main remaining vhdl08 issue about unbounded arrays.

umarcor, 2021-01-09:
I think this issue is another variant of #1055 and #1455.

commit f9e7eba9 (2021-08-04): testsuite/gna: add test and close #1589
closed 2021-08-04 by tgingold