ghdl/ghdl #1591: Synth/psl: Signals with init only result in PSL error when declared in block/generate declaration region.

tmeissner, 2021-01-11:
**Description**
Maybe a design has signals which are  declared inside a `block` or `generate` declaration region. If these signals are initialized only and not further assigned, they result in an error in the ghdl-yosys-plugin when using them in PSL code.

I created the Issue here in GHDL project because I assume the problem is located in GHDL synthesis and not the ghdl-yosys-plugin.

**Expected behaviour**
Signals with initialization only, but without any further assignments should work with PSL code. Even if they are declared inside a `block` or `generate` declaration region.

**How to reproduce?**
Get this repo: https://github.com/tmeissner/psl_with_ghdl
```
git clone https://github.com/tmeissner/psl_with_ghdl
```

Run the test case:
```sh :image: ghdl/synth:formal
cd issues
make issue1590
```
results in error message

```
$ make issue_1590
mkdir -p work
sed 's/__ISSUE__/issue_1590/g' template.sby > work/issue_1590.sby
ghdl --synth --std=08 -fpsl issue_1590.vhd -e issue > work/issue_1590_synth.vhd
issue_1590.vhd:14:10:warning: signal "a" is never assigned
sby --yosys "yosys -m ghdl" -f -d work/issue_1590 work/issue_1590.sby bmc
SBY 10:39:20 [work/issue_1590] Removing directory 'work/issue_1590'.
SBY 10:39:20 [work/issue_1590] Copy 'issue_1590.vhd' to 'work/issue_1590/src/issue_1590.vhd'.
SBY 10:39:20 [work/issue_1590] engine_0: smtbmc z3
SBY 10:39:20 [work/issue_1590] base: starting process "cd work/issue_1590/src; yosys -m ghdl -ql ../model/design.log ../model/design.ys"
SBY 10:39:20 [work/issue_1590] base: issue_1590.vhd:14:10:warning: signal "a" is never assigned
SBY 10:39:20 [work/issue_1590] base: ERROR: Assert `n.id != 0' failed in src/ghdl.cc:204.
SBY 10:39:20 [work/issue_1590] base: finished (returncode=1)
SBY 10:39:20 [work/issue_1590] base: job failed. ERROR.
SBY 10:39:20 [work/issue_1590] summary: Elapsed clock time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 10:39:20 [work/issue_1590] summary: Elapsed process time [H:MM:SS (secs)]: 0:00:00 (0)
SBY 10:39:20 [work/issue_1590] DONE (ERROR, rc=16)
SBY 10:39:20 One or more tasks produced a non-zero return code.
make: [Makefile:11: issue_1590] Error 16 (ignored)
```

**Context**

- OS: Debian Buster
- Origin:
  - [x] Docker (Image: ghdl/synth:formal)

tgingold, 2021-01-12:
In fact, this is not PSL related.  I can reproduce the same issue with:

```
entity repro4 is
  port (o : out boolean);
end entity repro4;

architecture arch of repro4 is
begin
  testG : if true generate
    signal b : boolean := true;
  begin
    o <= b;
  end generate testG;
end;
```

tmeissner, 2021-01-12:
Ah, okay. It was found by @meck during tests with PSL. I changed the title of the issue.

commit 2fb32502 (2021-01-13): synth: always finalize declarations.  Fix #1591
closed 2021-01-13 by tgingold