ghdl/ghdl #1593: 'element is not yet implemented

Paebbels, 2021-01-12:
**Is your feature request related to a problem? Please describe.**
When testing pre-compile scripts for UVVM, it was discovered, that array attribute `'element` is not implemented.
I assume it's a not yet implemented VHDL-08 feature.

**Describe the solution you'd like**
Please implement this feature to support UVVM (and OSVVM libraries with latest code).

**Describe alternatives you've considered**
*None*

**Additional context**
Here is a reproducer:
```vhdl
entity e1 is
	port (
		vector : in  bit_vector(3 downto 0);
		output : out bit
	);
end entity;

architecture a1 of e1 is
	signal zero : vector'element;
begin
	zero   <= vector(0);
	
	process(all) is
		variable z : vector'element;
	begin
		z := zero;
		
		output <= z;
	end process;
end architecture;

entity e2 is
end entity;

architecture top of e2 is
begin
	l1: entity work.e1
		port map (
			vector => (others => '0'),
			output => open
		);
end architecture;
```

Shown messages:
```
PS C:\GHDL> C:\Tools\GHDL\1.0.dev0-llvm\bin\ghdl.exe -a --std=08 C:\Temp\ghdl\TickElement.vhdl
C:\Temp\ghdl\TickElement.vhdl:9:29: port "vector" was not annotated with attribute "element"
C:\Temp\ghdl\TickElement.vhdl:9:29: a type mark must be a simple or expanded name
C:\Temp\ghdl\TickElement.vhdl:14:36: port "vector" was not annotated with attribute "element"
C:\Temp\ghdl\TickElement.vhdl:14:36: a type mark must be a simple or expanded name
C:\Temp\ghdl\TickElement.vhdl:18:26: can't match 'z' with type bit
C:\Temp\ghdl\TickElement.vhdl:26:21: entity 'e2' was not analysed
C:\Tools\GHDL\1.0.dev0-llvm\bin\ghdl.exe: compilation error
```

cartfjord, 2021-02-21:
Any hints on where to start implementing this feature?

tgingold, 2021-02-21:
You can follow the implementation of 'subtype.

Although parsing and semantic analysis should be easy, the code generation (translate) might be more difficult.

closed 2021-08-01 by tgingold