ghdl/ghdl #1596: Synth support for string literal in port map to unconstrained std_logic_vector

bl0x, 2021-01-13:
**Description**
When an entity 'v' with a unconstrained std_logic_vector port 'input' is instantiated by mapping a string literal to the port, ghdl reports during synthesis that a bug occurred. See below for the `GHDL Bug occurred` block.

**Expected behaviour**
It is expected, that the string literal is accepted at this point. A workaround is also shown in the MWE below. This explicitly adds a signal which is initialised to the string literal and then used in the port map.

**How to reproduce?**
Minimum working example. 'ent_bug' triggers the bug, while 'ent_working' is showing the workaround. Both are fine during elaboration, but the bug shows up in synthesis.

```vhd :file: ent.vhd
library ieee;
use ieee.std_logic_1164.all;

-- Instatiated component ---
entity v is
	-- input is unconstrained
	port (input : in std_logic_vector);
end;

architecture RTL of v is
	constant bits : positive := 4;
	signal i : std_logic_vector(bits - 1 downto 0);
begin
	i <= input;
end;

library ieee;
use ieee.std_logic_1164.all;
use work.v;

-- TOP WORKING ---
entity ent_working is end;
architecture RTL of ent_working is
	signal a : std_logic_vector(3 downto 0) := "0101";
begin
	inst_v : entity v
	port map (input => a);
end;

library ieee;
use ieee.std_logic_1164.all;
use work.v;

-- TOP BUG ---
entity ent_bug is end;
architecture RTL of ent_bug is
begin
	inst_v : entity v
	port map (input => "0000");
end;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl --elab-run ent_working # OK
ghdl --elab-run ent_bug # OK
ghdl --synth ent_working # OK
ghdl --synth ent_bug # FAILS!
```

**Context**
Please, provide the following information:

- OS: Manjaro Linux
- Origin: Built from source, GHDL release: 1.0-dev (0.37.0.r1278.g485a5741) [Dunoon edition]

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (0.37.0.r1278.g485a5741) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-pc-linux-gnu
/home/bloeher/git/rataser/tmp/
Command line:
ghdl --synth ent_bug
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:32
Call stack traceback locations:
0x48f5b3 0x560240 0x5681e4 0x56ab55 0x553d12 0x554328 0x56bf47 0x567377 0x596492 0x4facb7 0x4076a9 0x7fba8556e150 0x40771c 0xfffffffffffffffe
******************************************************************
```

eine, 2021-01-13:
@bl0x, did you try `port map (input => B"0000");`?

tgingold, 2021-01-13:
Yes, I can reproduce the issue.

bl0x, 2021-01-13:
@eine also with your suggestion, the same bug is encountered.

commit 5d367877 (2021-01-16): synth-expr: handle string literals in port associations.  Fix #1596
closed 2021-01-16 by tgingold