ghdl/ghdl #1597: GHDL does not enforce the  requirement for static signals for attribute 'active

sckoarn, 2021-01-13:
**Description**
While creating some tests, it has been found that GHDL does not enforce the static-ness of the prefix to 'active.
LRM 2000 section 14
"S'ACTIVE
Kind: Function.
Prefix: Any signal denoted by the static signal name S.
Result Type: Type Boolean.
Result: A value that indicates whether signal S is active. Specifically:"

**Expected behaviour**
GHDL should indicate an error at the point where the  'active is the suffix of a non-static signal.

**How to reproduce?**
The two code examples should show the issue

```  function active_bit(signal vec : std_logic_vector;
                      idx : integer) return boolean is
  begin
    return vec(idx)'active;
  end function;
```
```  function active_bit(signal vec : std_logic_vector;
                      idx : integer) return boolean is
  begin
    return vec(idx)'active;
  end function;
  

  function active_bits(signal vec : std_logic_vector) return std_logic_vector is
    variable rtnv : std_logic_vector(vec'range);
  begin
    for i in vec'range loop
      if active_bit(vec, i) then
        rtnv(i) := '1';
      else
        rtnv(i) := '0';
      end if;
    end loop;
    return rtnv;
  end function;
```

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:
GHDL 0.37 (v0.37) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.

- OS:  Unix

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
...
******************************************************************
```

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

tgingold, 2021-01-13:
Your reproducer is truncated!

tgingold, 2021-01-13:
On a simple example, I get:
```
repro.vhdl:10:15:error: prefix of 'active attribute must be a static name
```

Note: this restriction is stupid as you can work around it.

sckoarn, 2021-01-13:
Yes,  I see my code example is incomplete.  I will be better in the future.
Would you like me to post a more complete testcase for this?  It runs with out errors or warnings on my end.

I have coded around this issue to get to the feature I was targeting.

eine, 2021-01-13:
@sckoarn, the version of GHDL you are using seems to be almost a year old. Over 1300 commits were pushed since then. I suggest using nightly assets (see https://github.com/ghdl/ghdl/releases/tag/nightly and https://github.com/ghdl/setup-ghdl-ci).

sckoarn, 2021-01-13:
Yeah I know about the age.  Those that maintain our tools will only update to releases.  They are not into dynamic updates.  I can see if I can push for it though.   I do not want to make extra work for you guys.

Seeing if I can  push up a zip file with full code example.

sckoarn, 2021-01-13:
[ghdl_cmd.zip](https://github.com/ghdl/ghdl/files/5811035/ghdl_cmd.zip)

eine, 2021-01-13:
My only point is that Tristan might be getting an error/warning that you won't see, because it might have been introduced after 0.37.0. So, regardless of your local setup, you should try to test nightlies/master. You can do that without affecting the system version of GHDL. In other words, this is likely to be closed as "fixed in development version". But we need to confirm it first.

eine, 2021-01-13:
It tried unzipping your `ghdl_cmd.zip`:

```sh
# ghdl -a --std=02 -fsynopsys proc_pkg.vhdl
proc_pkg.vhdl:18:10: unit "subs_pkg" not found in library "work"
proc_pkg.vhdl:72:14: package "proc_pkg" was not analysed
C:\msys64\mingw64\bin\ghdl.exe: compilation error

# ghdl --version
GHDL 1.0-dev (0.37.0.r1261.g4d37b288) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 llvm code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2020 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

sckoarn, 2021-01-13:
[ghdl_cmd.zip](https://github.com/ghdl/ghdl/files/5811146/ghdl_cmd.zip)

Well, yes,  it is missing a file which should be part of this zip.
Sorry for the run around ...  I will get this process down  soon :)

sckoarn, 2021-01-13:
I am working on the powers that be regarding "being up to date"  on GHDL.  There are some questions.
How good do you think your nightly releases are?  Can we count on stability?
When will the next major release be?

I am trying to get them to run both versions.  This would enable me to test release and latest before submitting bug reports.

tgingold, 2021-01-14:
So this issue is already fixed.

closed 2021-01-14 by tgingold

eine, 2021-01-20:
@sckoarn,

> How good do you think your nightly releases are? Can we count on stability?

Nightly releases are exactly as stable as any other "regular" release, no more, no less. Exactly the same testsuite is executed. Nightly assets are only uploaded if all the tests are successful. Otherwise, users will get the latest known to work assets.

In fact, "stable" release packages do not exist anymore. Upcoming tags/releases will not contain assets. Those are to be used by packagers when creating recipes for apt, dnf, pacman, etc. Regular users will have nightlies available if they want to install GHDL without using the package manager in their system. This will hopefully avoid many misconceptions about the "stability" of regular releases, which was neither explicitly better (nor worse).

Anyway, you might want to read the following discussion in this regard: https://github.com/ghdl/ghdl/issues/1118#issuecomment-579946057. Several other users raised concerns as yours in the past, precisely trying their IT services to update GHDL.

> When will the next major release be?

Tristan typically bumps the release around february-march. Last year (exactly one year ago tomorrow) he created an issue for discussing about it: #1108.

sckoarn, 2021-01-22:
@eine Thank you for the reply.