ghdl/ghdl #1598: AXI4 Master Model of OSVVM does not Analyze

Paebbels, 2021-01-14:
**Description**
When running analysis of OSVVM (see new pre-compile scripts provided in #1594) GHDl cashes in the AXI4 Master model:

```
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:863
```

GHDL 1.0.dev0 was build in MinGW64 from sources (6496bc9045f9c07c71ca92dbe9d7afbc107c7d7d) with LLVM backend.

**Expected behavior**
GHDL shouldn't crash. I assume it triggers a not yet implemented VHDL-2008 feature.

This feature is needed to support OSVVM's new Verification IPs (verification models).

**How to reproduce?**
Analyze latest OSVVMLibraries. This can be done with latest pre-compile script for PowerShell provided in #1594. (A Bash version is coming soon.)

```
cd <workingdir>
C:\Tools\GHDL\1.0.dev0-llvm\lib\ghdl\vendors\compile-osvvm.ps1 -All -Source ..\OsvvmLibraries\ -Output precompiled -GHDL C:\Tools\GHDL\1.0.dev0-llvm\bin\ghdl.exe
```

* The script is located in `vendors` directory
* `-All` analyzes all parts of OSVVM (dependencies for e.g. AXI4 master)
* `-Source` specifies relative or absolute path to repository `OSVVMLibraries`
* `-Output` specifies where to write the resulting VHDL libraries
* `-GHDL` specifies the GHDL executable

**Context**
Please, provide the following information:

- OS: Windows 10 Pro
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: 6496bc9045f9c07c71ca92dbe9d7afbc107c7d7d

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (0.37.0.r1309.gd1112221.dirty) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
C:\git\GitHub\OSVVM\GHDL\OSVVM\osvvm_axi4\v08\
Command line:
C:\Tools\GHDL\1.0.dev0-llvm\bin\ghdl1-llvm.exe -fexplicit -frelaxed-rules --mb-comments -Wbinding -Wno-hide -Wno-others -Wno-static -fsynopsys --no-vital-checks --std=08 --work=osvvm_axi4 -PC:\git\GitHub\OSVVM\GHDL\OSVVM\ -PC:\Tools\GHDL\1.0.dev0-llvm\lib/ghdl\ieee\v08\ -PC:\Tools\GHDL\1.0.dev0-llvm\lib/ghdl\ -c -o Axi4Master.o C:\git\GitHub\OSVVM\OsvvmLibraries\AXI4\Axi4\src\Axi4Master.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : vhdl-nodes.adb:863
Call stack traceback locations:
0x652c66 0x474273 0x4a50ee 0x5710f0 0x573732 0x573903 0x57446c 0x574bd3 0x574e11 0x574f53 0x5349d7 0x53b527 0x52288f 0x637136 0x638e0a 0x63a6e5 0x402706 0x4013bf 0x4014f4 0x7ffd8f5a7032 0x7ffd8fb7d0cf
******************************************************************
C:\Tools\GHDL\1.0.dev0-llvm\bin\ghdl.exe: compilation error
```

**Additional context**
*None*

umarcor, 2021-10-24:
I believe that this issue can be closed, since the same command is being tested in https://github.com/ghdl/extended-tests/blob/main/verification/OSVVM/run.ps1 and results are successful.

Paebbels, 2025-04-27:
All of OSVVM works fine with GHDL v5.0.1.

---------------
/cc @JimLewis

closed 2025-04-27 by Paebbels