ghdl/ghdl #1601: -frelaxed / -frelaxed-rules does not work properly

Paebbels, 2021-01-15:
**Description**
In VHDL-2008 mode, shared variables create an error even when `-frelaxed` is on. A few days back when testing precompile scripts, this was not noticed.

This command was executed:
```powershell
...\lib\ghdl\vendors\compile-uvvm.ps1 -All -Source C:\git\GitHub\UVVM\UVVM -Output uvvm -GHDL ...\bin\ghdl.exe -HaltOnError -Debug
```

GHDL reports to enabled `frelaxed`, but it's already on.
```
C:\git\GitHub\UVVM\UVVM\uvvm_util\src\adaptations_pkg.vhd:70:19: type of a shared variable must be a protected type
C:\git\GitHub\UVVM\UVVM\uvvm_util\src\adaptations_pkg.vhd:70:19:note: (you can use -frelaxed to turn this error into a warning)
```

This test was executed on Windows 10 with GHDL 1.0.dev0 (d890f27e480cdb37428f1b7c4bb37475ed38348d) built with MinGW64 and LLVM backend.

**Expected behavior**
When `-frelaxed` is specified, shared variables should be a warning.
Moreover when `-Wno-shared` is set, also no warning should appear.

**How to reproduce?**
**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: d890f27e480cdb37428f1b7c4bb37475ed38348d

**Additional context**
```
Compiling library 'uvvm_util' ...
  Creating library uvvm_util ...
    mkdir C:\git\GitHub\OSVVM\GHDL\uvvm/uvvm_util/v08
    cd C:\git\GitHub\OSVVM\GHDL\uvvm/uvvm_util/v08
  Analyzing package file 'C:\git\GitHub\UVVM\UVVM\uvvm_util\src\types_pkg.vhd'
    & 'C:\Tools\GHDL\1.0.dev0-llvm\bin\ghdl.exe' -a -fexplicit -frelaxed --mb-comments -Wbinding -Wno-shared --ieee=synopsys --no-vital-checks --std=08 -PC:\git\GitHub\OSVVM\GHDL\uvvm --work=uvvm_util C:\git\GitHub\UVVM\UVVM\uvvm_util\src\types_pkg.vhd 2>&1
  Analyzing package file 'C:\git\GitHub\UVVM\UVVM\uvvm_util\src\adaptations_pkg.vhd'
    & 'C:\Tools\GHDL\1.0.dev0-llvm\bin\ghdl.exe' -a -fexplicit -frelaxed --mb-comments -Wbinding -Wno-shared --ieee=synopsys --no-vital-checks --std=08 -PC:\git\GitHub\OSVVM\GHDL\uvvm --work=uvvm_util C:\git\GitHub\UVVM\UVVM\uvvm_util\src\adaptations_pkg.vhd 2>&1
      ERROR: C:\git\GitHub\UVVM\UVVM\uvvm_util\src\adaptations_pkg.vhd:70:19: type of a shared variable must be a protected type
      C:\git\GitHub\UVVM\UVVM\uvvm_util\src\adaptations_pkg.vhd:70:19:note: (you can use -frelaxed to turn this error into a warning)
      ERROR: C:\git\GitHub\UVVM\UVVM\uvvm_util\src\adaptations_pkg.vhd:388:19: type of a shared variable must be a protected type
      ERROR: C:\git\GitHub\UVVM\UVVM\uvvm_util\src\adaptations_pkg.vhd:392:14: circular reference of package "adaptations_pkg"
      C:\Tools\GHDL\1.0.dev0-llvm\bin\ghdl.exe: compilation error
--------------------------------------------------------------------------------
Compiling UVVM packages [FAILED]
```

Paebbels, 2021-01-15:
I'm also testing with Bash variant of this script (not done yet / not pushed yet), but it shows the same behavior and it shows, that `ghdl1-llvm.exe` was started with `-frelaxed` too.

tgingold, 2021-01-15:
-frelaxed behaviour is now changed by --std=.  So be sure to place -frelaxed after --std.

eine, 2021-01-15:
There is a comment in https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-std, but not in https://ghdl.github.io/ghdl/using/InvokingGHDL.html#cmdoption-ghdl-frelaxed. We should update that.

Paebbels, 2021-01-15:
OK, my problem is solved / question is answered, but we keep it open until the documentation is improved for that point.

closed 2021-01-16 by umarcor