ghdl/ghdl #1604: When are assertions in entity_statement_part checked?

m-kru, 2021-01-16:
**Description**
It looks like assertions in the entity_statement_part are not checked before anything else is done within the entity on instantiation.

**Expected behaviour**
Checking assertions within the entity_statement_part should be done before calculations within the architecture.

**How to reproduce?**

```vhd :file: test.vhd
library ieee;
   use ieee.std_logic_1164.all;

entity test is
   generic (
      G_INPUT_WIDTH  : positive;
      G_OUTPUT_WIDTH : positive
   );
   port (
      i : in  std_logic;
      o : out std_logic
   );
begin
   assert G_OUTPUT_WIDTH < G_INPUT_WIDTH
      report "G_OUTPUT_WIDTH (" & integer'image(G_OUTPUT_WIDTH) & ") must be lower than G_INPUT_WIDTH (" & integer'image(G_INPUT_WIDTH) & ")"
      severity failure;
end entity;

architecture rtl of test is

   constant FOO : positive := G_INPUT_WIDTH - (G_INPUT_WIDTH / G_OUTPUT_WIDTH) * G_OUTPUT_WIDTH;

begin

   o <= i;

end architecture;
```
```vhdl :file:tb.vhd
library ieee;
   use ieee.std_logic_1164.all;

entity tb is
end entity;

architecture bar of tb is

   signal i, o : std_logic;

begin

   DUT : entity work.test
   generic map (
      G_INPUT_WIDTH => 3,
      G_OUTPUT_WIDTH => 3
   )
   port map (
      i => i,
      o => o
   );

end architecture;

```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a test.vhd
ghdl -a tb.vhd
ghdl --elab-run tb
```

`ghdl --elab-run tb` returns:

```
./tb:error: bound check failure at test.vhd:22
./tb:error: error during elaboration
```
This is true, `FOO` is positive and it can't be 0.
The question is why assertion has not fired?
Probably because it is checked after the calculations.
Changing type to `natural` fires the assertion.
```
test.vhd:14:4:@0ms:(assertion failure): G_OUTPUT_WIDTH (3) must be lower than G_INPUT_WIDTH (3)
./tb:error: assertion failed
in process .tb(bar).dut@test(rtl).P0
./tb:error: simulation failed
```

**Context**
GHDL 1.0-dev (v0.37.0-1121-g17d32256) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 llvm code generator

tgingold, 2021-01-16:
The assertions are run after elaboration.

m-kru, 2021-01-16:
Is it correct behavior for the assertions with the entity_statement_part?

tgingold, 2021-01-16:
Yes, even for statements of the entity.

closed 2021-08-01 by tgingold