ghdl/ghdl #1608: 'extended-tests' repository in the organisation

umarcor, 2021-01-17:
Together with @Paebbels, we've prototyped an 'extended-tests' repository for GHDL: https://github.com/umarcor/ghdl-extended-tests. The motivation is having quick feedback of relevant projects which we'd like GHDL to be compatible with, but which should not affect the regular development flow here.

For now, we setup CI for testing OSVVM and UVVM on several environments, using the vendor scripts that were updated yesterday. See https://github.com/umarcor/ghdl-extended-tests/actions/runs/490911192. Installation of GHDL is done with [setup-ghdl-ci](https://github.com/ghdl/setup-ghdl-ci), so that nightly assets are used.

The proposal is to have each target project added as a submodule. Then, [dependabot](https://dependabot.com/) allows bumping to be handled automatically. It can check repos daily, weekly or monthly; and it will open PRs for testing the bumped submodules. That allows keeping the latest known-to-work versions of all the targets tagged in the main branch, but easily test against any other version by creating a feature branch.

As referenced in the README of ghdl-extended-tests, some of the examples for ghdl-yosys-plugin, projects for synthesis or formal verification, etc. can be later added.

Ref #1291 #974 #1393

tgingold, 2021-01-17:
Great idea.  Better to automatically test and update other projects.

umarcor, 2021-01-17:
Will you create ghdl/extended-tests or ghdl/ghdl-extended-tests or the name you prefer?

tgingold, 2021-01-17:
Done

umarcor, 2021-01-17:
Thanks. The content is now upstreamed to https://github.com/ghdl/extended-tests. I removed and recreated my personal repo, so that it's now shown as a fork of the upstream.

closed 2021-01-17 by umarcor