ghdl/ghdl #1610: Add Yosys PSL custom attribute `gclk`

meck, 2021-01-17:
Currently the `Yosys` [non-standard attributes](https://github.com/YosysHQ/yosys/blob/master/README.md#non-standard-or-systemverilog-features-for-formal-verification) `anyconst` `anyseq` etc are handled and corresponding formal ports are added, however the attribute `gclk` mentioned in the same section seems not to be implemented, it does not work and a grep for `gclk` shows nothing.

**Describe the solution you'd like**
Add support for the `gclk` attribute.

tgingold, 2021-01-17:
Do you have a testcase ?

meck, 2021-01-18:
I'm not quite sure actually, maybe in above my head... if the  `any*`-family of attributes synthesizes to ports with `$any*` input, One would assume the the directive should be a port with a `$global_clock` input. I guesses this touches https://github.com/ghdl/ghdl-yosys-plugin aswell.

``` vhdl
library IEEE;
use IEEE.std_logic_1164.ALL;
use IEEE.numeric_std.ALL;

entity exp is
  port (
    clk : in    std_logic
  );
end entity exp;

architecture behav of exp is

  signal ver_clk : std_logic;
  signal count : integer := 0;

  attribute gclk : boolean;
  attribute gclk of ver_clk : signal is true;

begin

  default Clock is rising_edge(clk);

  process (ver_clk)
  begin
    if rising_edge(ver_clk) then
      count <= count + 1;
    end if;
  end process;

  assert always next count = prev(count) + 1;

end architecture behav;
```

tgingold, 2021-01-25:
Now implemented.

closed 2021-01-25 by tgingold