ghdl/ghdl #1612: GHDL Bug occurred

vdkeybus, 2021-01-17:
I will probably find a workaround for the moment.

I have removed a couple of filenames I've got to be a bit careful with. Let me know if you need them and I'll see what can be done.

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (0.37.0.r1340.g5cf09791) [Dunoon edition]
Compiled with GNAT Version: 10.2.1 20201202 
Target: x86_64-suse-linux
/home/vdkeybus/Src/qdesigns/***
Command line:
/usr/local/bin/ghdl1-llvm --workdir=work -P/usr/local/lib/ghdl/ieee/v93/ -P/usr/local/lib/ghdl/ -c -o work/***.o ***.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap6.adb:1173
Call stack traceback locations:
0x653de0 0x5a85e3 0x5a8a49 0x59ffe1 0x5a00c5 0x5a04ff 0x5a09fe 0x5a16ec 0x634a1e 0x630648 0x630d62 0x5ed7a2 0x5e0871 0x63b925 0x63d18c 0x63e776 0x40aead 0x7fe2bbf48150 0x40a02c 0xfffffffffffffffe
******************************************************************
ghdl:error: compilation error

eine, 2021-01-17:
Actually, we need any minimal reproducer you can provide... If you are not allowed to share the sources, we need you to try come up with a somehow equivalent piece of code with completely different names. Alternatively, you might try explaining which language features you are using, but that's more challenging.

Xiretza, 2021-01-17:
Looking at [the source](https://github.com/ghdl/ghdl/blob/5cf09791f2c14210226c0d3435e859107c1592f2/src/vhdl/translate/trans-chap6.adb#L1173), it's something about a constant/file/variable declaration having a `Mode_Signal` instead of a `Mode_Value`, maybe that can help in narrowing it down.

vdkeybus, 2021-01-17:
Ok thanks. Narrowed it down with a minimal example:

-------------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;

entity expose_problem is
  port (
    arg : in std_logic_vector);
end entity;

-------------------------------------------------------------------------------

library ieee;
use ieee.std_logic_1164.all;

entity expose is
  generic (
    ARG : std_logic_vector);
end entity;

architecture rtl of expose is

begin
  expose_problem_1 : entity work.expose_problem
    port map (
      arg => ARG
      );
end architecture;

eine, 2021-01-17:
By copying the whole minimal example into a single file named `exposep.vhd`, I can confirm the bug:

```
# ghdl -a --std=08 exposep.vhd 

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (0.37.0.r1270.gb65ee8c3) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
D:\data-dev\umarcor\ghdl\ghdl\temp\
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o exposep.o exposep.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap6.adb:1175
Call stack traceback locations:
0x179dce6 0x59ca31 0x59cec9 0x593f5d 0x594046 0x59443a 0x594943 0x595662 0x62fb33 0x62b1e1 0x62b942 0x5e59c6 0x5d7eaa 0x637218 0x638bba 0x63a495 0x402706 0x4013bf 0x4014f4 0x7ffaa4177032 0x7ffaa549d0cf
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

Thank @vdkeybus for reacting so fast :wink:

commit 3f1d6271 (2021-01-19): trans-chap5: use value mode to get object bounds.  Fix #1612
closed 2021-01-19 by tgingold