ghdl/ghdl #1618: Compiling Wishbone BFM for UVVM

m-kru, 2021-01-20:
@Paebbels I see you have modified compile scripts. Why have you removed support for compiling UVVM_VIP_WISHBONE from compile-uvvm.sh?

Paebbels, 2021-01-20:
@m-kru  
UVVM doesn't provide a Wishbone VIP.
Please see https://github.com/UVVM/UVVM/blob/master/script/component_list.txt

> Something is eating last letter in the paths. uti instead of util, framewor instead of frameowork.

The Problem with `uti` and `util` should be fixed in the current version on `master`.
As far as I know, CI is not complaining here. It was a problem some commits ago and was fixed.

> And why the heck introduce more bashism to the shell scripts instead of de bashing them?

The scripts are written for Bash, so Bash features are used! I'll not extend the script size to workaround
missing shell features because e.g. Dash is used.

m-kru, 2021-01-20:
@Paebbels

> UVVM doesn't provide a Wishbone VIP.

It does. https://github.com/UVVM/UVVM_Community_VIPs. I onced asked UVVM to put it into the UVVM, but the response was that the user should copy the directory with wishbone stuff to the UVVM himself. It can't be compiled when it is not in the UVVM directory.

>  The Problem with uti and util should be fixed in the current version on master.
> As far as I know, CI is not complaining here. It was a problem some commits ago and was fixed.

I use current master.

> The scripts are written for Bash, so Bash features are used! I'll not extend the script size to workaround
> missing shell features because e.g. Dash is used.

The scripts have been initially written for Bash, but this is not a reason to introduce more bash to them.
There are people using GHDL on FreeBSD and on Linux distributions without the bash.
I guess you are the oracle, which knows best what shell others _should_ be using.
Writing portable shell scripts is informal standard in *nix operating systems.

eine, 2021-01-20:
> It does. [UVVM/UVVM_Community_VIPs](https://github.com/UVVM/UVVM_Community_VIPs). I onced asked UVVM to put it into the UVVM, but the response was that the user should copy the directory with wishbone stuff to the UVVM himself. It can't be compiled when it is not in the UVVM directory.

It seems there is a miscommunication here. If users need to explicitly add an specific VIP, while all others are included by default, I would say it is obvious that the VIP is not provided as part of UVVM. It is an optional community contributed extension, as the name of the separated repo suggests. I guess that "UVVM Community" was never supported in GHDL's vendor scripts. Therefore, that's a new feature request. You might well talk to UVVM maintainers so they provide a "Full UVVM" repository/package, which includes community resources.

> I use current master.

Please, let us know which version of GHDL and UVVM are you using, and which procedure are you following.

> The scripts have been initially written for Bash, but this is not a reason to introduce more bash to them.
> There are people using GHDL on FreeBSD and on Linux distributions without the bash.

Using a single bash-only feature is enough for the scripts not to work on FreeBSD or other Linux distributions. If that single feature is not to be removed, it is pointless to avoid using others, since the premise will never be false.

> I guess you are the oracle, which knows best what shell others _should_ be using.
> Writing portable shell scripts is informal standard in *nix operating systems.

I believe he is not telling anyone what to do. He is explaining why he is not going to put effort into maintaining poorly-featured environments, just for the sake of purity. Of course, users of those environments are welcome to remove all the bashism and help maintain equivalent scripts that work "anywhere". Are you up to it?

Paebbels, 2021-01-20:
@m-kru we can discuss how the UVVM scripts can support another repository / directory providing the same structure as UVVM to be compiled into a single output directory. The script should work without copying sources and adjusting component files in the original UVVVM. If we go this step, it should be like a  `--addons ../UVVM_Community_VIPs` option to the main script.

If you reject using Bash scripts, you're free to use [PowerShell 7](https://github.com/PowerShell/PowerShell) and run the PowerShell variants of the scripts.

---------------
> I use current master.

I'll check with @eine if CI covered this case and come back to you.

m-kru, 2021-01-20:
> It seems there is a miscommunication here. If users need to explicitly add an specific VIP, while all others are included by default, I would say it is obvious that the VIP is not provided as part [...]

It is a dead lock. UVVM does not want to move wishbone to the UVVM repository. GHDL does not want to support it in the uvvm-compile script because it is not in the UVVM repository.

I will poke UVVM again.

> Please, let us know which version of GHDL and UVVM are you using, and which procedure are you following.

GHDL branch `master` commit `f90c`. UVVM version is irrelevant as the problem is with the path to the UVVM VIPs.

> Using a single bash-only feature is enough for the scripts not to work on FreeBSD or other Linux distributions. If that single feature is not to be removed, it is pointless to avoid using others, since the premise will never be false.

I agree, but it is easier to make script portable when there is single bash feature than when the script is full of bashism, isn't it?

eine, 2021-01-20:
@m-kru, we did not say "GHDL does not want to support it in the compile script". We both said that it can be supported. I did specify that it is not suprising for it to be missing: support was not removed, because it was non-existent. Hence this issue is a feature request, not a bug report in that regard.

> I will poke UVVM again.

Note that we do not retrieve UVVM in the scripts. We cannot support installation procedures that involve downloading and relocating multiple repositories, because scripts assume that sources are already available in the disk.

As Patrick said, `--addons ../UVVM_Community_VIPs` might be added to the script. Still, most users will need to find guidelines about how to download and relocate the repositories. Those guidelines belong in the UVVM or UVVM_Community_VIPs repos, since are independent from GHDL. We, as a simulator, cannot take care about the documentation and community management of third-party projects (libraries).

All of this is also true for OSVVM. Precisely, Jim is providing guidelines and recommended installation/testing procedures which we can use in CI, but which do not belong in the scripts. At the same time, we are discussing with him what we can handle in this org and what needs to be done elsewhere.

I strongly recommend UVVM maintainers, you, or any other contributor in the UVVM community to get involved in [ghdl/extended-tests](https://github.com/ghdl/extended-tests). The more love/care that repository receives, the better the user experience will be for users of UVMM + GHDL. There, we can install multiple repos (as submodules) and we can do some further tests than what's acceptable in the root repo.

> UVVM version is irrelevant as the problem is with the path to the UVVM VIPs.

It should be, but it might not. Please provide the version of UUVM.

> I agree, but it is easier to make script portable when there is single bash feature than when the script is full of bashism, isn't it?

Strictly, there is no "easier" or "harder" solution, because it's a purely hypothetical question. Is anyone willing to contribute to this repo so that vendor shell scripts are portable (including future enhancements/additions)? Should that person exist, I agree it'd be easier. Yet, no one stepped ahead in the last two years, even though Patrick's WIP PR was clearly shown, and many users reported problems with UVVM, OSVVM, Xilinx, Lattice... At the moment, there is still no one who volunteered to do it. It is not a priority for neither Tristan nor me, and Patrick did already state that he is ok with the bash constraints. That's why I referred to "purity" before. Hypothetically it would be nice. In practice, it is a very low relevance issue.

Honestly, if I were to put any effort into vendor scripts, I would remove them (both bash and psh). I would use Python instead. 5-6 years ago, I hated Python, and I even used VUnit without Python. Today, I still don't like it much, but "resistance is futile". Moreover, I would very seriously consider writing a single line of code in this repo. I would do my best for extending FuseSoC, edalize, pyIPCMI, tsfpga, PyFPGA, Xeda... Those are all "library management" projects. All vendor libs can be used with simulators other than GHDL, and it makes sense to have a single declaration that is reusable regardless of the simulator. The only motivation for vendor scripts to exist, is that those 6 projects are fragmented and not collaborating yet. When we get them to work together, vendor scripts won't be necessary. Precisely, that's why VUnit was removed from vendor scripts. That's the only project which does already provide a Python based cross-platform working solution for GHDL and many other simulators.

m-kru, 2021-01-20:
> I did specify that it is not surprising for it to be missing: support was not removed, because it was non-existent. 

It was supported. I have added support in commit `ba0bc`.

> Note that we do not retrieve UVVM in the scripts. We cannot support installation procedures that involve downloading and relocating multiple repositories, because scripts assume that sources are [...]

I understand your point. I do not think adding extra argument is good idea, it would be slick handling of some corner case. It is UVVM that is doing weird thing, God only knows why they put Wishbone BFM into separate repository. If they do not change their minds I guess I will switch to OSVVM or Vunit Wishbone BFM. I am not a big of adding complexity.

> Honestly, if I were to put any effort into vendor scripts, I would remove them (both bash and psh). I would use Python instead. 5-6 years ago, I hated Python, and I even used VUnit without Python. [...]

I agree, I use this scripts only because FuseSoc/GHDL does not cache results and reanalyzing verification frameworks take relatively long. Actually this is good question, whether it is GHDL or wrapper tool that should support caching?

eine, 2021-01-21:
> > I did specify that it is not surprising for it to be missing: support was not removed, because it was non-existent.
> 
> It was supported. I have added support in commit `ba0bc`.

Let me apologise. You are partially correct. Let's see if we can put it clear: https://github.com/ghdl/ghdl/commits/12d669cb9c048d09a0e103a6123cfe0a4aad0800/libraries/vendors/compile-uvvm.sh

- In December 2019, the wishbone vip was available in UVVM repos, along with other vips. You contributed a patch for having it added in `compile-uvvm.sh` (only).
- Sometime before August 2020, the wishbone vip was removed and located in a different not-submoduled repo.
- In August 2020, `compile-uvvm.sh` was updated according to the new list of vips in the main repo. The wishbone vip was removed.
- In January 2021, both bash and powershell scripts were updated, for reading the `compile_order.txt` files, where the wishbone vip is not found by default.

However, there is https://github.com/UVVM/UVVM_SUPPLEMENTARY repo, which does contain the wishbone vip, and which seems to be up to date 😕 .

> I understand your point. I do not think adding extra argument is good idea, it would be slick handling of some corner case.

The idea of the extra argument is not for this corner case only. It's also for allowing in-house or third-party vip libraries that follow the same pattern. If users provide sources and a `compile_order.txt` so that their library looks like a UVVM one, the same logic used for adding official vips can be used for those. 

> I agree, I use these scripts only because FuseSoc/GHDL does not cache results and reanalyzing verification frameworks take relatively long. Actually this is good question, whether it is GHDL or wrapper tool that should support caching?

My understanding is that neither FuseSoc nor GHDL need to handle that.

The strengh of FuseSoc is the CAPI format/syntax; the `*.core` files. It is an alternative to IP-XACT actually usable by humans. I believe that the Python part of FuseSoC is less relevant, unless it allows composition of SoCs by plugging multiple cores to certain buses. I don't think it does, and that's why migen/nmigen are used for that (or Chisel, SpinalHDL, etc.). As a community, we should try other "management" tools to understand `*.core` files, and to extend the format (v3?) if required. See https://docs.google.com/document/d/1qThGGqSVQabts-4imn5zY5BMptp1-Q2rGiNKHDH1Pbk. That document was created in the context of @hdl. 

GHDL does already have some features. When doing synthesis, you can provide all files at once, in any order, and let GHDL guess which is the top-level unit. You can optionally tell which one to use. Hence, it can potentially decide which files need to be just parsed quickly and which need to be thoroughly analyzed and elaborated. By the same token `ghdl -i` and `ghdl -m` provide a user experience similar to using `make`. However, as a user, when I do `ghdl -a`, I want GHDL to analyze; I'm asking it explicitly. I don't want it to be too clever. Since the build scripts of most verification frameworks use `-a` explicitly, that's what we get.

VUnit has an internal dependency scanner, and it keeps track of which libraries need to be recompiled. By default, it will build each project once in a local directory `vunit_out`. From there on, only the files which were modified need to be recompiled. That initial build of the framework(s) for each project might be not acceptable for some users. In those cases, the vendor scripts in this repo come handy. Users can pre-build the frameworks, and then use them in VUnit by passing `-P` to GHDL. Then, VUnit will only do dependency scan and conditional compilation on the project sources. If you are developing parts of UVVM, I do recommend using a minimal VUnit `run.py` file for handling re-compilation. You don't need to use any of VUnit's VHDL libraries, if you don't want to. You can ignore the logger, tha arrays and the verification components. You need 5 lines to setup the Python-VHDL coordination, and then you can write your regular UVVM tests.

The main caveat of VUnit's nice dependecy scanning and conditional compilation features is that it's for simulation only. Synthesis is out of the scope. It would be interesting if those features were used by some other project targeting synthesis too. For instance, tsfpga uses VUnit for simulation, but it's a project manager for Vivado too. The dependency scanner was also used by TerosHDL for some time. Then, they decided to try providing a JavaScript only solution. The other caveat is that VUnit's dependency scanner is based on regular expressions. It works nice enough, but we cannot ask much more than it already provides. Looking into the future, I think that pyVHDLModel is the way to go. A reusable dependency scanner should be written on top of that. Then, projects would use pyVHDLParser, pyGHDL or any other backend. When the time comes, that will probably be my proposal for VUnit.

In the mid term, I would like VUnit, tsfpga and PyFPGA to read/write (i.e. import/export) `*.core` files, which might require enhancing CAPI. If that happened, I assume that `*.core` files would automatically replace `hdl-prj.json` (for ghdl-language-server).

m-kru, 2021-01-21:
> BTW, some fixed were pushed to master.

There is still problem with the first line:
```
[WARNING] Vendor directory '/usr/local/lib/ghdl/vendors/.' already exists.
./compile-uvvm.sh: line 425: test: -eq: unary operator expected
./compile-uvvm.sh: line 425: test: -eq: unary operator expected
[ERROR] Compile order file '/home/mkru/workspace/UVVM/bitvis_vip_clock_generato/script/compile_order.txt' does not exist..
```
'r' got eaten. Other VIP compile or not depending if they require bitvis_vip_clock_generator which failed to compile.

>  I believe that the Python part of FuseSoC is less relevant, unless it allows composition of SoCs by plugging multiple cores to certain buses. I don't think it does, and that's why migen/nmigen are used for that (or Chisel, SpinalHDL, etc.).

This is the beauty of FuseSoc, it follows Unix philosophy much better than migen/nmigen etc. All the stuff related with bus and connecting modules to the bus is not a part of the FuseSoc. You couple FuseSoc with external generator responsible for the whole bus part. See [AGWB](https://github.com/wzab/agwb). Idea, that the dependency/built tool should know anything about the bus is at least weird.

eine, 2021-01-21:
> There is still problem with the first line:
>
> ```
> [WARNING] Vendor directory '/usr/local/lib/ghdl/vendors/.' already exists.
> ./compile-uvvm.sh: line 425: test: -eq: unary operator expected [...]

See https://github.com/ghdl/ghdl/commit/55abae8769c53171b1c3caa4dcac7f1fdbc5b65c.

There is still the error with line 425. But the build should succeed.

> This is the beauty of FuseSoc, it follows Unix philosophy much better than migen/nmigen etc. All the stuff related with bus and connecting modules to the bus is not a part of the FuseSoc. You couple [...]

Interesting reference. Thanks! Then, what I don't get yet is what is FuseSoC's Python part for. Does it provide an API based on classes, which represents the CAPI data (smilar to pyVHDLModel)?

m-kru, 2021-01-21:
It looks like the problems with script have been solved. It also looks like GHDL is not able to analyze correctly files from UVVM current master (commit c1aa)
```
  ERROR: /home/mkru/workspace/UVVM/uvvm_util/src/string_methods_pkg.vhd:1280:36: constant interface "val" was not annotated with attribute "element"
  ERROR: /home/mkru/workspace/UVVM/uvvm_util/src/string_methods_pkg.vhd:1320:36: constant interface "val" was not annotated with attribute "element"
  ERROR: /home/mkru/workspace/UVVM/uvvm_util/src/string_methods_pkg.vhd:1360:36: constant interface "val" was not annotated with attribute "element"
  FATAL: /usr/local/bin/ghdl: compilation error
```
https://github.com/UVVM/UVVM/blob/c1aa7ff906c4a0789f93eab95caf4605d954d278/uvvm_util/src/string_methods_pkg.vhd#L1280
but this is a different story, potentially a problem in the UVVM itself.

> Interesting reference. Thanks! Then, what I don't get yet is what is FuseSoC's Python part for. Does it provide an API based on classes, which represents the CAPI data (smilar to pyVHDLModel)?

Can you explain what you mean by 'FuseSoc Python part'?

eine, 2021-01-21:
> It looks like the problems with script have been solved. It also looks like GHDL is not able to analyze correctly files from UVVM current master (commit c1aa)

There are still some issues, but it does work, yes. 

Note that the submodule in https://github.com/ghdl/extended-tests/tree/main/verification/UVVM is not in master, but slightly behind. Master is not buildable at the moment. See #1614, #1600 and #1593.

> > Interesting reference. Thanks! Then, what I don't get yet is what is FuseSoC's Python part for. Does it provide an API based on classes, which represents the CAPI data (smilar to pyVHDLModel)?
> 
> Can you explain what you mean by 'FuseSoc Python part'?

To me, the interesting and reusable part of FuseSoc is https://fusesoc.readthedocs.io/en/latest/ref/capi2.html. However, 98.1% of the repo is Python code: https://github.com/olofk/fusesoc/tree/master/fusesoc.

m-kru, 2021-01-21:
CAPI2 (`.core`) is solely the format (based on YAML) for describing cores, dependencies etc.. FuseSoc reads `.core` files, resolves dependencies, calls generators and then directs the job to the proper EDA tool (such as GHDL, Vivado etc.) - this is all implemented in Python.

When all you cores (VHDL/SV code) and generators are ready, you describe your design in fully declarative way by writing `.core` files. You do **not** describe the design structure in Python files. You can use Python for example for writing generators, but such generators can be written in any language, as generator is simply an external program, which is called by FuseSoc with some arguments defined in proper `.core` file.

eine, 2021-01-21:
> FuseSoc reads .core files, resolves dependencies, calls generators and then directs the job to the proper EDA tool (such as GHDL, Vivado etc.) - this is all implemented in Python.

> as generator is simply an external program, which is called by FuseSoc with some arguments defined in proper .core file.

My point is that I don't need this to be tied to `*.core` files. I want to use them from VUnit, GHDL, PyFPGA, tsfpga. I don't want FuseSoC to call generators, because I don't want it to be the root/manager of the execution. I'd rather have FuseSoC be a library which provides a Python API to CAPI compliant declarative files. 

Precisely the main issue with the current FuseSoC-Edalize-VUnit integration is that FuseSoC-Edalize behave as a generator of `run.py` files. I don't want that, but users to write their `run.py` files as they have always done. All I want is to replace:

https://github.com/VUnit/vunit/blob/master/examples/vhdl/array_axis_vcs/run.py#L27
```py
VU.add_library("lib").add_source_files([SRC_PATH / "*.vhd", SRC_PATH / "**" / "*.vhd"])
```

with

```py
from fusesoc import fusesocParser

VU.importFusesocCore(fusesocParser(SRC_PATH / "*.core"))
```

or

```py
from fusesoc import fusesocParser, fusesocExportToVunit

_core = fusesocParser(SRC_PATH / "*.core")
fusesocExportToVunit(VU, _core)
```

or

```py
from fusesoc import fusesocParser, fusesocExportToVunit

_core = fusesocParser(SRC_PATH / "*.core")
_core.fusesocExportToVunit(VU)
```

There, `fusesocParser` would be a Python function provided by FuseSoC for reading a core file and providing a Python object/class representing the data in a pythonic way. Then, `fusesocExportToVunit` or `importFusesocCore` would pick that Python object/class and use VUnit existing API for declaring libraries, sources, options, etc. The difference between them is the location (VUnit's codebase or FuseSoC's library's codebase).

I think that FuseSoC's features as a package manager for resolving dependencies, etc. are very interesting, but that's a feature which should be built on top of a reusable library. Not as the only consumer of `*.core` files, in practice.

m-kru, 2021-01-21:
> My point is that I don't need this to be tied to *.core files. I want to use them from VUnit, GHDL, PyFPGA, tsfpga. I don't want FuseSoC to call generators, because I don't want it to be the [...]

This contradicts with FuseSoc paradigm.

I understand your point. I have faced the same issue one and a half year ago, when I wanted to integrate vunit into edalize (vunit was not supported back then). Then I have realized, that vunit does not fit well into FuseSoc paradigm. They somehow contradicts. Both tools want to control what libraries are created and what files are placed into which library. Some of their functionalities overlap. At that point, I made up my mind to implement verification runner based on FuseSoc. I was driven by following reasons:

1. Back then Vunit was monolithic (I do not know if it has changed). It was test bench runner + libraries + some minor things. I did not like this approach. In my humble opinion verification libraries should not be coupled with the test bench runner at all.
2. FuseSoc has already provided fully declarative way for defining libraries, dependencies, targets etc. in unify way for different EDA tools. In such case, it is much more reasonable to put test bench runner on top of the FuseSoc, not the other way. Compare this:
```
Test Bench Runner -> FuseSoc -> GHDL
```
with this:
```
FuseSoc -> Test Bench Runner (vunti in this case) -> GHDL
```
The second one is a bit pointless as FuseSoc can directly "talk" to the GHDL. Putting test bench runner in between is redundant.
 
What was needed, was a thin test bench runner that would simply collect test bench targets from `.core` files, and run these targets with `fusesoc`. This is why I have implemented [fsva](https://github.com/m-kru/fsva). This approach is extremely flexible:
1. Verification libraries are completely decoupled from the test bench runner. You can use Vunit, OSVVM, UVVM, it does not matter, at all. 
2. The same, fully declarative format is used for describing both cores and test benches files. They can be placed in the same `.core` if the test bench is core specific, or you can easily create special cores for test benches.
3. You can use the full power of FuseSoc hooks for cosimulation.

I would really appreciate, if you could clone https://github.com/m-kru/vhdl_simple, install [fsva](https://github.com/m-kru/fsva) with
```
pip install --user fsva
```
and play with it for a while. Run `fsva` within the repository. Take a look at `.core` files to see how test bench targets are declared. Also take a look at the `_fsva` output directory. Run single target with `fsva :vhdl_simple:register_chain tb_` and check `_fsva/tmp`. It automatically creates waveform for Gtkwave.

I do not want you to abandom Vunit verification libraries (they are great!). I just would like you to try another test bench runner and ask yourself a question which approach is better for being used with FuseSoc. Also compare lines of code of fsva and vunit. fsva is so lightweight, because it is built on top of FuseSoc, it does not overlap with FuseSoc at all.

tmeissner, 2021-01-21:
@m-kru Is there any more complex example in the vhdl_simple repo? Some with external dependencies for example. And another question (I don't know fusesoc at all), does fusesoc does all the necessary things to run the simulator? Compile, simulation options, coverage options etc.?

m-kru, 2021-01-21:
@tmeissner

> Is there any more complex example in the vhdl_simple repo? Some with external dependencies for example.

Unfortunately, I have more complex examples in the closed source repositories.
However, the whole dependency system works exactly the same as in FuseSoc (because it is FuseSoc), so you can look for FuseSoc examples with dependencies.

> And another question (I don't know fusesoc at all), does fsva/fusesoc all the necessary things to run the simulator? Compile, simulation options, coverage options etc.?

Yes it does.

Only coverage is lacking. I am looking for a way how to handle this, as coverage works only with gcc backend. However it generates waveforms and psl reports by default. No need to remember or write anywhere the options. The philosophy is quite simple. Enable anything related with verification by default.
Here a short snippet from the output directory:
```
_fsva/
├── 2021-01-16_13-58-49
│   ├── summary
│   ├── vhdl_simple-multiplexer
│   │   └── tb_
│   │       ├── ghdl.ghw
│   │       ├── ghdl_psl_report.json
│   │       └── output.txt
│   ├── vhdl_simple-narrowing_register
│   │   └── tb_
│   │       ├── ghdl.ghw
│   │       ├── ghdl_psl_report.json
│   │       └── output.txt
│   ├── vhdl_simple-piso_register
│   │   ├── tb_default_generics
│   │   │   ├── ghdl.ghw
│   │   │   ├── ghdl_psl_report.json
│   │   │   └── output.txt
│   │   └── tb_msb_first
│   │       ├── ghdl.ghw
│   │       ├── ghdl_psl_report.json
│   │       └── output.txt
```

eine, 2021-01-22:
@k-kru, in full honestly, I must say I found your initial messages in this issue rather unpolite, and that's why I was somehow rude. However, I'm really liking this very insightful dialogue about FuseSoC and VUnit. Please keeep up!

> This contradicts with FuseSoc paradigm.

I am aware of that and, it's the only relevant point I want to bring. There is some background here, so let me provide references (in chronological order):

- olofk/fusesoc#301 and olofk/edalize#49.
- [January 29, 2020 10:00 PM](https://gitter.im/vhdl/General?at=5e31f260fe0e6f74e9e1966d)
- rodrigomelo9/verifying-foss-hdl-synthesizers#7 and eine/vhdl-cfg#1 and rodrigomelo9/FOSS-for-FPGAs#1.
- [May 13, 2020 4:19 PM](https://gitter.im/VUnit/vunit?at=5ebc020e2baa191854684b2c)
- VUnit/vunit#651 and [ktbarrett/vunit-cocotb](https://github.com/ktbarrett/vunit-cocotb) and [umarcor/vunit-cocotb](https://github.com/umarcor/vunit-cocotb).

That was my motivation for derailing this discussion. I did tactically write several paragraphs about FuseSoC, CAPI, make and VUnit because I believe that is more valuable for the community from an strategical point of view; instead of wasting time arguing about a third-party project whose maintainers and employees are rarely (ever) proactive.

I now added cocotb to the cocktail. That is also tactical. The problem for integrating cocotb and VUnit is the same: both paradigms assume to be the top-level manager of the execution, plus both do have test runner features. However, unlike conflicts/duplications between other projects, in this case maintainers or advanced users of both frameworks discussed their points, addressed technical and demographical differences, and are now working towards making integration possible. There are two strategies, which are closer to the userbase of one project or the other, but the nice and ditchable features of each project were laid out; and both strategies benefit from the advances of the other.

NOTE: I must clarify that OSVVM and VUnit maintainers did also try to bring some features together, such as the logging in both frameworks. They put time and effort into it, but unfortunately it didn't work back then. I do honestly think the context changed, and it can be tried again in the next months/years, maybe with a less ambitious integration in mind.

NOTE: Not so many years ago, the market model of OSVVM and UVVM was traditional: based on keeping many of their resources non-public. Hence, their trainings and consultancy fees were the only legal mechanism for using those frameworks. That is not true anymore. OSVVM and UVVM are mostly open source now. As a result, they cannot prevent other users/companies making money with their libraries; i.e. learning about them, copying implementation ideas without attributing them, and/or offering training/consultancy independently. They had to switch their strategy. That was probably very painful for them, and is probably still. At the same time, it's the best thing that could happen to us, the wider open source community.

Overall, we all agree that VUnit has two very well designed and industry-proven Python interfaces. One of them is the simulator interface, which provides source compilation and simulation execution features. The other one is the interaction with VUnit's HDL runners, and how it allows managing multiple tests defined in multiple HDL sources. AFAIAA, all open source verification projects (UVVM, OSVVM, cocotb) have evaluated during the last 5 years whether they should use VUnit's Python features. That is not just a random thing. And the reason for it not to come true was not always technical.

Apart from that, it is very arguable whether many other features of VUnit are the best solution. If we dive into logging features, custom/helper types, verification components, etc. I do agree there is a lot of room for discussion. And that is very nice! Users can install VUnit + OSVVM + UVVM + (soon) cocotb all together, and compare/decide which specific features of each framework they want to use. Some are unique (OSVVM randomization, or cocotb's VPI/VHPI support), some other are a common ground. If they all did their homework properly, users can connect verification components from any of the frameworks, arbitrarily.

---

Back to edalize/fusesoc. Because of the points we are discussing, I found/find the split merely cosmetic. Two repositories were created. The tool excution was split from the declarative file parsing. But the architecture didn't change. It allows maintainers of other projects to use edalize transparently and without FuseSoC. But it does not allow them to use FuseSoc transparently and without edalize. As result, edalize is as far as I am aware the only backend that can be used with `*.core` files, unless someone writes a custom generator. Writing a custom generator implies duplicating the APIs of the projects for having them forwarded in FuseSoC. That is, the FuseSoC-edalize ecosystem is funneling multiple source declaration projects/solutions into a single backend ([forums.ohwr.org/t/hdlmake-vs-fusesoc](https://forums.ohwr.org/t/hdlmake-vs-fusesoc/)). As a community, we should do the opposite: funnel all the existing backends/projects to consume a single declarative/imperative format (a human-readable one, not IP-XACT). That would remove the main motivation for users not to try different backends: the need to describe and maintain duplicated configuration/declaration files. At the same time, it would be more [... cut ...] ps to peek over what others are doing. In the next Wilson Study, "open source verification" should be a single answer/choice, because we as a community should be able to state loudly that we could provide a complex and distributedly developed solution more feature complete than any vendor alternative. Which would allow us and vendors to stop wasting time on this and focus on pushing the state of the art forward. The competition was fun and intense for three decades. Let's grow up now (together)!

m-kru, 2021-01-22:
> @k-kru, in full honestly, I must say I found your initial messages in this issue rather unpolite, and that's why I was somehow rude.

It was, stand in my shoes:
1. removed functionality,
2. broken script,
3. step aside from portability.

Nevertheless, it lead to conclusions why these scripts are needed at all and that theoretically they could be removed solving all the problems they create right now.

> Back to edalize/fusesoc. Because of the points we are discussing, I found/find the split merely cosmetic. Two repositories were created. The tool excution was split from the declarative file [...]

A lot of points touched here, so let me refer to them one by one.

> Back to edalize/fusesoc. Because of the points we are discussing, I found/find the split merely cosmetic.

In my humble opinion FuseSoc/Edalize is very well structured (it follows the Unix philosophy). FuseSoc is like a common preprocessing module for different backends (such as GHDL, Vivado etc.) which are placed in the Edalize. Edalize contains code for "talking" to the EDA tools. Although they are in single repository, they are completely decoupled.

> The tool excution was split from the declarative file parsing. But the architecture didn't change. It allows maintainers of other projects to use edalize transparently and without FuseSoC. But it does not allow them to use FuseSoc transparently and without edalize.

Actually with a little effort it does, but no one has used it this way. For example, I didn't have such need as Edalize already supported what I needed, if it didn't I prepared PR.

> As result, edalize is as far as I am aware the only backend that can be used with *.core files, unless someone writes a custom generator. Writing a custom generator implies duplicating the APIs of the projects for having them forwarded in FuseSoC.

This part I do not understand, do you mean FuseSoc generator? What kind of API duplication do you mean?

> That is, the FuseSoC-edalize ecosystem is funneling multiple source declaration projects/solutions into a single backend (forums.ohwr.org/t/hdlmake-vs-fusesoc). 

By backend do you mean Edalize? 
I would not call it single backend.
It is much more like a set of decoupled backends with common API.

> As a community, we should do the opposite: funnel all the existing backends/projects to consume a single declarative/imperative format (a human-readable one, not IP-XACT). That would remove the main [...]

I understand your point, but in my humble opinion this is utopia, because:

1. Proprietary tools will never follow this approach, so you will always need to handle them with current approach.
2. Open source tools already define how users should interact with them. I do not believe all developers suddenly add support for CAPI2 (`.core` files) format as an input (even if there was perfect, simple API provided by FuseSoc).

> In the open source VHDL verification framework ecosystem, we don't need edalize and it does not provide any value for simulation. it is too far behind and there is much duplication work required for [...]

I do not agree, but everyone may have its own opinion. Using fsva -> FuseSoc -> Edalize for simulation/verificatoin made my life much simpler. I use the same format (CAPI2) for everything. I do not need to (and do not want to) use multiple formats depending on if I do simulation or synthesis.

> So, we do definitely want to allow interaction. But we want "traditional" companies/developers who are using vendor tooling to be exposed to the open source ecosystem by changing a few options, not by forcing them to learn a whole new tool.

To be honest I do not care about companies that make money out of open source and do not contribute or their contribution is minimal (Sigasi?). I care about backward compatibility though. But I am not going to choose one path over the other just because of some companies if neither path breaks backward compatibility. They do not share a cake with me.

> Moreover, it does not make sense to write a custom JSON import, if FuseSoC's CAPI is already available and used by other backends (edalize). At the same time, it doesn't make sense to write a *.core [...]

Yes, this is possible. Unfortunately, FuseSoc code is not well documented. I guess if someone does this (make FuseSoc accessible also as library) it will be accepted.

> Why is this not done yet? Because this is "open source hardware tooling". Each word in there makes the target audience smaller.

I like these 2 sentences :smile:.

> I did neither ignore nor forget your fsva. I have been following it since you mentioned some months ago, and I did also evaluate how to include it in the ecosystem. My understanding is that what you [...]

I do not know whether it is "test runner" or "task runner", we would need some formal definitions for these (maybe there already are some?) to identify. You can still define multiple tests within single file. You can also generate tests dynamically with FuseSoc generators feature.

> Nevertheless, fsva is still relevant and useful. It is two abstraction layers on top of VUnit's test runner. Therefore, it does not conflict with anything, it is a complement. Some users will stick [...]

I guess it will always remain. You can not suddenly changed paradigm of a mature tool.

[... cut ...]
Not really. I used Vunit for some time (before I have discovered FuseSoc), I was not a master of Vunit for sure. But then I just came to the conclusion that Vunit does not suit well with FuseSoc.

> Which would allow us and vendors to stop wasting time on this and focus on pushing the state of the art forward. The competition was fun and intense for three decades. Let's grow up now (together)!

These are big corporations. The EDA market is an oligopoly, they care about money. But hey, so far no one has prohibited daydreaming (although social media giants for sure will try).

closed 2021-03-29 by m-kru