ghdl/ghdl #1621: Error when the file location consists of non-ASCII characters

machitgarha, 2021-01-24:
**Description**
When a VHDL source file is in a location made up with at least one Persian (i.e. or maybe more generally, non-ASCII) characters, after analyzing it, the analyzation of next VHDL source file(s) will fail, if the work directory remains the same.

This happens because the work file (e.g. `work-obj93.cf`) will include those directory name characters, and GHDL reports it as an error ([here](https://github.com/ghdl/ghdl/blob/master/src/vhdl/vhdl-scanner.adb#L527)).

Is this really the default behavior? I don't think so.

**Expected behaviour**
Analyze the files without problems.

Instead, it gives a bunch of the following error:

```
work-obj93.cf:2:42:error: invalid character not allowed, even in a string
work-obj93.cf:2:44:error: invalid character not allowed, even in a string
...
```

**How to reproduce?**
1. Create a directory named Persian and navigate into it, like:

    ```
    mkdir آزمایشی && cd آزمایشی
    ```

2. Create one (or more) minimal working VHDL code, e.g. `test.vhd`.

    ```
    library ieee;
    use ieee.std_logic_1164.all;

    entity comparator_1_bit is
        port(dummy: in std_logic);
    end entity;
    ```

3. Analyze the file. No problems here.
4. Re-analyze either the same file, or another one. Errors should be appeared.

**Context**
- OS: Fedora Workstation 33 (Linux kernel 5.10.7)
- Origin:
  - [x] Package manager: `4.5.2` (DNF)
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

**Additional context**
- GHDL version: `1.0-dev`

tgingold, 2021-01-25:
Interesting.  That's a use case I didn't think about.

Which encoding are you using ?

I plan to rewrite the library manager.  I will consider this enhancement then.

machitgarha, 2021-01-26:
> Which encoding are you using?

UTF-8.

> I plan to rewrite the library manager. I will consider this enhancement then.

Good news!