ghdl/ghdl #1622: Multiple definitions linkage errors on elaboration after analyzing two different files with the same name

machitgarha, 2021-01-26:
**Description**
When you analyze two different entity files located in two different places but with exactly the same name, after elaboration, you will get a bunch of annoying linkage errors representing something is defined twice.

**Expected behaviour**
Elaboration (and running) without any errors. Instead, a bunch of multiple definition errors.

**How to reproduce?**
1. Create two directories, `src` and `tests`.
2. Create two files with the same name inside them, e.g. `src/something.vhd` and `tests/something.vhd`.
3. Create two distinct entities in the files, in which you create an instance of the first entity inside second one's architecture. A minimal working example:

    ```vhd :file: src/test.vhd
    -- src/test.vhd
    library ieee;
    use ieee.std_logic_1164.all;

    entity something is
        port(
            clock: out std_logic
        );
    end entity;

    architecture structural of something is
    begin
    end architecture;
    ```

    ```vhd :file: tests/test.vhd
    -- tests/test.vhd
    library ieee;
    use ieee.std_logic_1164.all;

    entity other_thing is
    end entity;

    architecture structural of other_thing is
        component something is
            port(
                clock: out std_logic
            );
        end component;

        signal clock: std_logic;
    begin
        something_instance: something port map(clock);
    end architecture;
    ```
4. Analyze the files (no problems should happen):

    ```sh :image: ghdl/ghdl:buster-mcode
    ghdl -a --workdir=build/ {src,tests}/test.vhd
    ```

5. Elaborate the files:

    ```sh :image: ghdl/ghdl:buster-mcode
    # You will get errors after this
    ghdl --elab-run --workdir=build/ other_thing
    ```

    Wow, now look at your terminal:

    ```text
    /usr/bin/ld: build/test.o:(.rodata+0x40): multiple definition of `work__other_thing__RTI'; build/test.o:(.rodata+0x40): first defined here
    /usr/bin/ld: build/test.o: in function `work__other_thing__PKG_ELAB':
    test.vhd:(.text+0x0): multiple definition of `work__other_thing__PKG_ELAB'; build/test.o:test.vhd:(.text+0x0): first defined here
    /usr/bin/ld: build/test.o: in function `work__other_thing__DECL_ELAB':
    test.vhd:(.text+0x19): multiple definition of `work__other_thing__DECL_ELAB'; build/test.o:test.vhd:(.text+0x19): first defined here
    ...
    /usr/bin/ld: e~other_thing.o: in function `work__other_thing__ARCH__structural__DEFAULT_CONFIG__something':
    e~other_thing:(.text+0x1d): undefined reference to `work__something__ARCH__structural__INSTSIZE'
    /usr/bin/ld: e~other_thing:(.text+0x44): undefined reference to `work__something__PKG_ELAB'
    /usr/bin/ld: e~other_thing:(.text+0x87): undefined reference to `work__something__ARCH__structural__DECL_ELAB'
    ...
    collect2: error: ld returned 1 exit status
    ghdl: compilation error
    ```
**Context**

- OS: Fedora Workstation 33 (Linux kernel 5.10.7)
- Origin:
  - [x] Package manager: `4.5.2` (DNF)
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

**Additional context**
- GHDL version: `1.0-dev`

tgingold, 2021-01-27:
Yes, this is a known fact.  It happens with the LLVM or GCC backend.

tgingold, 2021-03-11:
It is simply that the object file is created in the current directory using the basename of the source file.
So the first test.vhdl creates a test.o file, which is overwritten by the second test.vhdl.
During the link, the test.o file appear twice on the command line, creating duplicate symbols.

A fix could be to append a unique number to each object file.

abyszuk, 2021-03-12:
OK, understood.
Anyway, now you know you have at least 2 users waiting to fix that ;-)

jeras, 2021-04-12:
There are probably a few other build systems with the same issue (same file name different directory), which already have a solution.

Historically C compiler object files were placed in the same folder as the source file, this approach is not a good fit for GHDL.

CMAKE is closer to the GHDL approach, where all object files are placed into a build folder, and source folders are left unchanged.
I did a quick search for "where does cmake place object files from source files of same name" and found the next:

https://cmake.org/cmake/help/latest/command/file.html

See the option "CONFLICTING_DEPENDENCIES_PREFIX".

umarcor, 2021-04-12:
@jeras it seems that CONFLICTING_DEPENDENCIES_PREFIX does not actually solve the issue but provides information about which are the conflicting files, isn't it? I mean, in the case of GHDL it can be handled silently, because it can guess whether another file was first analysed.

jeras, 2021-04-12:
Sorry, I did not mean to point to a specific argument.
I just wanted to point out there are other tools with similar issues, and they might have a well thought through solution.

jpf91, 2024-12-23:
Is there any reason why we can't just simply keep the original folder structure? E.g.:
* `src/test.vhd` => `obj/mylib/src/test.o`
* `tests/test.vhd` => `obj/mylib/tests/test.o`

where `--work=mylib` and `--workdir=obj/mylib`.

This would also help writing proper Makefile wildcard rules: Commonly in make, you specify the outputs. The rule then derives the input filename. E.g. output `mylib/src/test.o` looks for `src/test.vhd`. If we only know `mylib/test.o` as output, we can't reconstruct the source path.

dalex78, 2026-08-25:
# Issues #539 and #1622 — object files collide when two sources share a base name

Analysis by an AI agent.

## Why these two are one report

Both describe the same thing: with the gcc or llvm backend, `ghdl -a` writes an object file named after the source file's *base* name into the library directory, so two design files with the same base name in different directories overwrite each other's object file. The link then puts that one object on the command line twice and fails with a wall of "multiple definition" errors about symbols the user never wrote, plus "undefined reference" errors for the units whose object was overwritten.

tgingold confirmed it on both. On #539 (2018):

> GHDL with the gcc or llvm backend creates a .o file from the vhdl file. So in your case, pkg1/pkg.vhd generates pkg.o which is then overwritten by the .o file generated by pkg2/pkg.vhd. So, you have to rename the files. Maybe I should add code to detect this situation.

and on #1622 (2021):

> It is simply that the object file is created in the current directory using the basename of the source file. So the first test.vhdl creates a test.o file, which is overwritten by the second test.vhdl. During the link, the test.o file appear twice on the command line, creating duplicate symbols.

mcode is not affected: it never writes an object file.

## What the fix does

It implements the sanity check tgingold considered ("Maybe I should add code to detect this situation" on #539, "adding a sanity check is doable" in reply to the same suggestion). `Ghdldrv.Add_File_List` is where the link list is turned into object file names, so the collision is visible there and nowhere earlier:

```
ghdl:error: 'pkg2/pkg.vhd' and 'pkg1/pkg.vhd' are both compiled to 'pkg.o'
ghdl:error: (object files are named after the source file base name; rename one of these files)
```

The check keeps the object name and source name of each design file already added, and stops with that message when a later design file maps to an object name already produced by a *different* source. The same source listed twice is not a collision and is left alone.

## What the fix deliberately does not do

It does not change how object files are named. The thread has three proposals — a hash of the full path (#539), a unique number per object (tgingold on #1622), and mirroring the source directory structure under the object directory (#1622, 2024) — and each changes where users' object files land, so which one to adopt is the maintainer's call. Until then, a design that hits this gets told what is wrong in one line instead of being handed linker errors about `work__pkg2__ELAB_SPEC`.

The two issues are therefore reported as detected, not as resolved; they stay open for the naming decision.

## Testing

- `testsuite/gna/issue539/` — two files both called `pkg.vhd`, in `pkg1/` and `pkg2/`, linked with `ghdl -m top`.
- `testsuite/gna/issue1622/` — two files both called `test.vhd`, in `src/` and `tests/`, one instantiating the other.

Both assert the new diagnostic and assert that the link is never reached (no "multiple definition" in the output). Both skip on mcode, which produces no object files.

dalex78, 2026-08-25:
@tgingold I can try to see if a fix can be provided. Do you have a preference for the fix method ? Or do I just try ?

closed 2026-08-28 by tgingold