ghdl/ghdl #1624: The conversion of a product of incompatible types leads to a crash

ldbo, 2021-01-28:
**Description**

A file that tries to convert the product of a ``time`` ratio and a ``real`` value to a ``real`` makes GHDL crash on analysis.

**Expected behaviour**

Instead of complaining about the ``*`` operator not being implemented, GHDL crashes.

**How to reproduce?**

```vhd :file: buggy.vhd
entity buggy is
begin
end entity buggy;

architecture a of buggy is
begin

p: process
    variable v: real;
begin
    -- Remove the next line and GHDL does not crash
    v := real((now / (1 ns)) * 1.0e-9);
end process;

end architecture a;
```

```sh :image: ghdl/ghdl:*
ghdl -a buggy.vhd
```

The output of the previous command is then

```
translate_predefined_operator(2): cannot handle IIR_PREDEFINED_UNIVERSAL_I_R_MUL

******************** GHDL Bug occurred ***************************
...
******************************************************************
ghdl:error: compilation error
```

**Context**

- OS: Arch 64-bit
- Origin:
  - Package: ``community/ghdl-llvm``
  - Version: ``0.37-10``

**GHDL Bug occured output**

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (tarball) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-pc-linux-gnu
/tmp/
Command line:
/usr/bin/ghdl1-llvm -P/usr/lib/ghdl/ieee/v93/ -P/usr/lib/ghdl/ -c -fpic -o buggy.o buggy.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:39
Call stack traceback locations:
0x55e45bbd7c81 0x55e45bce56a9 0x55e45bcedcba 0x55e45bcedf06 0x55e45bce5bbc 0x55e45bd4dbe3 0x55e45bd650f6 0x55e45bd6520f 0x55e45bd69b4a 0x55e45bd6f351 0x55e45bd0829f 0x55e45bcfad77 0x55e45bd7b23a 0x55e45bd7cbfb 0x55e45bd7e5c6 0x55e45bb254f0 0x7eff93000150 0x55e45bb2464c 0xfffffffffffffffe
******************************************************************
```

tgingold, 2021-01-28:
Apparently you are the first one to use that!

closed 2021-01-29 by tgingold

ldbo, 2021-02-01:
> Apparently you are the first one to use that!

Thank you serendipity !

commit e77a1e10 (2021-01-29): trans-chap7.adb: handle universal integer/real multiplication.  Fix #1624