ghdl/ghdl #1625: Exception TYPES.INTERNAL_ERROR raised

MJoergen, 2021-02-02:
**Description**
GHDL crashes when trying to run a simulation.

**Expected behaviour**
GHDL should not crash. In case of error, GHDL should report the problem, rather than crashing.

**How to reproduce?**
The following code reproduces the crash. As it stands, the code on itself is meaningsless, i.e. it was cut out from a larger project. Enough of the code was preserved in order to reproduce the crash.

```vhd :file: level0.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;

entity level0 is
end entity level0;

architecture synthesis of level0 is

   signal clk       : std_logic;
   signal rst       : std_logic;
   signal src_reg   : std_logic_vector(3 downto 0);
   signal src_val   : std_logic_vector(15 downto 0);
   signal dst_reg   : std_logic_vector(3 downto 0);
   signal dst_val   : std_logic_vector(15 downto 0);
   signal flags_out : std_logic_vector(15 downto 0);
   signal flags_we  : std_logic;
   signal flags_in  : std_logic_vector(15 downto 0); 
   signal reg_we    : std_logic;
   signal reg_addr  : std_logic_vector(3 downto 0);
   signal reg_val   : std_logic_vector(15 downto 0);

begin

   inst : entity work.level1
      port map (
         clk_i       => clk,
         rst_i       => rst,
         src_reg_i   => src_reg,
         src_val_o   => src_val,
         dst_reg_i   => dst_reg,
         dst_val_o   => dst_val,
         flags_o     => flags_out,
         flags_we_i  => flags_we,
         flags_i     => flags_in,
         reg_we_i    => reg_we,
         reg_addr_i  => reg_addr,
         reg_val_i   => reg_val
      );

end architecture synthesis;
```

```vhd :file: level1.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;

entity level1 is
   port (
      clk_i      : in  std_logic;
      rst_i      : in  std_logic;
      src_reg_i  : in  std_logic_vector(3 downto 0);
      src_val_o  : out std_logic_vector(15 downto 0);
      dst_reg_i  : in  std_logic_vector(3 downto 0);
      dst_val_o  : out std_logic_vector(15 downto 0);
      flags_o    : out std_logic_vector(15 downto 0);
      flags_we_i : in  std_logic;
      flags_i    : in  std_logic_vector(15 downto 0);
      reg_we_i   : in  std_logic;
      reg_addr_i : in  std_logic_vector(3 downto 0);
      reg_val_i  : in  std_logic_vector(15 downto 0)
   );
end entity level1;

architecture synthesis of level1 is

   signal sr : std_logic_vector(15 downto 0);
   signal dst_val_lower : std_logic_vector(15 downto 0);

begin

   inst : entity work.level2
      generic map (
         G_ADDR_SIZE => 11,
         G_DATA_SIZE => 16
      )
      port map (
         clk_i     => clk_i,
         rst_i     => rst_i,
         rd_addr_i => sr(15 downto 8) & dst_reg_i(2 downto 0),
         rd_data_o => dst_val_lower,
         wr_addr_i => sr(15 downto 8) & reg_addr_i(2 downto 0),
         wr_data_i => reg_val_i,
         wr_en_i   => reg_we_i and not reg_addr_i(3)
      ); -- i_ram_lower_dst

end architecture synthesis;
```

```vhd :file: level2.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std_unsigned.all;

entity level2 is
   generic (
      G_ADDR_SIZE : integer;
      G_DATA_SIZE : integer
   );
   port (
      clk_i         : in  std_logic;
      rst_i         : in  std_logic;
      -- Read interface
      rd_addr_i     : in  std_logic_vector(G_ADDR_SIZE-1 downto 0);
      rd_data_o     : out std_logic_vector(G_DATA_SIZE-1 downto 0);
      -- Write interface
      wr_addr_i     : in  std_logic_vector(G_ADDR_SIZE-1 downto 0);
      wr_data_i     : in  std_logic_vector(G_DATA_SIZE-1 downto 0);
      wr_en_i       : in  std_logic
   );
end entity level2;

architecture synthesis of level2 is

begin

end architecture synthesis;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -i --std=08 level0.vhd level1.vhd level2.vhd
ghdl -m --std=08 level0
ghdl -r --std=08 level0
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

> NOTE: Large files can be uploaded one-by-one or in a tarball/zipfile.

**Context**
Please, provide the following information:

- OS: Linux Mint 20 Ulyana
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [*] Built from sources: `commit b459b854` (Date:   Mon Feb 1 19:14:14 2021 +0100)

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0rc4 (1.0.0rc4.r0.gb459b854) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/mike/git/MJoergen/formal/
Command line:
ghdl -r --std=08 level0
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x55d431653c0b 0x55d431653feb 0x55d431654267 0x55d431865970 0x55d431874751 0x55d431874809 0x55d43185e57e 0x55d43183389d 0x55d431833b55 0x55d431834051 0x55d4318344ab 0x55d431835f7a 0x55d431838eb1 0x55d431840a9e 0x55d431840654 0x55d43180b122 0x55d43180b24d 0x55d43180b8a6 0x55d43180c03c 0x55d43180dbd6 0x55d43180dd2b 0x55d431811147 0x55d431815a2a 0x55d43188396a 0x55d4317eba7b 0x55d4317e8bce 0x55d4317f6cd2 0x55d4318882d9 0x55d4317c74ed 0x55d4316fff8b 0x55d43188bbf6 0x55d4314b6a67 0x7fb5a08fa0b1 0x55d4314b51cc 0xfffffffffffffffe
******************************************************************
```

This bug was present in an earlier version (commit 36ae5c1f, Date:   Thu Dec 31 22:28:43 2020 +0100), so I upgraded to the latest to see if the bug was still present, which it apparently is.

eine, 2021-02-02:
Can confirm the bug on MSYS2 (LLVM):

```
# ghdl -m --std=08 level0
analyze level2.vhd
analyze level1.vhd

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0rc3 (1.0.0rc1.r2.g4dc37524) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: x86_64-w64-mingw32
T:\
Command line:
C:\msys64\mingw64\bin\ghdl1-llvm.exe --std=08 -PC:\msys64\mingw64\lib/ghdl\ieee\v08\ -PC:\msys64\mingw64\lib/ghdl\ -c -o level1.o level1.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:553
Call stack traceback locations:
0x495d84 0x4961e5 0x496483 0x57ba7a 0x58ae62 0x58af26 0x597087 0x5b997c 0x5b9d1c 0x5ba2af 0x5ba799 0x5bcac8 0x5c0d5e 0x5cbee1 0x5cb909 0x61f75a 0x61f8aa 0x620042 0x620ac2 0x622f11 0x623049 0x626e34 0x62c211 0x5e6306 0x5d87ea 0x637b58 0x6394fa 0x63add5 0x402706 0x4013bf 0x4014f4 
0x7ffec3af7032 0x7ffec553d0cf
******************************************************************
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

MJoergen, 2021-02-02:
I found a workaround:

It seems the problem is related to the lines:
```
   rd_addr_i => sr(15 downto 8) & dst_reg_i(2 downto 0),
   wr_addr_i => sr(15 downto 8) & reg_addr_i(2 downto 0),
```

If I instead use temporary signals like so:
```
   rd_addr <= sr(15 downto 8) & dst_reg_i(2 downto 0);
   wr_addr <= sr(15 downto 8) & reg_addr_i(2 downto 0);

...

   rd_addr_i => rd_addr,
   wr_addr_i => wr_addr,
```
then GHDL no longer crashes, and the simulation (in the original project) proceeds as expected.

GeraldCarda, 2021-02-04:
I also run into a TYPES.INTERNAL_ERROR exception. This is a older tarball version of the nighly build available as download.

OS is Ubuntu 18.04LTS 64 Bit - lxd container with ghdl from tarball extract.
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1334-g2bb2e298) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/mnt/development/logic/Vivado/axiemc/
Command line:
/bin/ghdl1-llvm --workdir=work --std=08 -fsynopsys -frelaxed -Wno-hide -Wno-shared -P../../UVVM_Light/GHDL/util/ -P../../unisims/work/ -P./work/ -P/lib/ghdl/ieee/v08/ -P/lib/ghdl/ -c -fpic -o work/tb_axiemc.o vhdl/tb_axiemc.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x4a461e 0x62787a 0x6284ec 0x62ac05 0x6291a2 0x629480 0x5e756e 0x5e798f 0x5e7908 0x5e9dfa 0x5e9eb6 0x5ed86f 0x5edbb8 0x5edd15 0x5a832f 0x5a565d 0x5aa409 0x5b0c4d 0x5b0cb7 0x5b4b3c 0x5b9d59 0x638ee5 0x58967f 0x63f858 0x6411b8 0x642822 0x4091f1 0x7f870cbf3bf5 0x4084d8 0xfffffffffffffffe
******************************************************************

I am not able to extract the 'failing' part as this is a simulation using Xilinx unisims and UVVM util and bfm's. Other simulations (existing older projects) do not show this issue. I did not managed to compile a newer version from sources yet, so I just use the tarball images.

Anyhow, the above error log my help to track the issue. I can also run some tests if needed. 

Regards,
Gerald

GeraldCarda, 2021-02-04:
Same simulation using release version 0.37 crashes with:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 0.37 (v0.37) [Dunoon edition]
Compiled with GNAT Version: 4.9.3
Target: x86_64-linux-gnu
/media/filemaster/development/logic/Vivado/axiemc/
Command line:
/bin/ghdl1-llvm --workdir=work --std=08 -frelaxed -Wno-hide -Wno-shared -P../../UVVM_Light/GHDL/util/ -P../../unisims/work/ -P./work/ -P/lib/ghdl/synopsys/v08/ -P/lib/ghdl/ -c -o work/tb_axiemc.o vhdl/tb_axiemc.vhd
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
Exception name: SYSTEM.ASSERTIONS.ASSERT_FAILURE
Message: vhdl-utils.adb:1135
Call stack traceback locations:
0x7f416494fa81 0x4c7fd0 0x4c7ffc 0x5a93a3 0x5b6f92 0x5791f6 0x5795a1 0x5575d0 0x5543e5 0x5599de 0x561b09 0x561b70 0x536d4c 0x53c74c 0x5d2dcc 0x514ff0 0x5e1a7b 0x5e331e 0x41be84 0x408a01 0x7f4163dd683e 0x407ca7 0xfffffffffffffffe
******************************************************************

Gerald

eine, 2021-02-04:
> I am not able to extract the 'failing' part as this is a simulation using Xilinx unisims and UVVM util and bfm's. Other simulations (existing older projects) do not show this issue. I did not managed to compile a newer version from sources yet, so I just use the tarball images.

You can use the nightly assets: https://github.com/ghdl/ghdl/releases. However, I think that your issue is unrelated to the OP. Might be better to handle it in a different issue. Moreover, UVVM is known to fail: https://github.com/ghdl/ghdl/issues?q=is%3Aissue+is%3Aopen+sort%3Aupdated-desc+label%3A%22ThirdParty%3A+UVVM%22

GeraldCarda, 2021-02-04:
I will try the nightly assets. Other than that I could compile the UVVM without any problem. I was also able to run the UVVM demo_tb example and some axilite testbenches I created on my own (using the UVVM bfm's).
The reported crash only pops up while using the axi bus bfm (not axi lite, see above) AND using the axi_write function supplied by this bfm.
I will now setup a build system to compile ghdl from the sources. This may help to further investigate the crash cause.

Gerald

diogratia, 2021-02-04:
See IEEE Std 1075-2008 6.5.6.3 Port clauses. ghdl as yet doesn't support the declaration of anonymous signals for expressions as actuals in a port association, a feature introduced in -2008.

If you were to comment out the use clauses for package numeric_std_unsigned and attempt to analyze the various design unit's you'd get error messages `actual expression must be globally static`. 

What's missing with --std=08 absent implementation of the feature is detecting the actual expressions aren't static.

commit a20bce35 (2021-08-23): Rework inertial associations.

Fix #1625
Fix #1672
closed 2021-08-23 by tgingold