ghdl/ghdl #1629: Please respect the CC environment variable for C compiler executable name

yurivict, 2021-02-04:
**Description**
On different systems the compiler executable has different names. On FreeBSD it is ```cc```, which is passed to build through the ```CC``` environment variable which is a common practice.

Currently we have to patch the sources to replace the ```gcc``` executable name hardcoded in the sources.

Particularly, this patch is applied:
```
--- src/ghdldrv/ghdldrv.adb.orig	2021-02-02 19:52:50 UTC
+++ src/ghdldrv/ghdldrv.adb
@@ -483,7 +483,7 @@ package body Ghdldrv is
          Cmd.Assembler_Cmd := new String'("as");
       end if;
       if Cmd.Linker_Cmd = null then
-         Cmd.Linker_Cmd := new String'("gcc");
+         Cmd.Linker_Cmd := new String'("cc");
       end if;
    end Set_Tools_Name;
 
@@ -755,8 +755,8 @@ package body Ghdldrv is
       Put_Line ("   Set the path of the ghdl1 compiler");
       Put_Line (" --AS=as");
       Put_Line ("   Use as for the assembler");
-      Put_Line (" --LINK=gcc");
-      Put_Line ("   Use gcc for the linker driver");
+      Put_Line (" --LINK=cc");
+      Put_Line ("   Use cc for the linker driver");
       Put_Line (" -S");
       Put_Line ("   Do not assemble");
       Put_Line (" -o FILE");
```

It would be nicer if the ```configure``` step could pick the value of the ```CC``` environment variable and substitute its value in these places.

Thank you,
Yuri

eine, 2021-02-04:
It feels to me that using the value in `--LINK=cc` might be more sensible than using the envvar, for this specific case. What do you think?

yurivict, 2021-02-04:
Linking of C code should also be done with $(CC).

eine, 2021-02-04:
According to https://github.com/ghdl/ghdl/blob/master/src/ghdldrv/ghdldrv.adb#L637 the value of `--LINK` is used indeed. So, the modification you are suggesting is to allow changing the default, but GHDL can already be used without patching, isn't it?

yurivict, 2021-02-04:
> {...} GHDL can already be used without patching, isn't it?

Not on systems that don't have ```gcc```.

eine, 2021-02-04:
Then, option `--LINK` does not work for setting/overriding the `Linker_Cmd` default?

yurivict, 2021-02-04:
Its default should be based on $(CC) or $(LD) but then the user would be able to set it to to any other value.

tgingold, 2021-02-04:
Maybe the simplest is to use 'cc' as default linker.  That should work on most systems.

closed 2021-08-01 by tgingold

umarcor, 2021-09-23:
It seems that some jobs in extended-tests are failing since https://github.com/ghdl/ghdl/commit/b72f46a0 was merged. Precisely, when GHDL with LLVM backend is installed on MSYS2 and used from PowerShell (through the vendor scripts), it cannot find `cc`:

https://github.com/ghdl/extended-tests/runs/3681282727?check_suite_focus=true#step:6:34
```
D:\a\_temp\msys\msys64\MINGW64\bin\ghdl.exe: installation problem: cc not found
```

I tried adding `--LINK=gcc` to the vendor scripts (https://github.com/umarcor/ghdl/commit/50211118930484d87fa939fcb2701013b5ba8da5). That fixes the problem with LLVM backend, but it break the jobs with mcode backend, because the option is only available for LLVM/GCC: 

https://github.com/umarcor/ghdl-extended-tests/runs/3682032108?check_suite_focus=true#step:7:34
```
D:\a\_temp\msys\msys64\MINGW32\bin\ghdl.exe: unknown command option '--LINK=gcc'
```

How shall we proceed? Should we detect the backend in the vendor scripts, or should we handle it in the CI of extended-tests by creating symlink for cc to exist?

/cc @Paebbels

tgingold, 2021-09-23:
We can create a symlink in the docker image.  Easy.

We could also add a configuration option to set the name of the C compiler.

Or GHDL could try first cc, then gcc, then clang.

umarcor, 2021-09-23:
> We can create a symlink in the docker image.  Easy.

The failing tests are on Windows, with GHDL installed through MSYS2 (setup-ghdl-ci) and executed from PowerShell. Hence, I'm not sure about the symlink option being so straightforward, although it might be possible.

> We could also add a configuration option to set the name of the C compiler.

I think it'd be desirable if this was overriden at runtime. The same GHDL build does work on MSYS2 without issues.
Suprisingly, https://github.com/ghdl/ghdl/runs/3681890432?check_suite_focus=true works on this repo. Maybe that's because we are testing pyGHDL/pyunit only?

> Or GHDL could try first cc, then gcc, then clang.

What about using envvar CC, as suggested in this issue originally?

commit b72f46a0 (2021-08-01): ghdldrv.adb: use cc (instead of gcc) as linker driver.  Fix #1629