ghdl/ghdl #1632: GHDL fuzzing

Arcnor, 2021-02-05:
So after I got & reported #1631 I was curious to see if fuzzing might find any useful errors on the project. (For those that are not familiar, fuzzing is basically running totally/partially random inputs to any piece of software to find bugs on it).

I found at least 3 errors using this technique, and I was wondering if such an approach is interesting to this project. My idea is either fuzz & report each bug that I find and/or setting some CI job that takes care of finding them.

So the question(s) are:
1. Will bugs reported using this technique be accepted? This might seem an odd question, but a bug found doing this might be "GHDL crashes when completely random bytes are passed as input" although that's an extreme case and not a real one either. Also, only "crash" bugs will be reported (the ones that contain "GHDL Bug") given that's probably impossible to differentiate others.
2. Is there any interest in having automation for this kind of thing?

I was also thinking that at least for the first stage (manual reporting) a new tag like `fuzz` can be created so bugs reported like this can be filtered out if that's desired.

Anyway, I hope this is useful, please let me know what you think.

tgingold, 2021-02-05:
Why not.  I tried a fuzzer a few years ago, but I wasn't convinced by the outcome.  Maybe it is a matter of configuration or tooling.

I am ok with such bug reports, but that's not very useful if they are very artificial.  Also, don't mass report.

Arcnor, 2021-02-05:
Ok, I'll report the few I've found, then (around 7 I think). Not sure what you mean by "mass report", though. You mean automatically creating many issues?

I'll make sure they all refer to a different issue and that they are not "artificial".

eine, 2021-02-05:
I propose using ghdl/extended-tests for this purpose. We can have a CI job which generates reports either as artifacts or as a pre-release. Then, we don't need to handle them as regular issues, but we can have a list of crashes to check/fix. Ideally, the reports would be MWEs in multiple tarballs, so that Tristan can just download a tarball and run it for reproducing the crash.

Arcnor, 2021-02-05:
Ok, then I'll hold off reporting for now until that can be done? I'm not sure where to start, honestly. Also the reports I currently have are not that "crazy" anyway (things like overflow errors and similar) but I'll wait for any of you to decide how to proceed.

eine, 2021-02-06:
> I'm not sure where to start, honestly.

Copy the following job in any fork/repo of yours:

```yml
  test:
    runs-on: ubuntu-latest
    steps:

    - uses: actions/checkout@v2

    - uses: ghdl/setup-ghdl-ci@master
      with:
        backend: llvm

    - run: ./yourscript.sh
```

Write your fuzzing and MWE generation logic in `yourscript.sh`. Make it find/generate the reproducible crashes in some specific subdir.

When that's done, I can help with moving it to extended-tests.

tgingold, 2021-02-06:
You can submit one at a time.  My only fear is that they aren't representative.
(This was not the case of 1631 - it was a 'good' issue)

Arcnor, 2021-02-10:
@eine thanks for the detailed writeup. I'm a bit constrained on time at the moment, so I'll open them one by one for now until I can automate it
@tgingold I think they are, it's not garbage what they're processing (in most cases at least) but I'll let you be the judge of that. I'll open 2 today, then I'll wait for you to fix them before adding more.

closed 2021-08-01 by tgingold