ghdl/ghdl #1634: VHDL errors cause ghdl-mcode to crash on Windows

SiliconWizard, 2021-02-06:
**Description**
Any error in VHDL code analyzed by ghdl (mcode version) causes it to crash on Windows after displaying the error message.
Just before crashing, the following additional message is displayed on the console:

"This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information."

**Expected behaviour**
Expecting to see an error message, but not the above message, and no crash.

**How to reproduce?**
Tell us how to reproduce this issue. Please provide a Minimal Working Example (MWE). With sample code it's easier to reproduce the bug and it's much faster to fix it. For example:

```vhd :file: ErrorBug.vhd
entity ErrorBug is
end entity;

architecture Behavioral of ErrorBug1 is
end Behavioral;
```

```
ghdl -a ErrorBug.vhd
```

Yields the following output on console:

```
ErrorBug.vhd:5:1: 'begin' is expected instead of 'end'

This application has requested the Runtime to terminate it in an unusual way.
Please contact the application's support team for more information.
```

Followed by an application "crash".

**Context**
Please, provide the following information:

- OS: Windows 7/64
- Origin: Built from the latest git source, with latest MSYS2/mingw32

**Additional context**
When no error is issued by ghdl-mcode, execution works as normal.

The above crash behavior seems to be triggered by a call to the abort() C standard function. On Windows, unless specific flags are set with the _set_abort_behavior() function, abort() will cause the output of the "This application has requested the Runtime to terminate it in an unusual way." message, and will then cause a apparent "crash".

I'm not sure 1/ using abort() is a good way of ending application termination here and 2/ why ghdl executes this when it detects errors in source files.

The released pre-built 0.37 version doesn't seem to exhibit this behavior.

eine, 2021-02-06:
I cannot reproduce with https://github.com/ghdl/ghdl/releases/download/nightly/mingw-w64-i686-ghdl-mcode-ci-1-any.pkg.tar.zst:

```
# ghdl -a ErrorBug.vhd
ErrorBug.vhd:5:1: 'begin' is expected instead of 'end'

# ghdl --version
GHDL 2.0.0-dev (1.0.0.r10.g982f1d7c) [Dunoon edition]
 Compiled with GNAT Version: 10.2.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

Did you use the PKGBUILD recipe (https://github.com/ghdl/ghdl/blob/master/scripts/msys2-mcode/PKGBUILD) for building GHDL?

SiliconWizard, 2021-02-06:
I was building it "manually". Thanks for pointing me to this, I figured what caused my issue. It's indeed all related to the call of the abort() function. The difference with what I did is that the PKGBUILD recipe configures the build with the "LDFLAGS=-static" option, whereas I wasn't doing that. So I think my build was using the abort() function from the standard Windows C library, whereas with a static build, it's using its own mingw32 implementation. With this "static" option, the crash disappears.

Thanks for the quick heads up. Worth noting this abort() issue on Windows.

closed 2021-02-06 by SiliconWizard

eine, 2021-02-07:
@SiliconWizard, precisely I pointed you to the recipe because of those configure/make options. Other than that, it's ok to build manually (without the recipe).

> Worth noting this abort() issue on Windows.

Development on Windows does need some love. We do provide two very specific builds (mcode on MINGW32 and LLVM on MINGW64), which we know to be stable and usable inside MSYS2. However, there are many details and other build options which were not thoroughly tested/debugged yet. For instance, https://github.com/ghdl/ghdl/tree/master/scripts/windows contains build scripts for powershell using AdaCore's GNAT. See #1599.