ghdl/ghdl #1636: ASSERT_FAILURE on synthesis of memory with asymmetric access

egorxe, 2021-02-08:
**Description**
ghdl -- synth crashes when trying to synthesize memory with different read and write width.

**Expected behaviour**

Not to crash. Maybe produce a warning that bit is not written?

**How to reproduce?**

```vhd :file: asym_test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity asym_test is
  port (
    clk_i  : in  std_logic; 
    dat_i  : in  std_logic_vector(15 downto 0);
    addr_i : in  std_logic_vector(3 downto 0); 
    dat_o : out std_logic_vector(15 downto 0)
  );
end asym_test;

architecture arch of asym_test is

  type   reg_file_t is array (15 downto 0) of std_logic_vector(15 downto 0);
  signal reg_file : reg_file_t;

begin

  process(clk_i)
  begin
    if rising_edge(clk_i) then
        reg_file(to_integer(unsigned(addr_i)))(15 downto 1) <= dat_i(15 downto 1);
    end if;
  end process;

  process(addr_i, reg_file)
  begin
      dat_o <= reg_file(to_integer(unsigned(addr_i)));
  end process;

end arch;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a asym_test.vhd
ghdl --synth asym_test
```

**Context**
Please, provide the following information:

- OS: Debian 10
- Origin:
  - Source from `https://github.com/ghdl/ghdl/releases/tag/v1.0.0`

```
asym_test.vhd:36:10:note: found RAM "reg_file", width: 16 bits, depth: 16

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0.0 (tarball) [Dunoon edition]
Compiled with GNAT Version: 8.3.0
Target: x86_64-linux-gnu
/home/test/
Command line:
ghdl --synth asym_test
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : netlists-builders.adb:1266
Call stack traceback locations:
0x7febe716d4f1 0x55f8bba727ea 0x55f8bbb02698 0x55f8bbb029fb 0x55f8bbb043bc 0x55f8bbb04f9a 0x55f8bbbfb238 0x55f8bbbfb2c6 0x55f8bbbf4358 0x55f8bbc61751 0x55f8bbb839f6 0x55f8bbd10d29 0x55f8bb93ca2b 0x7febe6d55099 0x55f8bb93b1a8 0xfffffffffffffffe
******************************************************************

```

tgingold, 2021-02-11:
I suppose the workaround is to write all the bit.

dalex78, 2026-08-20:
# Issue #1636 — Crash: memory with asymmetric (partial) write

Analysis by an AI agent.

## High-level summary

A memory (`reg_file`, array of `std_logic_vector`) written only in bits `15 downto 1` (bit 0 never written) crashed `ghdl --synth` with an internal `ASSERT_FAILURE` at `netlists-builders.adb:1266` instead of handling the partial write.

## Investigation / current status: already fixed on master

Reproduced the exact repro from the report against current `master`: `ghdl --synth asym_test` now produces a correct netlist, splitting the memory into two separate inferred memories along the write-enable boundary (a 1-bit `mem_type` for the never-written bit 0, and a 15-bit `mem_type` for bits `14 downto 0`) instead of crashing. This must have been fixed as part of general improvements to partial-write memory inference in `src/synth` since this was reported (originally against GHDL 1.0.0, 2020). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1636/` (the exact repro from the report) whose `testsuite.sh` synthesizes the design and re-analyzes the resulting netlist. This locks in the current, correct behavior as a regression guard.

dalex78, 2026-08-24:
@tgingold should this issue be closed ? (PR has been merged)

closed 2026-08-24 by tgingold