ghdl/ghdl #1638: Semantics of 'out' mode procedure parameter not correct for record type arguments

sidneycadot, 2021-02-09:
**Description**
GHDL doesn't seem to properly implement the proper initialization behavior for record-type variables of mode 'out' passed to a procedure. Inside the procedure, the initial value should be the default record value, rather than the value of the actual parameter.

**Expected behaviour**
I expect to see -2147483648 as the value of rec.field printed inside the procedure (and also as the value after execution of the procedure).

**How to reproduce?**
Analyze and elaborate using ghdl.
I did this in --std=08 mode; no idea about other modes.

```vhd :file: ent.vhd
entity bug_proc_inout is
end entity bug_proc_inout;

architecture arch of bug_proc_inout is

type record_type is record
        field : integer;
    end record record_type;

procedure show_bad_behavior(rec: out record_type) is
begin
    -- The variable 'rec' with out mode should be initialized to its default value,
    -- rather than the value of the actual parameter.
    -- The reported value is the value
    report "inside procedure call: " & to_string(rec.field);
end procedure show_bad_behavior;

begin

    process is
    variable rec : record_type;
    begin
        rec.field := 42;
        report "before procedure call: " & to_string(rec.field);
        show_bad_behavior(rec);
        report "after procedure call: " & to_string(rec.field);
        wait;
    end process;

end architecture arch;
```

I used a built-from-source GHDL under Linux:
```
GHDL 2.0.0-dev (1.0.0.r11.ga75c135b) [Dunoon edition]
```

umarcor, 2021-02-09:
@JimLewis, for your information.

diogratia, 2021-02-10:
IEEE Std 1076-2008 10.7 Procedure call statement

>Execution of a procedure call includes evaluation of the actual parameter expressions specified in the call and  evaluation  of  the  default  expressions  associated  with  formal  parameters  of  the  procedure  that  do  not  have actual parameters associated with them. In both cases, the resulting value shall belong to the subtype of the  associated  formal  parameter. 

These two quoted sentences are mode-agnostic and are also found in -1993.

The formal parameter has an actual parameter associated with it. The default expression isn't evaluated while the actual parameter expression is.

Your premise doesn't appear valid.

sidneycadot, 2021-02-11:
I think it's best to consult the part of the standard that actually deals with the semantics of parameter modes.

Section 4.2.2.2 of IEEE Std 1076-2008, and, more specifically, Note 2, says:

"NOTE 2—The value of an actual associated with a formal variable parameter of mode **out** is not copied into the formal parameter. Rather, the formal parameter is initialized based on its declared type, regardless of whether the implementation chooses to pass the parameter by copy or by reference. When a formal variable parameter of mode out is read, the current value of the formal parameter is read."

Ashenden's Designer's Guide To VHDL (3rd edition) is a bit more explicit still (Section 6.2):

"[...] The difference is that an **out**-mode parameter is not initialized with the value of the actual parameter. Instead, it is initalized in the same way as a locally declared variable, with the default initial value for the type of the parameter."

tgingold, 2021-02-11:
I agree with you.  This point was clarified with LRM 2008.

sidneycadot, 2021-02-11:
Note that I tried to simplify the minimal bug example with an integer instead of a record type. For integers, GHDL seems to follow the specification.

diogratia, 2021-02-12:
Don't rely on notes.

>1.3.4 Front matter, examples, notes, references, and annexes  
>  
... Notes are meant to emphasize consequences of the rules described in the clause or elsewhere.  ... Examples, notes, and cross-references are not part of the definition of the language.

Generally elsewhere. Notes are just as hard to find the semantics they allude to as Peter Ashenden's book.

However as Tristan mentions -2008 clarifies:

>14.6 Dynamic elaboration  
>  
>The execution of certain constructs that involve sequential statements rather than concurrent statements also involves elaboration. Such elaboration occurs during the execution of the model.
There are three particular instances in which elaboration occurs dynamically during simulation. These are as follows:  
>  
>> ...  
>>b) Execution of a subprogram call involves the elaboration of the parameter association list. This involves the elaboration of the parameter association element or elements in the association list associated with each interface declaration. Elaboration of the parameter association elements associated with a formal parameter declaration proceeds as follows:  
>>  
>>> 1) The subtype indication of the corresponding formal parameter declaration is elaborated.  
>>> 2) The formal part or parts of the parameter association elements corresponding to the formal parameter declaration are elaborated.  
>>> 3) If the type of the formal parameter is an array type or contains a subelement that is of an array type, the rules of 5.3.2.2 are applied to determine the index ranges.  
>>> 4) For each parameter association element associated with the formal parameter declaration, the parameter or subelement or slice thereof designated by the formal part is then associated with the actual part.  
>>> 5) If the formal parameter is a variable of mode out, then the implicit initial value for the object is determined.

Where b) 5) specifies the implicit initial value is used.

Without clarification in -1993 12.5 Dynamic elaboration you'd expect the semantics of  8.6 Procedure call statements  to hold sway and the notes in 2.1.1.1 Constant and variable parameters aren't the same. It could be more on the order of functional modification for those implementations not previously clarified by the new semantics. The difference relates to historical non-portability. On the other it's a contrived test case which conveys little information the mode out parameter lacking an assignment to insure compliance detection with 14.6.

Should the semantics of 10.7 yield to 14.6 here? Yes for a strictly compliant implementation.

sidneycadot, 2021-02-12:
@diogratia I am not sure what wou are arguing..

Do you now agree we're looking at a bug, or not?

If not, is your opinion that both Note 2 and Ashenden are simply wrong?

diogratia, 2021-02-12:
>Should the semantics of 10.7 yield to 14.6 here? Yes for a strictly compliant implementation.

Not having stated it wasn't a bug, there's semantic conflict between 10.7 and 14.6 that should be resolved in favor of the latter.

It's a bug on semantics new to -2008 and here relies on the procedure not assigning the formal to be detectable, otherwise it's a tree falling in the forest with no one to hear.

Note 2 and Ashenden's 6.2 are simply not authoritative. They happen to be correct here while neither one provides the clause and subclause with the actual semantic requirement.

The -2008 change derives from -2002 Issue Report 2085 where the author was aware of two implementations not matching the present semantics of -2008 14.6. No change was recommended by ISAC yet there was a change to the later standard, the documentation trail is incomplete. See LCS-2006-046, 5.    In 12.5, page 250, change list item b), where sub items 1) - 5) are inserted anecdotally to a deleted paragraph the sub items can be inferred from a referenced -2002 subclause in IR2085.