ghdl/ghdl #1639: Overflow on 'after'

Arcnor, 2021-02-10:
**Description**
Part of the bugs found during #1632 . There is an overflow error using `after`

**How to reproduce?**

```vhd :file: ap_a_04.vhd
 entity ap_a_04 is

 end entity ap_a_04;

 library ieee;  use ieee.std_logic_1164.all;

 architecture test of ap_a_04 is

   signal a, b, y : std_ulogic;

 begin

   -- code from book

   y <= a or b;

   -- end code from book

   a <= '0', '1' after 10 ns;
   b <= '0', '1' after 9223372036854775802 ns, '0' after 10 ns, '1' after 15 ns;

 end architecture test;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ap_a_04.vhd
```

**Context**
```
******************** GHDL Bug occurred ***************************
 Please report this bug on https://github.com/ghdl/ghdl/issues
 GHDL release: 2.0.0-dev (1.0.0.r6.g8e0c7e8f) [Dunoon edition]
 Compiled with GNAT Version: Community 2019 (20190517-83)
 Target: x86_64-apple-darwin17.7.0
/tmp/
 Command line:
ghdl_mcode -a fuzz-ap_a_ap_a_04.vhd-5.vhd
 Exception CONSTRAINT_ERROR raised
 Exception information:
 raised CONSTRAINT_ERROR : vhdl-evaluation.adb:60 overflow check failed
 Load address: 0x10ccc9000
 Call stack traceback locations:
 0x10cefc86b 0x10cf9e519 0x10cf9ee92 0x10cfa308a 0x10cfa33b8 0x10cfa350d 0x10cf5c569 0x10cf63e64 0x10cf48479 0x10d00c2d2 0x10cf45846 0x10d0cd5f1 0x10cccbe96
 ******************************************************************
```

diogratia, 2021-02-10:
TIME'HIGH is 9223372036854775807 fs for ghdl (ghdl  --disp-standard).

The error occurs in evaluating the physical literal 9223372036854775802 ns where the multiplication by the secondary unit causes overflows.

While you could get past this particular error with
```none
ghdl -a --time-resolution=ns ap_a_04.vhdl
```
thanks to the use of the mcode version the resolvable ghdl error in the example as shown is a lack of exception handling for the overflow.

There are two further semantic errors in the waveform being assigned to b. Both 10 ns and 15 ns are not greater that 9223372036854775802 ns in the last two waveform elements.

All three errors are defined in the same place. IEEE Std 1076-2008 10.5.2.2 Executing a simple assignment statement "It is an error if the sequence of new transactions is not in ascending order with respect to time. It is also an error if the value of any value expression in the waveform does not belong to the subtype of the target."

Arcnor, 2021-02-10:
Hey @diogratia , thanks for the detailed explanation!

This issue is actually about the exceptions when parsing certain code, which GHDL should parse without exploding (IMHO) and it's part of a bigger effort on my part to make GHDL a bit more robust (you can see the linked issue for discussion if you're interested).

commit 899eab63 (2021-02-10): vhdl-sem_stmts: avoid overflow. Fix #1639
closed 2021-02-10 by tgingold