ghdl/ghdl #1640: Cannot handle IIR_KIND_OVERFLOW_LITERAL

Arcnor, 2021-02-10:
**Description**
Part of #1632 . Cannot handle IIR_KIND_OVERFLOW_LITERAL

**How to reproduce?**

```vhd :file: test.vhd
package cpu_types is
   constant word_size : positive := 2;

   subtype word is bit_vector(word_size downto 0);
 end package cpu_types;

 package fg_08_01 is

 end package fg_08_01;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a test.vhd
```
**Context**
```
eval_pos: cannot handle IIR_KIND_OVERFLOW_LITERAL (/tmp/test.vhd:6:40)

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r6.g8e0c7e8f) [Dunoon edition]
Compiled with GNAT Version: Community 2019 (20190517-83)
Target: x86_64-apple-darwin17.7.0
/tmp/
Command line:
ghdl_mcode -a /tmp/test.vhd
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : vhdl-errors.adb:30
Load address: 0x10443a000
Call stack traceback locations:
0x10461bafe 0x10467ab92 0x10467a715 0x10467a8d4 0x1046f06a9 0x1046f0738 0x1046f0c16 0x1046dc79a 0x1046dc890 0x1046dcbb9 0x1046ded9c 0x1046df35f 0x1046ea584 0x1046ee2b3 0x1046ee618 0x1046d3927 0x1046d4e70 0x1046b9479 0x10477d2d2 0x1046b6846 0x10483e5f1 0x10443ce96
******************************************************************
```

tgingold, 2021-02-10:
Are you sure about this one ?  I cannot get a crash and the test looks ok.

Arcnor, 2021-02-10:
My apologies, I tried to simplify it too much and I probably tested the wrong thing at some point. The whole test is:

```
package cpu_types is

  constant word_size : positive := 2147483649;
  constant address_size : positive := 24;

  subtype word is bit_vector(word_size - 1 downto 0);
  subtype address is bit_vector(address_size - 1 downto 0);

  type status_value is ( halted, idle, fetch, mem_read, mem_write,
                         io_read, io_write, int_ack );

end package cpu_types;

-- end code from book

package fg_08_01 is

  constant status :
    -- code from book
    work.cpu_types.status_value
    -- end code from book
    :=
    -- code from book
    work.cpu_types.status_value'(work.cpu_types.fetch)
    -- end code from book
    ;

end package fg_08_01;
```

eine, 2021-02-10:
I can confirm that the previous one did not crash and this one does (`1.0.0.r0.g2fb2384d`).

commit 8e832e1b (2021-02-11): eval_is_range_in_bound: avoid crash in case of overflow.  Fix #1640
closed 2021-02-11 by tgingold