ghdl/ghdl #1646: file type: index constraint not allowed

umarcor, 2021-02-12:
The following MWE shows two solutions which I'd expect to be equivalent. However, GHDL accepts only one of them.

```vhd :file: test.vhd
entity test is
end entity;

architecture arch of test is
    -- This works:
    --subtype y_vec is real_vector(0 to 3);
    --type realvec_file is file of y_vec;

    -- This complains bout 'index constraint not allowed here'
    type realvec_file is file of real_vector(0 to 3);
begin
end arch;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 test.vhd
```

```
# ghdl -a --std=08 test.vhd 
test.vhd:10:45: index constraint not allowed here
C:\msys64\mingw64\bin\ghdl.exe: compilation error
```

**Context**

```
GHDL 1.0.0 (1.0.0.r0.g2fb2384d) [Dunoon edition]
```

tgingold, 2021-02-12:
Sorry, but GHDL follows the LRM here.  I agree it's a stupid rule (and not the only one), but that's the rule.

Paebbels, 2021-02-13:
@umarcor please  report this at [IEEE-P1076/VHDL-Issues](https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues). I think we should improve it in the  next revision.

@tgingold can you narrow down, what part of the LRM causes this restriction for file types,  but not for other types?

--------------
/cc @JimLewis

tgingold, 2021-02-13:
It is in the grammar of file types (5.5).

Attribute declarations (6.7) has the same useless restriction.

umarcor, 2021-02-13:
Reported: [gl:IEEE-P1076/VHDL-Issues#43](https://gitlab.com/IEEE-P1076/VHDL-Issues/-/issues/43)

closed 2021-02-13 by tgingold