ghdl/ghdl #1648: Adding VHPI support

marlonjames, 2021-02-15:
As far as I am aware, there are no open-source VHDL simulators that have more than rudimentary VHPI support.
I think that the VHDL community would benefit from having a mature implementation of VHPI in an open-source project, and GHDL is the obvious target.

Is the GHDL project receptive to contributions towards that goal?

VHPI is divided into the different capability sets, which helpfully provide subsets of the full functionality, and can be worked in turn. My initial goal would be loading a library and calling the registration function(s), and the minimal implementation as noted in the LRM:

> A minimal implementation of the VHPI need only provide the function interface described in Clause 23 and
Annex B, with none of the capability sets described in this subclause being implemented by the tool. In such a minimal
implementation, calls to functions would, in most cases, raise a VHPI error condition.

To be followed by some of the capability sets, in whatever order seems most useful/requested, with an eye towards cocotb support (e.g., `vhpiProvidesDebugRuntime`, `vhpiProvidesHierarchy`).

A related issue: I've started working on a set of VHPI compliance tests. The plan is to have a freely available testsuite that simulators can run to test their VHPI implementation. I'm not sure how that should integrate with testing of a GHDL implementation.

tgingold, 2021-02-15:
Of course, contributions for VHPI are welcome!

There is a base to implement it: Avhpi, which mimic the VHPI interface but with much less dynamic allocation.

umarcor, 2021-02-15:
> A related issue: I've started working on a set of VHPI compliance tests. The plan is to have a freely available testsuite that simulators can run to test their VHPI implementation. I'm not sure how that should integrate with testing of a GHDL implementation.

See https://github.com/VHDL/Compliance-Tests.

umarcor, 2021-02-15:
> There is a base to implement it: Avhpi, which mimic the VHPI interface but with much less dynamic allocation.

Ref: https://ghdl.github.io/ghdl-cosim/vhpidirect/grt.html

marlonjames, 2021-03-02:
What's the preferred path for the VHPI thunk?

1. separate `libghdlvhpi`
2. combine VPI/VHPI to `libghdlpli` similar to what other sims do, and hard link from `libghdlvpi` for backwards compatibilty?

tgingold, 2021-03-02:
Yes, combine in the same library.  I am not sure we even need to rename it.