ghdl/ghdl #1649: Behavior of function "*" in openieee lib in GHDL 0.37 : bug or feature ?

jserot, 2021-02-16:
When one of its (`signed`or `unsigned`) argument is undefined, the function `"*"` provided by the `numeric_std` package of the `openieee` library in release 0.37 of `ghdl` returns `0`.

One would have expected - as happened in previous `ghdl` versions (at least the 0.35 i was using until recently) - the function to return an undefined result (`X...X`).

Examining the code of the function in `src/openieee/v93/numeric_std-body.vhd` confirms this behavior (see line 913, where the result is initialized with `0`s and not altered afterwards when an `X` is detected in an argument).

Is this a bug or (added) specification of the `openieee` lib ???

This behavior is easy to reproduce with this pair of files

```
-- file mul.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity mul is
  port (x   : in  signed(7 downto 0);
        y   : in  signed(7 downto 0);
        r   : out signed(15 downto 0));
end entity;

architecture arch of mul is
begin
      r <= x*y;
end architecture;
```
```
-- file tb.vhd
library ieee;
use ieee.std_logic_1164.all;	   
use ieee.numeric_std.all;

entity tb is
end tb;

architecture seq of tb is
  signal x, y   : signed (7 downto 0);
  signal r   : signed (15 downto 0);

begin
  
  U1: entity work.mul port map(x, y, r);

  STIM: process
  begin
    x <= to_signed(0, 8); wait for 10 ns;
    y <= to_signed(2, 8); wait for 10 ns;
    wait;
  end process;
end architecture;
```

compiled with

```
	ghdl -a --std=93 ieee_fake.vhd	
	ghdl -a --std=93 mul.vhd	
	ghdl -a --std=93 tb.vhd
	ghdl -e --std=93 tb	
	ghdl -r --std=93 tb --stop-time=30ns --vcd=tb.vcd
```
From t=0 to t=10ns, `r` should be `X` since `y` is. It is `0`.

**Context**
- OS: Linux Ubuntu (Linux Mint)
- Origin:
  - [x] Package manager: `GHDL 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]`
  

**Additional context**
This problem did not show with the `ieee/v93` version of the `numeric_std` package bundled with `ghdl` v 0.35

tgingold, 2021-02-21:
I think this is a bug.  I will fix it.

eine, 2021-02-21:
Note that recent versions of GHDL do not include openieee by default anymore. Still, the lib is in the codebase, so users can optionally install it if they want to.

diogratia, 2021-02-21:
The behavior of the IEEE package is considered semantically definitive.

The openieee [version](https://github.com/ghdl/ghdl/blob/ghdl-0.35/libraries/openieee/numeric_std-body.v93) in v0.35 exhibits the same behavior while the IEEE numeric_std package returns all 'X's.

One possible solution to match the IEEE package behavior would be to detect metavalues in the left and right operands and return (others => 'X') like the IEEE version. This could also interject the added warning on detecting non-logical values only once.

Something like:

```vhdl
    function "*" (L, R : SIGNED) return SIGNED
    is
       alias la : SIGNED (L'Length - 1 downto 0) is l;
       alias ra : SIGNED (R'Length - 1 downto 0) is r;
       variable res : SIGNED (L'length + R'Length -1 downto 0) := (others => '0');
       variable rb, lb, vb, carry : sl_x01;
    begin
       if la'length = 0 or ra'length = 0 then
          return null_SIGNED;
       end if;
       if is_X(std_logic_vector(la)) or is_X(std_logic_vector(ra)) then
            res := (others => 'X');
            assert NO_WARNING
                report "NUMERIC_STD.""*"": non logical value detected"
                severity warning;
            return res;
        end if;
       --  Shift and add L.
       for i in natural range 0 to ra'left - 1 loop
         rb := sl_to_x01 (ra (i));
         if rb = '1' then
            --  Compute res := res + shift_left (l, i).
            carry := '0';
            for j in la'reverse_range loop
              lb := la (j);
              vb := res (i + j);
              res (i + j) := compute_sum (carry, vb, lb);
              carry := compute_carry (carry, vb, lb);
            end loop;
            --  Sign extend and propagate carry.
            lb := la (la'left);
            for j in i + l'length to res'left loop
               vb := res (j);
               res (j) := compute_sum (carry, vb, lb);
               carry := compute_carry (carry, vb, lb);
            end loop;
         -- elsif rb = 'X' then
         --  assert NO_WARNING
         --    report "NUMERIC_STD.""*"": non logical value detected"
         --    severity warning;
         end if;
       end loop;
       if ra (ra'left) = '1' then
          --  R is a negative number.  It is considered as:
          --   -2**n + (Rn-1 Rn-2 ... R0).
          --  Compute res := res - 2**n * l.
          carry := '1';
          for i in la'reverse_range loop
            vb := res (ra'length - 1 + i);
            lb := not la (i);
            res (ra'length - 1+ i) := compute_sum (carry, vb, lb);
            carry := compute_carry (carry, vb, lb);
          end loop;
          vb := res (res'left);
          lb := not la (la'left);
          res (res'left) := compute_sum (carry, vb, lb);
       end if;
       return res;
    end "*";
```
There are other openieee operators that would exhibit warnings per bit needlessly. Any place the added warning message shows up likely should warrant review. 

The openieee packages are deprecated in favor of revision rationalized versions of the IEEE -2019 packages which are distributed under more permissive license (Apache License Version 2.0).

jserot, 2021-02-22:
Thanks for your answers. 
But i'm a bit confused regarding the status of the `ieee` and `openieee` libs in the GHDL distributions.
I.e. 
- there's no `openieee` lib in the GHDL 0.35 dist (at least the `GHDL 0.35 (tarball) [Dunoon edition]` i was using before; only `ieee` (in the `v93` subdir)
- there's no `ieee` lib the GHDL 0.37 dist i have switched to (`GHDL 0.37 (Ubuntu 0.37+dfsg-1ubuntu1) [Dunoon edition]`). More precisely, there's a file `.../ghdl/mcode/ieee/v93/ieee-obj93.cf` but only one (and buggy) file `numeric_std-body.vhd', in dir `.../ghdl/src/openieee/v93`

When compiling, is GHDL using a pre-compiled version of `ieee` (and in this case, where is it located in the distro ?) or the code provided in the `src` subdir ?

Thanks for help 

ps : btw, i've found a workaround to the bug mentioned in this issue : i simply copied the file `numeric_std.vhd` and `numeric_std-body.vhd` from the `ieee` dir in the 0.35 distro to the directory containing my application code, compiled it and replaced `use ieee.numeric_std.all` by `use work.numeric_std.all`. Not very elegant but works.

eine, 2021-02-22:
@jserot, diogratia and me already explained it in our previous messages. However, I understand it might not be easy to get. Let me try rewording it:

- IEEE libraries were not open source until December 2019.
- For many years, GHDL distributed IEEE libraries without modifications.
- IEEE libraries could not be included in Debian, because of the license. GHDL was removed from Debian.
- Tristan wrote openieee, for allowing GHDL to be distributed in Debian with some minimal functionality.
- GHDL was built/distributed with IEEE libs in all platforms, except on platforms requiring GPL compatibility; i.e. Debian (and Ubuntu), where openieee was used.
- In December 2019, VHDL 2019 was released. The LRM is not open source yet, but standard and IEEE libs were published under Apache 2.0.
- Tristan backported those Apache 2.0 sources to 93 and 08.
- Now, either "regular" or "gplcompat" builds of GHDL do include the same complete IEEE libraries. The only difference is that VITAL libraries are not included in "gplcompat" (because those were not released in 2019).
- openieee libs are preserved in the codebase for historical purposes and because there are performance gains when dealing with math features. However, openieee is neither built or distributed anymore.

> there's no openieee lib in the GHDL 0.35 dist i was using before; only ieee (in the v93 subdir)

That was/is likely a non GPL compatible build.

> there's no ieee lib the GHDL 0.37 dist i have switched to 

This was/is a GPL compatible build.

> When compiling, is GHDL using a pre-compiled version of ieee (and in this case, where is it located in the distro ?)

The libraries are located in subdir https://github.com/ghdl/ghdl/tree/master/libraries. They are built and installed along with the GHDL binary and shared libs, because VHDL is not usable without std libs, and it's barely usable without IEEE libs. You can find the built libs in `$PREFIX/lib/ghdl/ieee`, `$PREFIX/lib/ghdl/std` and `$PREFIX/lib/ghdl/vendors`. For LLVM or GCC backends, you will find `*.o` files along with VHDL sources.

> ps : btw, i've found a workaround to the bug mentioned in this issue : i simply copied the file numeric_std.vhd and numeric_std-body.vhd from the ieee dir in the 0.35 distro to the directory [...]

My suggestion is: use GHDL >= 1.0.0, and forget about these ieee vs openieee issues. Your workaround was effectively using IEEE instead of openieee. That's the default on any platform for GHDL >= 1.0.0.

jserot, 2021-02-23:
@eine
Thanks a lot for your detailed explanations.
IIUC, and as usual, unfortunately, licensing issues complicates things...
My 0.35 ghdl version was running on Mac OS and the 0.37 under Ubuntu. I now therefore understand the observed discrepancy.
In the latter,  the reason i was not able to find the .o files was that i installed the mcode version, wasn't it ?

I will switch to 1.0.0 ASAP.

eine, 2021-02-23:
> IIUC, and as usual, unfortunately, licensing issues complicates things...

Agree. But they also make users free or dependent. We nead to deal with these issues. Fortunately, VHDL seems to be moving to open source, which will make everything easier. It is already.

> In the latter, the reason i was not able to find the .o files was that i installed the mcode version, wasn't it ?

That is correct. mcode is a built-in in-memory assembler code generator. Therefore, it does not use `*.o` files and it does netiher support linking simulation models to other C sources or objects (which LLVM or GCC do support). mcode can execute shared libraries, tho. Hence co-simulation features are possible regardless of the backend.

> I will switch to 1.0.0 ASAP.

I'm not sure about the updating policy in Ubuntu, but I'd expect packagers to update GHDL. See https://packages.ubuntu.com/search?keywords=ghdl (note "hirsute").