ghdl/ghdl #1652: grt fails to build with GCC/GNAT 11

sharkcz, 2021-02-17:
grt fails to build with latest gcc (GCC) 11.0.0 20210210 (Red Hat 11.0.0-0) in Fedora
````
...
gcc -c -I./ -I/builddir/build/BUILD/gcc-10.2.1-20200826/ghdl-llvm/./src/grt -I.. -gnatec/builddir/build/BUILD/gcc-10.2.1-20200826/ghdl-llvm/./src/grt/grt.adc -gnatdY -g -I- /builddir/build/BUILD/gcc-10.2.1-20200826/ghdl-llvm/src/grt/grt-signals.adb
grt-signals.adb:556:20: illegal attribute for discriminant-dependent component
gnatmake: "/builddir/build/BUILD/gcc-10.2.1-20200826/ghdl-llvm/src/grt/grt-signals.adb" compilation error
````
The offending line is https://github.com/ghdl/ghdl/blob/master/src/grt/grt-signals.adb#L556

tgingold, 2021-02-17:
Does it work if you replaced `'Access` by `'Unrestricted_Access` ?

sharkcz, 2021-02-17:
seems it doesn't
````
grt-signals.adb:556:37: no selector "Unrestricted_Access" for type "Driver_Type" defined at grt-signals.ads:76
````

tgingold, 2021-02-17:
Did you put a dot before (instead of a single quote) ?
The error message is not expected.

sharkcz, 2021-02-17:
yes, there was a dot
````
         if Sig.S.Drivers (I).Proc = Proc then
            return Sig.S.Drivers (I)'Unrestricted_Access;
         end if;
````
^^ compiles

commit 8356ea3b (2021-02-18): grt-signals.adb: fix access.  Fix #1652
closed 2021-02-18 by tgingold