ghdl/ghdl #1654: PSL: abort operator acts synchronous, should act asynchronous

tmeissner, 2021-02-18:
**Description**
The PSL `abort` operator should work on asynchronous events which aren't aligned to clock. It is defined like an alias for the `sync_abort` operator, as defined in IEEE 1850-2010 6.2.1.5.1:

> The abort and async_abort operators accept asynchronous abort conditions as well.

At the moment, GHDL implements the `abort` operator like the `sync_abort` operator is defined:

> The sync_abort operator expects the abort condition to occur in a cycle in which the context clock holds.

Further relevant sections about the behavior of the `abort`operators in that section 6.2.1.5.1:

> An abort / async_abort property holds in the current cycle of a given path iff
> - The FL Property that is the left operand holds, or
> - The FL Property that is the left operand does not fail (see 4.4.5) prior to the first cycle (of the path defined by the base clock context) in which the Boolean condition that is the right operand holds.

If I understand that correctly, the property fails synchronously in the cycle in which the abort condition holds. But the `abort` condition itself can be asynchronous. For comparison, the description of the `sync_abort` behaviour:

> A sync_abort property holds in the current cycle of a given path iff
> - The FL Property that is the left operand holds, or
> - The FL Property that is the left operand does not fail (see 4.4.5) prior to the first cycle (of the path defined by the clock context of the abort property) in which the Boolean condition that is the right operand holds.

**Expected behaviour**
The `abort` operator should work asynchronous as defined in the IEEE standard.

**How to reproduce?**
Get this file: https://github.com/tmeissner/psl_with_ghdl/blob/master/issues/issue_165x.vhd

```sh :image: ghdl/vunit:llvm
ghdl -a --std=08 issue_165x.vhd
ghdl -e --std=08 test_issue
```

If you run it with `stop-time` option:

```sh :image: ghdl/vunit:llvm
./test_issue 
issue_165x.vhd:82:3:@5ns:(psl assertion error): Assertion violation
simulation stopped @29ns with status 0
```
The assertion should hold with the used `abort` operator.

- OS: Debian Buster
- Origin:
  - [x] Docker (Image:ghdl/vunit:llvm)

tgingold, 2021-02-20:
I fear handling nested async_abort wouldn't easy.  But maybe we could handle only outer async_abort.

tmeissner, 2021-02-20:
I think, outer `async_abort` would be enough for most use cases.

closed 2021-08-31 by tgingold

tgingold, 2021-08-31:
So async_abort and sync_abort are now handled both in simulation and in synthesis.

Should abort now behaves like async_abort ?  I think this is the last step.

tmeissner, 2021-08-31:
I think so. If I read the LRM correctly, `abort` is like an alias to `async_abort`:

In section **6.2.1.5.1** of IEEE 1850-2010:
> NOTE 1—The abort operator is identical to the async_abort operator. It is currently maintained in the language
for reasons of backward compatibility.

commit 09192336 (2021-08-31): trans-chap9.adb: handle async_abort, sync_abort.  Fix #1654