ghdl/ghdl #1655: Run Bug with -fsynopsys --wae option

samlei-research, 2021-02-18:
**Description**
I am trying to simulate my testbench. At first I run the ghld -a command with all related files and after that: ghdl -r -fsynopsys target --wave=target.ghw 

**Expected behaviour**
The expected behaviour would open Gtkwave with the .ghw file. Instead the following message in the terminal arises:

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 1.0-dev (v0.37.0-1076-g9db35195) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-linux-gnu
/home/mzdev/StarkStromWorkspace/Umrichter/Endat/absenc/src/
Command line:
ghdl -r -fsynopsys endat_tb --wave=endat_tb.ghw
Exception SYSTEM.ASSERTIONS.ASSERT_FAILURE raised
Exception information:
raised SYSTEM.ASSERTIONS.ASSERT_FAILURE : trans-chap5.adb:820
Call stack traceback locations:
0x7f499e5797b1 0x55e835ba7a7e 0x55e835ba80d4 0x55e835ba0d15 0x55e835c0c226 0x55e835c0c395 0x55e835c0e330 0x55e835b72d08 0x55e835b6fc90 0x55e835b7e174 0x55e835c1039f 0x55e835b4fe94 0x55e835a890b6 0x55e835c13cad 0x55e83583e47b 0x7f499db2eb95 0x55e83583cd58 0xfffffffffffffffe
******************************************************************

**How to reproduce?**
In the following structure are all files that are needed.
Just use "make sim" in there.

[src_ghdl_bug.zip](https://github.com/ghdl/ghdl/files/6004415/src_ghdl_bug.zip)

tgingold, 2021-02-19:
I can reproduce the issue.  But I doubt it is related to `--wave` or `-fsynopsys`

samlei-research, 2021-02-19:
> I can reproduce the issue. But I doubt it is related to `--wave` or `-fsynopsys`

If i comment out "--wave" the error still occurs. So the issue is because of -fsynopsys
Has anybody a workaround or a quick fix?

tmeissner, 2021-02-19:
I think @tgingold means that he doesn't believe that the issue is caused by these two options. Using of synopsys non-standard libs is always a bad thing, but probably not the root of yiur issue.

samlei-research, 2021-02-19:
> I think @tgingold means that he doesn't believe that the issue is caused by these two options. Using of synopsys non-standard libs is always a bad thing, but probably not the root of your issue.

oh yes right I misunderstood, but as you guessed it doesn't work without the Synopsys lib either.

diogratia, 2021-02-19:
Potential workaround
--
If you checkout the source for

Commit:9db35195bbafe7e7adf5dcf347fc1a30cc6873f8 [9db35195]

src/vhdl/translate/trans-chap5.adb:

```ada
   procedure Elab_Generic_Map_Aspect
     (Header : Iir; Map : Iir; Formal_Env : Map_Env)
   is
      Actual_Env : Map_Env;
      Assoc  : Iir;
      Formal : Iir;
      Inter : Iir;
   begin
      Save_Map_Env (Actual_Env, Formal_Env.Scope_Ptr);

      --  Elab generics, and associate.
      Assoc := Get_Generic_Map_Aspect_Chain (Map);
      Inter := Get_Generic_Chain (Header);
      while Assoc /= Null_Iir loop
         Formal := Get_Association_Formal (Assoc, Inter);
         Open_Temp;
         case Get_Kind (Assoc) is
            when Iir_Kind_Association_Element_By_Expression =>
               declare
                  Targ : Mnode;
               begin
                  if Get_Whole_Association_Flag (Assoc) then
                     Set_Map_Env (Formal_Env);
                     Chap4.Elab_Object_Storage (Formal);
                     Targ := Chap6.Translate_Name (Formal, Mode_Value);
                     Set_Map_Env (Actual_Env);
                     Chap4.Elab_Object_Init
                       (Targ, Formal, Get_Actual (Assoc), Alloc_System);
                  else
                     Set_Map_Env (Formal_Env);
                     Targ := Chap6.Translate_Name (Formal, Mode_Value);
                     Set_Map_Env (Actual_Env);
                     Chap7.Translate_Assign
                       (Targ, Get_Actual (Assoc), Get_Type (Formal));
                  end if;
               end;
            when Iir_Kind_Association_Element_Open =>
               case Get_Kind (Formal) is
                  when Iir_Kind_Interface_Constant_Declaration =>
                     declare
                        Value : constant Iir := Get_Default_Value (Formal);
                     begin
                        pragma Assert (Is_Valid (Value));
                        Set_Map_Env (Formal_Env);
                        Chap4.Elab_Object_Value (Formal, Value);
                        Chap9.Destroy_Types (Value);
                        Set_Map_Env (Actual_Env);
                     end;
                ...
```
The assertion that's failing at line 820 is `pragma Assert (Is_Valid (Value));` in a generic map aspect.

If in endat_tb.vhdl you comment out the configuration specification:
```vhdl
-- for u_master: master use entity work.master(absenc_master_rtl);
```
Elaboration will not produce the unhandled assertion:
```none
diogratia@Macbook: make
ghdl -a -fsynopsys  absenc_pkg.vhdl
ghdl -a -fsynopsys  absenc_utils.vhdl
ghdl -a -fsynopsys  absenc_master.vhdl
ghdl -a -fsynopsys  absenc_master_endat.vhdl
ghdl -a -fsynopsys  endat_tb.vhdl
ghdl -e -fsynopsys  endat_tb
endat_tb.vhdl:60:3:error: no actual for generic "clk_freq"
endat_tb.vhdl:60:3:error: no actual for generic "enable_endat"
endat_tb.vhdl:60:3:error: no actual for generic "enable_biss"
endat_tb.vhdl:60:3:error: no actual for generic "enable_ssi"
absenc_master.vhdl:490:1:warning: instance "master_biss" of component "master_biss" is not bound [-Wbinding]
absenc_master.vhdl:52:14:warning: (in default configuration of master(absenc_master_rtl))
absenc_master.vhdl:522:1:warning: instance "master_ssi" of component "master_ssi" is not bound [-Wbinding]
absenc_master.vhdl:52:14:warning: (in default configuration of master(absenc_master_rtl))
ghdl:error: compilation error
make: *** [endat_tb] Error 1
diogratia@Macbook: mate absenc_master.vhdl
```
(Using a makefile with a missing dependency added.)

The errors are due to a lack of not supplying generic values to generics without default values. Here besides the three generic map aspects found in absenc_master.vhdl there is a missing generic_map aspect in endat_tb.vhdl wherein the bug likely occurred.

With only one master architecture (absenc_master_rt) the configuration specification is not needed, instead depending on a default binding indication that generates the three errors.

Because a generic map aspect is required for those four generic constants lacking default values it could be added to the binding indication in the (uncommented) configuration specification, e.g.:

```vhdl
-- Configuration
-- for u_master: master use entity work.master(absenc_master_rtl);
 for u_master: master use entity work.master(absenc_master_rtl)
  generic map (42, FALSE, FALSE, FALSE);
```
(Without chasing down how CLK_FREQ is use and the appropriateness of the integer literal this allows elaboration to complete.)

tgingold, 2021-02-19:
Yes, that's correct.  There is a missing check that should catch this error.

closed 2021-02-20 by tgingold

samlei-research, 2021-02-20:
> ## Potential workaround
> 
> If you checkout the source for
> 
> Commit:9db35195bbafe7e7adf5dcf347fc1a30cc6873f8 [[9db3519](https://github.com/ghdl/ghdl/commit/9db35195bbafe7e7adf5dcf347fc1a30cc6873f8)]
> 
> src/vhdl/translate/trans-chap5.adb:
> 
> ```ada
>    procedure Elab_Generic_Map_Aspect
>      (Header : Iir; Map : Iir; Formal_Env : Map_Env)
>    is
>       Actual_Env : Map_Env;
>       Assoc  : Iir;
>       Formal : Iir;
>       Inter : Iir;
>    begin
>       Save_Map_Env (Actual_Env, Formal_Env.Scope_Ptr);
> 
>       --  Elab generics, and associate.
>       Assoc := Get_Generic_Map_Aspect_Chain (Map);
>       Inter := Get_Generic_Chain (Header);
>       while Assoc /= Null_Iir loop
>          Formal := Get_Association_Formal (Assoc, Inter);
>          Open_Temp;
>          case Get_Kind (Assoc) is
>             when Iir_Kind_Association_Element_By_Expression =>
>                declare
>                   Targ : Mnode;
>                begin
>                   if Get_Whole_Association_Flag (Assoc) then
>                      Set_Map_Env (Formal_Env);
>                      Chap4.Elab_Object_Storage (Formal);
>                      Targ := Chap6.Translate_Name (Formal, Mode_Value);
>                      Set_Map_Env (Actual_Env);
>                      Chap4.Elab_Object_Init
>                        (Targ, Formal, Get_Actual (Assoc), Alloc_System);
>                   else
>                      Set_Map_Env (Formal_Env);
>                      Targ := Chap6.Translate_Name (Formal, Mode_Value);
>                      Set_Map_Env (Actual_Env);
>                      Chap7.Translate_Assign
>                        (Targ, Get_Actual (Assoc), Get_Type (Formal));
>                   end if;
>                end;
>             when Iir_Kind_Association_Element_Open =>
>                case Get_Kind (Formal) is
>                   when Iir_Kind_Interface_Constant_Declaration =>
>                      declare
>                         Value : constant Iir := Get_Default_Value (Formal);
>                      begin
>                         pragma Assert (Is_Valid (Value));
>                         Set_Map_Env (Formal_Env);
>                         Chap4.Elab_Object_Value (Formal, Value);
>                         Chap9.Destroy_Types (Value);
>                         Set_Map_Env (Actual_Env);
>                      end;
>                 ...
> ```
> 
> The assertion that's failing at line 820 is `pragma Assert (Is_Valid (Value));` in a generic map aspect.
> 
> If in endat_tb.vhdl you comment out the configuration specification:
> 
> ```vhdl
> -- for u_master: master use entity work.master(absenc_master_rtl);
> ```
> 
> Elaboration will not produce the unhandled assertion:
> 
> ```
> diogratia@Macbook: make
> ghdl -a -fsynopsys  absenc_pkg.vhdl
> ghdl -a -fsynopsys  absenc_utils.vhdl
> ghdl -a -fsynopsys  absenc_master.vhdl
> ghdl -a -fsynopsys  absenc_master_endat.vhdl
> ghdl -a -fsynopsys  endat_tb.vhdl
> ghdl -e -fsynopsys  endat_tb
> endat_tb.vhdl:60:3:error: no actual for generic "clk_freq"
> endat_tb.vhdl:60:3:error: no actual for generic "enable_endat"
> endat_tb.vhdl:60:3:error: no actual for generic "enable_biss"
> endat_tb.vhdl:60:3:error: no actual for generic "enable_ssi"
> absenc_master.vhdl:490:1:warning: instance "master_biss" of component "master_biss" is not bound [-Wbinding]
> absenc_master.vhdl:52:14:warning: (in default configuration of master(absenc_master_rtl))
> absenc_master.vhdl:522:1:warning: instance "master_ssi" of component "master_ssi" is not bound [-Wbinding]
> absenc_master.vhdl:52:14:warning: (in default configuration of master(absenc_master_rtl))
> ghdl:error: compilation error
> make: *** [endat_tb] Error 1
> diogratia@Macbook: mate absenc_master.vhdl
> ```
> 
> (Using a makefile with a missing dependency added.)
> 
> The errors are due to a lack of not supplying generic values to generics without default values. Here besides the three generic map aspects found in absenc_master.vhdl there is a missing generic_map aspect in endat_tb.vhdl wherein the bug likely occurred.
> 
> With only one master architecture (absenc_master_rt) the configuration specification is not needed, instead depending on a default binding indication that generates the three errors.
> 
> Because a generic map aspect is required for those four generic constants lacking default values it could be added to the binding indication in the (uncommented) configuration specification, e.g.:
> 
> ```vhdl
> -- Configuration
> -- for u_master: master use entity work.master(absenc_master_rtl);
>  for u_master: master use entity work.master(absenc_master_rtl)
>   generic map (42, FALSE, FALSE, FALSE);
> ```
> 
> (Without chasing down how CLK_FREQ is used and the appropriateness of the integer literal this allows elaboration to complete.)

Thanks a lot. Its working now.

commit a18c2b2b (2021-02-20): vhdl-canon.adb: add a missing check on generic associations.  Fix #1655