ghdl/ghdl #1657: Concatenation with User Defined Arrays results in a GHDL Bug

jwprice100, 2021-02-19:
**Description**
The problem requires creating an array of some type, say a std_logic_vector. 
```
type std_logic_vector_array is array(natural range <>) of std_logic_vector;
```
If one we're to then try to assign an array the result of a concatenation of another array and a std_logic_vector:
```
array_1 <= array_2 & std_logic_vector'(x"FF");
```
Then GHDL throws a GHDL bug.

**Expected behaviour**
I expect GHDL to either generate a new std_logic_vector_array OR for a compiler error to be thrown for no matching function & for types std_logic_vector_array and std_logic_vector. Either is acceptable. 

**How to reproduce?**
I've attached a MWE: [concat_array_test.zip](https://github.com/ghdl/ghdl/files/6012474/concat_array_test.zip)

**Context**
Please, provide the following information:

- OS: Ubuntu 20.04
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [X] Built from sources: `commit SHA`

If a `GHDL Bug occurred` block is shown in the log, please paste it here:

```
******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (tarball) [Dunoon edition]
Compiled with GNAT Version: 9.3.0
Target: x86_64-linux-gnu
/home/james/ghdl_examples/concat_array_test/
Command line:
ghdl -r --std=08 test
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : trans.adb:1672
Call stack traceback locations:
0x560e77a3a24e 0x560e77c535b6 0x560e77c540c4 0x560e77c56103 0x560e77c54b24 0x560e77c54d83 0x560e77c16071 0x560e77c15a08 0x560e77c158c9 0x560e77c155f4 0x560e77c185ab 0x560e77c20198 0x560e77c1fd4e 0x560e77bea81c 0x560e77bea947 0x560e77beafa0 0x560e77beb736 0x560e77bed2d0 0x560e77bed425 0x560e77bdb7e2 0x560e77bdbbb5 0x560e77bed2a3 0x560e77bed425 0x560e77bf0841 0x560e77bf5124 0x560e77c63074 0x560e77bcb175 0x560e77bc82c8 0x560e77bd63cc 0x560e77c679e3 0x560e77ba6bf1 0x560e77adf567 0x560e77c6b300 0x560e77895a67 0x7f680748d0b1 0x560e778941cc 0xfffffffffffffffe
******************************************************************
```
Do note if I define my own & operator, it works as expected.

tgingold, 2021-02-20:
Not everything from vhdl 2008 is implemented!

jwprice100, 2021-02-20:
No worries, it supports way more of VHDL 2008 than I anticipated! For more information, It does seem to work if I manually overload the & operator. Since the LRM talks about 1D arrays, I think requiring overloading the & operator is fine. The only bug then is not reporting that there is no matching function for &.

commit 904efedb (2021-09-07): trans-chap7: Handle unbounded elements in Translate_Concatenation

Fix #1831
Fix #1657
closed 2021-09-07 by tgingold