ghdl/ghdl #1659: Error: invalid instruction suffix for `push'

bmmajeed, 2021-02-22:
**Description**
I'm new to GHDL. I've installed it successfully but when I'm trying to compile my code, I'm getting the following error. I tried it for multiple files/circuits but the error is the same. I tried running my circuit at EDAPlayground using the GHDL simulator and it works fine there. 

**Error**
Adder.s: Assembler messages:
Adder.s:144: Error: invalid instruction suffix for `push'
Adder.s:284: Error: invalid instruction suffix for `push'
Adder.s:339: Error: invalid instruction suffix for `push'
Adder.s:432: Error: invalid instruction suffix for `push'
ghdl: compilation error

**Expected behaviour**
Normal compilation.

**How to reproduce?**
Here is my code for Adder.vhdl

```-- Code your testbench here
library IEEE;
use IEEE.std_logic_1164.all;

entity adder is
-- `i0`, `i1`, and the carry-in `ci` are inputs of the adder.
-- `s` is the sum output, `co` is the carry-out.
	port (i0, i1 : in bit; ci : in bit; s : out bit; co : out bit);
end adder;

architecture rtl of adder is
begin
-- This full-adder architecture contains two concurrent assignments.
-- Compute the sum.
s <= i0 xor i1 xor ci;
-- Compute the carry.
co <= (i0 and i1) or (i0 and ci) or (i1 and ci);
end rtl;
```

- OS: Ubuntu 20.04.2.0
- GHDL Version: ghdl-0.29-i686-pc-linux

eine, 2021-02-22:
@bmmajeed, GHDL 0.29 is several years old. Please, use a more recent version. You can find pre-built tarballs for Ubuntu 20.04 in https://github.com/ghdl/ghdl/releases.

eine, 2021-02-22:
@Xiretza, you would be surprised about how many people arrive to http://ghdl.free.fr/download.html

Xiretza, 2021-02-22:
Oh wow, yep, that page along with [some old wiki](http://ghdl.free.fr/site/pmwiki.php?n=Main.Download) are currently the top two duckduckgo results for "ghdl download". They should probably either be converted to a redirect to github or at least get a giant red warning banner - @tgingold?

tmeissner, 2021-02-22:
Maybe we should also update this old download subpage with a hint to go to github.

eine, 2021-02-22:
@tmeissner, strictly, there is no "we" here. That is Tristan's very personal page/server/domain. I believe he is the only one who has access (or who knows where is it located). He did update the homepage, indeed, for pointing to GitHub: http://ghdl.free.fr/. Still, search engines are sometimes better than they should...

I remember scrapping it a few times with https://www.httrack.com/, when updating the docs. There are several pages that we might relocate and add redirections/placeholders instead.

bmmajeed, 2021-02-22:
> @Xiretza, you would be surprised about how many people arrive to http://ghdl.free.fr/download.html

Exactly, this is the reason I downloaded Version 0.29 because this one seems to be the latest on this webpage. Anyways, can someone please share that 

1- How do I uninstall GHDL 0.29 now? 
2- The latest version I found here is https://github.com/ghdl/ghdl/releases/tag/v1.0.0 Please correct me if this is not the latest stable version. 
3- Is there any chance that I'll get the same error of 'push' and 'pull' on the latest version as well? 

Thanks.

Xiretza, 2021-02-22:
> 1- How do I uninstall GHDL 0.29 now?

That depends entirely on how you installed it - if you used `make install`, the installed files will be scattered over the PREFIX, which probably defaulted to `/usr/local/`. Maybe there's a `make uninstall` target in those old releases, if not, you'll have to find those files and delete them manually.

> 2- The latest version I found here is https://github.com/ghdl/ghdl/releases/tag/v1.0.0 Please correct me if this is not the latest stable version.

That's correct.

> 3- Is there any chance that I'll get the same error of 'push' and 'pull' on the latest version as well?

I'd say it's very unlikely, this looks like a very low-level code-generation problem, which is done by gcc. Especially considering there are now several independent backends, not just gcc, a more recent version should work fine.

bmmajeed, 2021-02-22:
> That depends entirely on how you installed it - if you used `make install`, the installed files will be scattered over the PREFIX, which probably defaulted to `/usr/local/`. Maybe there's a `make [...]

Thanks so much. Just one more quick question. Is there any guide available to install this new version?

diogratia, 2021-02-22:
>1- How do I uninstall GHDL 0.29 now?

You can recover the list of files from the archive used to install. See [How i can remove ghdl 0.29 from Debian?](https://stackoverflow.com/questions/40556430/how-i-can-remove-ghdl-0-29-from-debian) on Stackoverflow.

Presumably from [Downloading GHDL](http://ghdl.free.fr/download.html).

eine, 2021-02-22:
Instead of downloading the sources of v1.0.0 and building GHDL, I suggest downloading the pre-built tarballs from https://github.com/ghdl/ghdl/releases/tag/nightly. You can get any of the following:

- https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-mcode.tgz
- https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-llvm.tgz
- https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-gcc.tgz

After uninstalling the previous one as diogratia said, you can install any of these as you installed the other one (http://ghdl.free.fr/INSTALL):

```sh
sudo tar -C /usr/local -xvf TARBALL.tgz
```

Anyway, I would recommend not to install GHDL from tarballs. GHDL is available through `apt`.

bmmajeed, 2021-02-23:
> Instead of downloading the sources of v1.0.0 and building GHDL, I suggest downloading the pre-built tarballs from https://github.com/ghdl/ghdl/releases/tag/nightly. You can get any of the following:
> 
> * https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-mcode.tgz
> * https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-llvm.tgz
> * https://github.com/ghdl/ghdl/releases/download/nightly/ghdl-gha-ubuntu-20.04-gcc.tgz
> 
> After uninstalling the previous one as diogratia said, you can install any of these as you installed the other one (http://ghdl.free.fr/INSTALL):
> 
> ```shell
> sudo tar -C /usr/local -xvf TARBALL.tgz
> ```
> 
> Anyway, I would recommend not to install GHDL from tarballs. GHDL is available through `apt`.

Thanks, Eine. 

1- Why do you recommend not to unstall GHDL from tarballs?
2- How to get and install this version of GHDL through apt? 

I'm new to the Linux so I want to get through the basic reasons of the stuff.

eine, 2021-02-23:
@bmmajeed on GNU/Linux, tools are typically installed to some predefined locations (e.g. `/usr/local/bin`, `/usr/local/lib`...). Unlike Windows, there is no one specific subidr for each app/tool. All the binaries of all tools are put together, all the libs are put together, etc. As a result, it is very painful to manually install/uninstall all the files related to some tool. Unless you do it from a tarball and keep that around, to use diogratia's trick.

At the same time, GNU/Linux distributions are known for having so-called "package managers" together with "repositories". A repository is a server containing thousands of pre-built tools. Unlike Windows, on Linux you don't need to search each tool in the browser and then download and install it manually. On Linux, you use a package manager:

- `apt update`: fetch the repository and update the local list of available packages.
- `apt search ghdl`: search if a package named ghdl exists.
- `apt install ghdl`: install ghdl for me.

Next time you do `apt update` and `apt upgrade`, it will automatically install a new version of GHDL (if available).

Moreover, the packages you get through the manager are guaranteed to work for your OS, because those were built specifically for the libraries and versions you have. 

When you download a tarball for Linux, you have no guarantee about it working. In this case, we do generate pre-built tarballs for Ubuntu 20.04. However, if you used any othe Linux distribution, you might have issues.

`apt` (or `apt-get`) is the default package manager on Debian/Ubuntu. On other distributions, managers are `pacman`, `dnf`, `yum`, etc. All of the are equivalent and serve the same purpose.

Therefore, installing GHDL from tarballs or building it from sources is only suggested in case the version available through the package manager is not enough. That might be because it is outdated or because it was built with some options that don't fit your needs. In this specific case, it might be problematic if the GHDL installed through `apt` was built with openieee. However, that should be fixed as soon as they update to `1.0.0`, which was released a month ago.

> I'm new to the Linux so I want to get through the basic reasons of the stuff.

Please, do see some Youtube video or some introduction to the GNU/Linux file structure and how package managers are used. I'm sure you will find examples specific about Ubuntu which will be easy to follow. I did a very basic introduction above, but explaining it in deep is out of the scope of this project.

tgingold, 2021-02-23:
I am closing this issue.  I have added a warning to the old html page.

closed 2021-02-23 by tgingold