ghdl/ghdl #1661: Synthesis: unhandled IIR_PREDEFINED_INTEGER_TO_STRING

marph91, 2021-02-23:
**Description**
Hi, I'm trying to synthesize a project, which uses [JSON-for-VHDL](https://github.com/Paebbels/JSON-for-VHDL). The synthesis fails. Below is a reproducer:

reproducer.vhd
```vhd
library ieee;
use ieee.std_logic_1164.all;

use work.json.all;

entity reproducer is
end entity;

architecture rtl of reproducer is
    constant C_CONFIG_FILE : STRING := "../json_file.json";
    constant C_JSON_CONTENT : T_JSON := jsonLoad(C_CONFIG_FILE);
    constant C_INTEGER_VECTOR : integer_vector := jsonGetIntegerArray(C_JSON_CONTENT, "integer_vector");
begin
end architecture;
```

reproducer.sh
```sh
set -e

git clone https://github.com/Paebbels/JSON-for-VHDL.git

rm -rf build
mkdir -p build
cd build

ghdl -a --std=08 ../JSON-for-VHDL/src/Encodings.pkg.vhdl
ghdl -a --std=08 ../JSON-for-VHDL/src/JSON.pkg.vhdl
ghdl -a --std=08 ../reproducer.vhdl

ghdl --synth --std=08 reproducer
```

json_file.json
```json
{
    "integer_vector": [
        0, 1, 0, 1
    ]
}
```

When executing `reproducer.sh`, ghdl synthesis fails with:
```
> ./reproducer.sh 
../JSON-for-VHDL/src/JSON.pkg.vhdl:298:24:(report note): jsonLoad: Filename ../json_file.json
../JSON-for-VHDL/src/JSON.pkg.vhdl:313:16:(report note): jsonReadFile: ../json_file.json
../JSON-for-VHDL/src/JSON.pkg.vhdl:1646:61:error: synth_static_monadic_predefined: unhandled IIR_PREDEFINED_INTEGER_TO_STRING

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r14.gb3c17274) [Dunoon edition]
Compiled with GNAT Version: 7.5.0
Target: x86_64-suse-linux
/home/martin/dev/vhdl/test_ghdlsynth/test_json/build/
Command line:
ghdl --synth --std=08 reproducer
Exception TYPES.INTERNAL_ERROR raised
Exception information:
raised TYPES.INTERNAL_ERROR : synth-static_oper.adb:723
Call stack traceback locations:
0x5ea3a8 0x5e4597 0x5e6cac 0x5f742d 0x5df7b2 0x5f6736 0x5d4872 0x5d6b40 0x5d6d82 0x5db685 0x5f744f 0x5d967e 0x5dacbc 0x5d64e7 0x5d6bfa 0x5d6d82 0x5db685 0x5f744f 0x5fbb5f 0x5fd1e7 0x5fd40d 0x5cb33a 0x5cb502 0x5c46f0 0x6476b0 0x557556 0x650b69 0x403beb 0x7f2b96177348 0x402b78 0xfffffffffffffffe
******************************************************************
```

**Context**
Please, provide the following information:

- OS: Opensuse Leap
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `commit b3c17274`

eine, 2021-02-23:
I can confirm on MSYS2 (`1.0.0 (1.0.0.r0.g2fb2384d)`). I'm not sure about textio being fully supported for synthesis yet. See https://github.com/ghdl/ghdl/issues/1004.

dalex78, 2026-08-19:
# Issue #1661 — Synthesis: unhandled `IIR_PREDEFINED_INTEGER_TO_STRING`

Analysis by an AI agent. Recommend (optionally merging the test and) closing the ticket.

## High-level summary

Statically folding an `integer'image(...)` call (GHDL's internal `IIR_PREDEFINED_INTEGER_TO_STRING`) during `--synth`'s constant evaluation failed with `synth_static_monadic_predefined: unhandled IIR_PREDEFINED_INTEGER_TO_STRING` and then crashed with an internal `TYPES.INTERNAL_ERROR` at `synth-static_oper.adb:723`. The original report hit this indirectly, through the third-party JSON-for-VHDL library's JSON-parsing-at-elaboration-time code.

## Investigation / current status: already fixed on master

The original repro needs cloning the external JSON-for-VHDL repository and a local JSON file, which doesn't fit a self-contained regression test. Instead, reproduced the exact named predefined function directly: a constant computed via `integer'image(42)`, whose length is then used to size a synthesized output. Against current `master`:

```vhdl
constant n1 : std_logic_vector (7 downto 0) := "00000010";
```

`--synth` now correctly statically folds `integer'image(42)'length` (= 2, i.e. `"00000010"` as an 8-bit vector) instead of failing on the unhandled predefined function. This must have been fixed as part of general improvements to predefined-function static folding in `src/synth` since this was reported (originally against a 2020 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/synth/issue1661/` with a minimal, self-contained reproducer that exercises `IIR_PREDEFINED_INTEGER_TO_STRING` directly (`integer'image(42)`, its length fed into a synthesized output), whose `testsuite.sh` synthesizes the design, re-analyzes the resulting netlist, and asserts the correctly folded value appears. This locks in the current, correct behavior as a regression guard.

closed 2026-08-20 by tgingold