ghdl/ghdl #1662: Segmentation fault and memory usage problem

VisitThePenguin, 2021-02-23:
A very ordinary ram declaration causes a segmentation fault (using ghdl -r) above a certain array size on: 1) linux target built from fresh git clone, 2) linux target obtained by ubuntu pacakge, both using mcode.
The same code does not crash on pre-compiled binary distribution for windows.

This isn't even a testbench, no simulation time will elapse, the elaborate unit should load in the simulator and then exit after time 0 evaluation with no more events to process.  This is what happens when launched with the windows build.

It should be noted that with the array depth at 2**18, 2**19, 2**20, you can see (using a process monitor) the memory usage of the ghdl -r process grow up to the physical limit of ram (assuming you already used ulimit to make the stack unlimited, without which it dies more quickly).

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity memtest is
  generic (G_AWIDTH : Positive := 17);
  port ( clk : in std_logic;
         addr : in std_logic_vector(G_AWIDTH-1 downto 0);
         din  : in std_logic_vector(31 downto 0);
         we   : in std_logic;
         qout : out std_logic_vector(31 downto 0) );
end entity;

architecture struct of memtest is
Constant C_DEPTH : Positive := 2**G_AWIDTH;

SubType Tdata is std_logic_vector(31 downto 0);
Type Tarr is Array(0 to C_DEPTH-1) of Tdata;
signal mem : Tarr := (OTHERS=>(OTHERS=>'0'));

signal int_addr : integer range 0 to C_DEPTH-1;

begin

int_addr <= to_INTEGER(UNSIGNED(addr));

p_reg: process(clk)
begin
if rising_edge(clk) then
  qout <= mem(int_addr);
  if we='1' then
    mem(int_addr)<=din;
  end if;
end if;
end process;

end struct;

====================
REPRODUCE WITH:

ghdl -a memtest.vhd
ghdl -e memtest
ghdl -r memtest

with the address width at 17 bits or below, ghdl does not segfault.  Depending on ulimit stack restrictions, it will segfault or get the process killed at deeper memory depths.

tried on linux mint (based on ubuntu 16.04)
tried on ubuntu 18.04
worked on windows 10 with ghdl 0.29 prebuild binary.

umarcor, 2021-02-23:
> It should be noted that with the array depth at 2**18, 2**19, 2**20, you can see (using a process monitor) the memory usage of the ghdl -r process grow up to the physical limit of ram (assuming you already used ulimit to make the stack unlimited, without which it dies more quickly).

When you say "grow up" you mean one fixed size for each generic value, isn't it? That is, you change it manually and it's constant during the execution?

> with the address width at 17 bits or below, ghdl does not segfault. Depending on ulimit stack restrictions, it will segfault or get the process killed at deeper memory depths.

I've seen similar issues (#822, #752), but `2^18 * 32 = 8 MB` seems too small for triggering those errors.

> tried on linux mint (based on ubuntu 16.04)
> tried on ubuntu 18.04
> worked on windows 10 with ghdl 0.29 prebuild binary.

That version of Windows is 10 years old! Anyway, I tested it with 1.0.0 on Windows (MSYS2) and I can confirm that no error is produced with `2^17` to `2^22`.

At the same time, I can confirm that a segmentation fault is produced in container `ghdl/ghdl:buster-mcode` with `G_AWIDTH>=18`:

```
# docker run --rm -itv /$(pwd)://src ghdl/ghdl:buster-mcode bash
root@14588767ef67:/# cd /src

root@14588767ef67:/src# ghdl -a memtest.vhd 
root@14588767ef67:/src# ghdl -e memtest
root@14588767ef67:/src# ghdl -r memtest
Segmentation fault

root@14588767ef67:/src# ghdl --version
GHDL 2.0.0-dev (v1.0.0-49-gba23c5a2) [Dunoon edition]
 Compiled with GNAT Version: 8.3.0
 mcode code generator
Written by Tristan Gingold.

Copyright (C) 2003 - 2021 Tristan Gingold.
GHDL is free software, covered by the GNU General Public License.  There is NO
warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
```

However, it is not produced in `ghdl/ghdl:buster-llvm-7`. Therefore, I think this issue is related to mcode backend.

VisitThePenguin, 2021-02-24:
yes, I mean that on several invocations in a row, I manually edited the address width, and watched the memory used by the process grow and grow until it crashed.  The important thing is that you reproduced it.  I did not try building and using the llvm backend... that is next.

VisitThePenguin, 2021-02-24:
this may be the same issue:

using the follow testbench does not work (no execution time, no waveform output) with G_AWIDTH at 18 or larger.
With G_AWIDTH at 17 or smaller, the simulation runs and the waveforms can be viewed.
(I rebuilt from source with llvm backend, so it doesn't seg fault on just the memtest entity, anyway....)

library ieee;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

entity tb is
  generic (G_AWIDTH : Positive := 18);
end entity;

architecture sim of tb is

signal clk  : std_logic:='0';
signal addr : std_logic_vector(G_AWIDTH-1 downto 0):=(OTHERS=>'0');
signal din  : std_logic_vector(31 downto 0):=(OTHERS=>'0');
signal we   : std_logic:='0';
signal qout : std_logic_vector(31 downto 0);
signal stop : std_logic:='0';

begin

uut : entity work.memtest
  generic map (G_AWIDTH => G_AWIDTH )
  port map ( clk => clk,
         addr => addr,
         din  => din,
         we   => we,
         qout => qout );

clk <= not clk after 500 ns when (stop='0');
stop <= '1' after 20 us;

p_main: process
begin
wait until rising_edge(clk);
wait until rising_edge(clk);
we<='1';
addr<=STD_LOGIC_VECTOR(to_UNSIGNED(125,G_AWIDTH));
din<=X"ABCDEF01";
wait until rising_edge(clk);
we<='1';
addr<=STD_LOGIC_VECTOR(to_UNSIGNED(126,G_AWIDTH));
din<=X"12345678";
wait until rising_edge(clk);
we<='0';
addr<=STD_LOGIC_VECTOR(to_UNSIGNED(125,G_AWIDTH));
wait until rising_edge(clk);
addr<=STD_LOGIC_VECTOR(to_UNSIGNED(126,G_AWIDTH));
wait until rising_edge(clk);
addr<=STD_LOGIC_VECTOR(to_UNSIGNED(127,G_AWIDTH));
wait until rising_edge(clk);
wait until rising_edge(clk);
wait until rising_edge(clk);
wait until rising_edge(clk);
wait;
end process;

end sim;

umarcor, 2021-02-24:
I cannot confirm that. I saved that last file as `test.vhd`, together with the one in the firs message above. Then:

```sh
# ghdl -a memtest.vhd 

# ghdl -a test.vhd

# ghdl -e tb

# ./tb.exe 

# ./tb.exe --wave=wave.ghw

# ./tb.exe --vcd=wave.vcd

# ls -lah
total 40M
drwxr-xr-x 1 umarcor None    0 Feb 24 16:05 .
drwxr-xr-x 1 umarcor None    0 Feb 24 16:03 ..
-rw-r--r-- 1 umarcor None 4.8K Feb 24 16:04 e~tb.o
-rw-r--r-- 1 umarcor None  12K Feb 24 16:04 memtest.o
-rw-r--r-- 1 umarcor None  819 Feb 24 16:04 memtest.vhd
-rwxr-xr-x 1 umarcor None 1.6M Feb 24 16:04 tb.exe
-rw-r--r-- 1 umarcor None  16K Feb 24 16:04 test.o
-rw-r--r-- 1 umarcor None 1.4K Feb 24 16:03 test.vhd
-rw-r--r-- 1 umarcor None  38M Feb 24 16:05 wave.ghw
-rw-r--r-- 1 umarcor None 2.5K Feb 24 16:07 wave.vcd
-rw-r--r-- 1 umarcor None  352 Feb 24 16:04 work-obj93.cf
```

GtkWave does not open the GHW file. The VCD file works:

```
# gtkwave wave.vcd 

GTKWave Analyzer v3.3.108 (w)1999-2020 BSI

[0] start time.
[20000000000] end time.
WM Destroy
```

[image]

umarcor, 2021-02-24:
The previous test was on MSYS2 (MINGW64) with LLVM backend. Using Debian containers, I get a segfault with mcode and an empty waveform with LLVM:

```sh
# docker run --rm -itv /$(pwd)://src -w //src  ghdl/ghdl:buster-mcode bash
root@b9c206a1e5bf:/src# ghdl -a memtest.vhd 
root@b9c206a1e5bf:/src# ghdl -a test.vhd
root@b9c206a1e5bf:/src# ghdl -e tb
root@b9c206a1e5bf:/src# ghdl -r tb --wave=wave.ghw
Segmentation fault
```

```sh
# docker run --rm -itv /$(pwd)://src -w //src  ghdl/ghdl:buster-llvm-7 bash
root@d1df997a6292:/src# ghdl -a memtest.vhd 
root@d1df997a6292:/src# ghdl -a test.vhd    
root@d1df997a6292:/src# ghdl -e tb       
root@d1df997a6292:/src# ghdl -r tb --wave=wave.ghw
root@d1df997a6292:/src# ls -la
total 1231
drwxrwxrwx 1 root root    4096 Feb 24 15:13 .
drwxr-xr-x 1 root root    4096 Feb 24 15:13 ..
-rwxr-xr-x 1 root root    6512 Feb 24 15:13 e~tb.o
-rwxr-xr-x 1 root root   15504 Feb 24 15:13 memtest.o
-rwxr-xr-x 1 root root     819 Feb 24 15:04 memtest.vhd
-rwxr-xr-x 1 root root 1201760 Feb 24 15:13 tb
-rwxr-xr-x 1 root root   21048 Feb 24 15:13 test.o
-rwxr-xr-x 1 root root    1349 Feb 24 15:03 test.vhd
-rwxr-xr-x 1 root root       0 Feb 24 15:13 wave.ghw
-rwxr-xr-x 1 root root     345 Feb 24 15:13 work-obj93.cf
```