ghdl/ghdl #1663: Stack overflow / memory problem?

treffnix1983, 2021-02-24:
I have small testbench in conduction with vuinit. This reads an image "test.bin" and the test does nothing than going trough this image and read byte per byte and puts this on an std_logic_vector.

This will work for an image size of 300 x 280. If I increase this size to 320x 280 I receive this error.
The file test.bin is just a hexdump with appropriate size, created with a hex editor.
----------------------------------------------------------------------------------------
library IEEE;
use IEEE.STD_LOGIC_1164.ALL;
use IEEE.NUMERIC_STD.ALL;

LIBRARY vunit_lib;
CONTEXT vunit_lib.vunit_context;
CONTEXT vunit_lib.vc_context;

entity tb_design_1 is
    GENERIC (runner_cfg : STRING);
end tb_design_1;

architecture Behavioral of tb_design_1 is

constant c_width      : integer := 300;
constant c_heigth     : integer := 280;  

signal test : std_logic_Vector(7 downto 0);
-- type of data   
subtype t_data is std_logic_vector(7 downto 0);

-- type of two dimensional array of data type  
type t_two_dim_array is array(0 to c_heigth-1, 0 to c_width-1) of t_data;

type t_file is file of character;
signal i : integer := 0;
signal j : integer := 0;

impure function init_two_dim_array return t_two_dim_array is
        use std.textio.all;
        type character_file is file of character;
        file file_pointer : character_file;
        file f_in         : t_file;
        variable v_char   : character;
        use ieee.numeric_std.all;
        variable v_output : t_two_dim_array;
    begin
        file_open(file_pointer, "test.bin", READ_MODE);
        for i in 0 to c_heigth-1 loop
            for j in 0 to c_width-1 loop
              if not endfile(file_pointer) then
                read(file_pointer, v_char); 
                v_output(i, j) := std_logic_vector(to_unsigned(character'pos(v_char),8));
              else
                report "UNEXPECTED END OF FILE (INPUT FILE)" severity FAILURE;
              end if;     
            end loop;
        end loop;
        file_close(file_pointer);
        return v_output;
    end function;

    

signal image : t_two_dim_array := init_two_dim_array;
signal nclk	:	std_logic;
begin

clk_process :process
begin
  nclk <= '0';
  wait for 5 ns;  -- for c_clk_period/2 signal is '0'.
  nclk <= '1';
  wait for 5 ns; 
end process;

process(nclk)
begin
  if (rising_edge(nclk)) then
	test <= image(i,j);
	j <= j + 1;
	if (j = c_width -1) then
		j <= 0;
		i <= i + 1;
	end if;
	if (i = c_heigth -1) then
		i <= 0;
	end if;
  end if;
end process;

  
main : PROCESS
BEGIN
    test_runner_setup(runner, runner_cfg);

    WHILE test_suite LOOP
        IF run("Example_Test1") THEN
			wait for 10 ms;
        END IF;
    END LOOP;

    test_runner_cleanup(runner);
END PROCESS;
test_runner_watchdog(runner, 20 ms); --set watchdog

end Behavioral;

-----------------------------------------------------------------------------------

- OS: Windows
- Origin:
  - [ ] Package manager: `version`
  - [*] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`

******************** GHDL Bug occurred ***************************
Please report this bug on https://github.com/ghdl/ghdl/issues
GHDL release: 2.0.0-dev (1.0.0.r49.gba23c5a2) [Dunoon edition]
Compiled with GNAT Version: 10.2.0
Target: i686-w64-mingw32
C:\Users\m.selke\Desktop\simpleGHDLTest\Unit-Test\
Command line:
C:\Program Files (x86)\GHDL\bin\ghdl.exe --elab-run --std=08 --work=ImageProcessing --workdir=C:\Users\m.selke\Desktop\simpleGHDLTest\Unit-Test\vunit_out\ghdl\libraries\ImageProcessing -PC:\Users\m.selke\Desktop\simpleGHDLTest\Unit-Test\vunit_out\ghdl\libraries\vunit_lib -PC:\Users\m.selke\Desktop\simpleGHDLTest\Unit-Test\vunit_out\ghdl\libraries\osvvm -PC:\Users\m.selke\Desktop\simpleGHDLTest\Unit-Test\vunit_out\ghdl\libraries\ImageProcessing --ieee=synopsys tb_design_1 behavioral -grunner_cfg=active python runner : true,enabled_test_cases : Example_Test1,output path : C::/Users/m.selke/Desktop/simpleGHDLTest/Unit-Test/vunit_out/test_output/ImageProcessing.tb_design_1.Example_Test1_92f187f088ef502246d6d6c8994c8aee2dc1a7e3/,tb path : C::/Users/m.selke/Desktop/simpleGHDLTest/testbench/,use_color : true --assert-level=error --vcd=C:\Users\m.selke\Desktop\simpleGHDLTest\Unit-Test\vunit_out\test_output\ImageProcessing.tb_design_1.Example_Test1_92f187f088ef502246d6d6c8994c8aee2dc1a7e3\ghdl\wave.vcd
Exception CONSTRAINT_ERROR raised
Exception information:
raised CONSTRAINT_ERROR : grt-vcd.adb:465 access check failed
Call stack traceback locations:
0x125319 0xf446d 0xf46ed 0xa3136 0x1c119c 0x1c1246 0x3f99a9 0x347570 0x29822e 0x3fa160 0xa2c7c 0xa1394 0x7607fa27 0x773876b2 0x77387682
******************************************************************

umarcor, 2021-02-24:
@treffnix1983, please use the following "block" syntax:

~~~
```vhdl
-- VHDL code here
-- For example:
entity ent is;
end entity;
```
~~~

That's shown as follows:

```vhdl
-- VHDL code here
-- For example:
entity ent is;
end entity;
```

BTW, which version of VUnit are you using?

treffnix1983, 2021-02-24:
Next time I will use the block syntax directly. 
Vunit version? I will look for it tomorrow at work, but I guess the newest.

treffnix1983, 2021-02-25:
I use vunit version 4.4.0

dalex78, 2026-08-20:
# Issue #1663 — Crash: large 2D array signal + `--vcd` wave dump

Analysis by an AI agent.

## High-level summary

A large two-dimensional array signal (a 300×280 array of 8-bit elements, an image buffer read from a file via VUnit in the original report) with `--vcd` wave dumping crashed with an internal `CONSTRAINT_ERROR` at `grt-vcd.adb:465 access check failed`. Despite the issue title ("stack overflow / memory problem"), the crash location (`grt-vcd.adb`, the VCD wave-dump code) points at wave-dump handling of the large array, not file reading or stack sizing per se.

## Investigation / current status: already fixed on master

The original repro depends on VUnit (`vunit_lib`/`vunit_context`), not available in this session. Reconstructed a standalone, VUnit-free MWE preserving the actual trigger (the large 2D array signal + `--vcd`), replacing the file-read initialization with a plain constant initial value (the file I/O isn't part of what crashes — the crash location is in VCD dumping, not file reading).

Reproduced against current `master`: `ghdl --elab-run --std=08 tb --vcd=wave.vcd` completes cleanly and produces a valid, non-empty VCD file. This must have been fixed as part of general improvements to multi-dimensional array wave-dump handling in `src/grt` since this was reported (originally against a 2021 build). No source change was needed.

## What the fix does

No source fix in this commit. Added `testsuite/gna/issue1663/` (a VUnit-free reconstruction of the repro, preserving the large 2D array + `--vcd` trigger) whose `testsuite.sh` elaborates and simulates the design with `--vcd` and asserts a non-empty wave file is produced with no `GHDL Bug occurred` crash. This locks in the current, correct behavior as a regression guard.

dalex78, 2026-08-24:
@tgingold should this ticket be closed ? (PR merged)

closed 2026-08-24 by tgingold