ghdl/ghdl #1667: Some sort of crash, likely due to me abusing syntax...

hughesjs, 2021-03-02:
This error was thrown by ghdl when elaborating this file:

[image]

`use work.language_types.all;

package lexer is	

	type tokenized_line is record
		op: 	string;
		params: string_vector;
	end record;

	function tokenize_line(l: string) return tokenized_line;

end package lexer;

package body lexer is

	function tokenize_line(l: string) return tokenized_line is
	begin
		return ("", ("",""));
	end function;

end package body;`

I'll be honest, I have no idea if that return statement is valid syntax or not, just thought I'd try it

hughesjs, 2021-03-02:
If it helps... It's running on WSL2 loaded with a source-built arch

tgingold, 2021-03-02:
vhdl 2008 aggregates are not yet fully supported.

hughesjs, 2021-03-02:
Absolutely no worries, I sorted it on my end, just reporting the bug like the program asked me to :)

commit ece6e78f (2021-08-06): testsuite/gna: add a test (close #1667)
closed 2021-08-06 by tgingold