ghdl/ghdl #1668: synthesis: unhandled (static) function: IIR_PREDEFINED_IEEE_1164_SCALAR_IS_X

dl3yc, 2021-03-02:
**Description**
I get the following error when trying to synthesize using ghdl:
```
../../src/ieee2008/fixed_generic_pkg-body.vhdl:3268:12:error: unhandled (static) function: IIR_PREDEFINED_IEEE_1164_SCALAR_IS_X
```
The code generates no errors during simulation.

**Expected behaviour**
I expect the code to synthesize without errors.

**How to reproduce?**
This demonstrates the problem:

```vhd :file: test.vhd
library ieee;
use ieee.std_logic_1164.all;
use ieee.fixed_pkg.all;

entity test is
end entity test;

architecture synthesis of test is
  signal a : sfixed(0 downto 0);
  signal b : std_logic;
begin

a <= (others => '0');

--b <= '1' when a >= to_sfixed(0, a); -- works
b <= '1' when a >= to_sfixed(0, 0, 0); -- fails

end architecture synthesis;
```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a --std=08 test.vhd
ghdl --synth --std=08 test
```
**Context**
Please, provide the following information:

- OS:
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources: `GHDL 2.0.0-dev (1.0.0.r55.gf5e3ef10) [Dunoon edition]`

treforsouthwell, 2021-05-14:
I'm seeing similar issues with other IEEE logic e.g:

 IIR_PREDEFINED_IEEE_STD_LOGIC_MISC_AND_REDUCE_SUV 

Is this an unimplemented feature?

tgingold, 2021-05-14:
Yes, please create an issue with a testcase.  In general that's rather quick to implement.