ghdl/ghdl #1669: Building on macOS ARM

controversial, 2021-03-02:
### Is your feature request related to a problem? Please describe.
There is no build available for ARM macs, and I’m having trouble building it myself.

The README in this repository makes it seem like it should be possible to compile GHDL on ARM, but I haven't managed to compile it even after a few hours of effort.

### What I’ve tried
I know the [`mcode` backend won’t run on ARM](https://github.com/ghdl/ghdl/issues/1535#issuecomment-744038012), so I'm trying with the llvm backend. I’ve tried:
```bash
brew install llvm
cd ghdl
mkdir build
cd build
../configure --with-llvm-config=/opt/homebrew/opt/llvm/bin/llvm-config --prefix=/Users/luke/opt/ghdl
make
```
I need an Ada compiler, though, and I haven't been able to find an ARM distribution of `gnatmake` anywhere (does [this](https://github.com/AdaCore/gps/issues/76) suggest that that is intentional?). I downloaded an older macOS x86_64 version of GNAT from [adacore.com/download](https://www.adacore.com/download) and tried again to set up my build environment using
```bash
cd build
GNATMAKE=/opt/GNAT/2020/bin/gnatmake ../configure --with-llvm-config=/opt/homebrew/opt/llvm/bin/llvm-config --prefix=/Users/luke/opt/ghdl
make
```
However, the x86_64 version of gnatmake doesn’t seem capable of compiling binaries for the architecture it’s not running under, and seems to try to generate an x86_64 binary, generating a bunch of linking errors along the way.

<details>
<summary>Full compiler output:</summary>
<pre><code>mkdir -p /Users/luke/opt/ghdl
mkdir -p /Users/luke/opt/ghdl/bin
mkdir -p /Users/luke/opt/ghdl/lib
mkdir -p /Users/luke/opt/ghdl/lib/ghdl
mkdir -p /Users/luke/opt/ghdl/include
GHDL_VER_DESC=${GHDL_VER_DESC:-tarball}; \
	GHDL_VER_REF=${GHDL_VER_REF:-unknown}; \
	GHDL_VER_HASH=${GHDL_VER_HASH:-unknown}; \
	if test -d ../.git && desc=`cd ..; git describe --dirty --long`; then \
	  GHDL_VER_DESC=`echo $desc | sed -e 's/\([^-]*-g\)/r\1/' -e 's/-/./g' -e 's/^v//g'`; \
	  GHDL_VER_REF=`cd ..; git rev-parse --abbrev-ref HEAD`; \
	  GHDL_VER_HASH=`cd ..; git rev-parse HEAD`; \
	fi; \
	sed \
	  -e "s#@VER@#2.0.0-dev#" \
	  -e "s#@DESC@#${GHDL_VER_DESC}#" \
	  -e "s#@REF@#${GHDL_VER_REF}#" \
	  -e "s#@HASH@#${GHDL_VER_HASH}#" \
	  < ../src/version.in > version.tmp;
if [ ! -r version.ads ] || ! cmp version.tmp version.ads > /dev/null; then cp version.tmp version.ads; fi
make -f ../src/ortho/llvm6/Makefile \
	 ortho_srcdir=../src/ortho ortho_exec=ghdl1-llvm \
	 GNATFLAGS="-aI../src -aI../src/vhdl -aI../src/synth -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm6 -aI../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe" LDFLAGS="-Wl,-no_compact_unwind" \
	 LLVM_CONFIG="/opt/homebrew/opt/llvm/bin/llvm-config" CXX="clang++" \
	 CFLAGS=" -g" \
	 GNATMAKE="/opt/GNAT/2020/bin/gnatmake" all
clang++ -c `/opt/homebrew/opt/llvm/bin/llvm-config --cxxflags` -g -o llvm-cbindings.o ../src/ortho/llvm6/llvm-cbindings.cpp
/opt/GNAT/2020/bin/gnatmake -o ghdl1-llvm -aI../src/ortho/llvm6 -aI../src/ortho \
	-aI../src -aI../src/vhdl -aI../src/synth -aI../src/grt -aI../src/psl -aI../src/vhdl/translate -aI../src/ghdldrv -aI../src/ortho -aI../src/ortho/llvm6 -aI../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe ortho_code_main -bargs -E \
	-largs llvm-cbindings.o --LINK=clang++ \
	-Wl,-no_compact_unwind `/opt/homebrew/opt/llvm/bin/llvm-config --ldflags --libs --system-libs`
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/ortho/llvm6/ortho_code_main.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/vhdl/translate/ortho_front.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/ortho/llvm6/ortho_llvm.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/bug.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/dyn_interning.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/dyn_maps.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/dyn_tables.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/errorout.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/errorout-console.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/flags.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/hash.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/interning.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/libraries.adb
gcc -c -I./ -I../src/ortho/llvm6 -I../src/ortho -I../src -I../src/vhdl -I../src/synth -I../src/grt -I../src/psl -I../src/vhdl/translate -I../src/ghdldrv -I../src/ortho -I../src/ortho/llvm6 -I../src/synth -gnat12 -gnaty3befhkmr -gnatwa -gnatwC -gnatf -g -gnata -gnatwe -I- /Users/luke/Developer/es4/ghdl/src/name_table.adb
[... 289 lines cut ...]
ld: warning: ignoring file /opt/homebrew/Cellar/llvm/11.1.0/lib/libLLVMDemangle.a, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: warning: ignoring file /opt/homebrew/Cellar/llvm/11.1.0/lib/libc++.dylib, building for macOS-x86_64 but attempting to link with file built for macOS-arm64
ld: in '/opt/homebrew/Cellar/llvm/11.1.0/lib/libunwind.dylib', building for macOS-x86_64 but attempting to link with file built for macOS-arm64
[... cut ...]
My end goal is to run ghdl-plugin-yosys under a native (arm64) installation of yosys. There doesn't seem to be a way to install `ghdl-yosys-plugin` that doesn't involve linking a `.so` file, and I can’t link an arm64 distribution of yosys to an x86_64-built `.so` file:
```sh
cd ghdl-yosys-plugin
LIBGHDL_LIB=ghdl-macos-10.15-llvm/lib LIBGHDL_INC=ghdl-macos-10.15-llvm/include arch -x86_64 make
file ghdl.so  # ghdl.so: Mach-O 64-bit dynamically linked shared library x86_64
file $(which yosys)  # /opt/homebrew/bin/yosys: Mach-O 64-bit executable arm64
```
(note `x86_64` vs `arm64` in the output from the final two lines)

umarcor, 2021-03-02:
@controversial, is it an option to use a GNU/Linux system instead of macOS? Ada tools are available there. It's the best solution for using GHDL on arm64 today. However, we do want your use case to be supported too. Let's go!

> I know the mcode backend won’t run on ARM, so I'm trying with the llvm backend.

This is correct. 

> I need an Ada compiler, though, and I haven't been able to find an ARM distribution of gnatmake anywhere (does this suggest that that is intentional?). I downloaded an older macOS x86_64 version of GNAT from adacore.com/download and tried again to set up my build environment using

There is an "ARM ELF (32 bits) (hosted on darwin)" available in the dropdown list: https://www.adacore.com/download/more. I don't know if that will work. Are Apple's new ARMs aarch64 only?

> However, the x86_64 version of gnatmake doesn’t seem capable of compiling binaries for the architecture it’s not running under, and seems to try to generate an x86_64 binary, generating a bunch of linking errors along the way.

There is https://github.com/hackfin/ghdl-cross.mk for generating a cross-compiling GHDL. That's not the solution you want, but it might work.

> I'd like to see arm64 binaries of GHDL distributed alongside the other binaries, or at least have a viable path to compile them myself.

While we can provide ARM64 binaries for GNU/linux, I don't think we have CI resources for generating ARM64 binaries for macOS. Unless someone provides guidelines for cross-compiling GHDL on x86-64 macOS, the solution to this issue will need to be instructions for users to build GHDL themselves.

> There doesn't seem to be a way to install ghdl-yosys-plugin that doesn't involve linking a .so file, and I can’t link an arm64 distribution of yosys to an x86_64-built .so file:

You can build ghdl-yosys-plugin without generating `ghdl.so`. As explained in https://github.com/ghdl/ghdl-yosys-plugin#build-as-part-of-yosys-not-recommended, you can build ghdl-yosys-plugin as part of Yosys. However, I don't know whether that's useful for you. The plugin uses the `libghdl` shared library. Therefore, you need to build GHDL, even if you can avoid building the plugin. However, an arm64 Yosys with the built-in plugin might be able to load the x86-64 shared library through Rosetta.

controversial, 2021-03-03:
Thanks for your response! I will see where I can get with virtualizing linux ARM. In pursuit of a native solution, however:

First, I don't think it will be possible to have arm64 Yosys load the x86_64 shared library through Rosetta, since
> The system prevents you from mixing arm64 code and x86_64 code in the same process. Rosetta translation applies to an entire process, including all code modules that the process loads dynamically ([source](https://developer.apple.com/documentation/apple_silicon/about_the_rosetta_translation_environment)).

I downloaded both available releases of “ARM ELF (32 bits) (hosted on darwin)” from the page you linked, but neither comes with an executable `gnatmake` nor any of the sources required to build it (they only include `gprbuild` and it seems like that‘s intentional?). The executables that _do_ come with it are x86_64 executables, which I’m a little confused about.

I don't know anything about cross-compilation, but what are the steps I would need to take to start exploring that path? It seems like there won’t be any other path if no GNAT for arm64 Darwin exists.

umarcor, 2021-03-03:
Let's ping @hackfin, as he is the author of ghdl-cross. I believe he didn't have macOS as a target, but he might provide valuable info.

controversial, 2021-03-03:
Okay! Fwiw, I also looked into building `gnatmake` myself, but GNAT doesn't come with the Homebrew distribution of `gcc` for ARM (only the fortran and C compilers) and it looks like it's not possible to build the GNAT compiler without a previous GNAT installation (nor is it possible to build [gnat-llvm](https://github.com/AdaCore/gnat-llvm)).

diogratia, 2021-03-04:
You can't help but wonder if you couldn't boot strap through an x86_64 gnat-llvm, either producing llvm assembly or bytecode and using that to cross target arm64.

One of the LLVM guys Iain Sandoe has a [project for a native gcc](https://github.com/iains/gcc-darwin-arm64) (and ominously has gcc/ada in the source tree).

hackfin, 2021-03-04:
Pong...I'm afraid, no Apple ARM target here. `ghdl-cross` focuses on buildin an 'ortho' GHDL spitting out a different target's .exe (mostly MS Windows).
But you could do the dance manually by first cross-compiling gcc and gnat-tools for the Apple aarch64 target from an existing GNAT development environment, then use that to compile GHDL. Not sure if crosstools-ng provides that though..

tgingold, 2021-09-14:
I am closing this issue as we won't make any progress until there is an Ada compiler for this platform.
I don't have access to any Apple ARM machine.

closed 2021-09-14 by tgingold

myzinsky, 2023-07-04:
Anything new on this front? I have an m2 and could help investigate...

myzinsky, 2023-07-04:
> I am closing this issue as we won't make any progress until there is an Ada compiler for this platform.
> I don't have access to any Apple ARM machine.

Would that help: https://github.com/simonjwright/building-gcc-macos-native/tree/aarch64-apple-darwin ? 
@tgingold

tgingold, 2023-07-05:
I do think so.  I think it would be possible to build ghdl with the llvm backend using this Ada compiler.

myzinsky, 2023-07-05:
Just to note to prevent others from compilation nirvana: the oss cad suite of Darwin x86 runs smoothly with Rosetta on arm m2 Mac. So it is not an urge to have native binaries for now...

AasmundN, 2024-09-16:
Hello,
I've been trying to compile the ghdl-plugin-yosys on my mac m1 for a few hours now with no luck. Did anyone here make it work? I am running mcode ghdl and yosus with no problems.

myzinsky, 2024-09-17:
I had the same adventure, sorry for that. Just downloaded the oss cad suite for Intel Macs. Rosetta does its job perfectly.

tgingold, 2024-09-17:
What are the issues ?  Given ghdl can be built on macOS ARM, there shouldn't be blocking points.

AasmundN, 2024-09-17:
Thank you for answering,

It did not occur to me that the intel build comes prebuilt with yosys-plugin-ghdl. Installed that now and things seem to be working. Thank you!

As for the compilation issues; I did not try building ghdl myself. I used the prebuilt binary that came with oss-cad-suite for arm. That might have been the problem I suppose. I just needed something that works, so I will not look more into this unless you really want me to.

tgingold, 2024-09-18:
If it works for you, that's fine.  We could revisit the issue when it will be blocking.

kms1212, 2024-09-21:
In the last comment on issue #1115 , I wrote about how to build GHDL.
I think it's too late but I hope I can help someone to build GHDL on macOS aarch64.

Paebbels, 2024-09-21:
@kms1212 I'm currently reworking GHDL's pipeline in https://github.com/ghdl/ghdl/pull/2718 on branch [Paebbels/GHDL:simplify-pipeline](https://github.com/Paebbels/ghdl/tree/simplify-pipeline).I have problems on macOS 12, 13, 14 (x86_64 and aarch64) with respect to `RPath`.

Building GHDL on macOS works, but then copying the results as an artifact to a new job (also a new virtual machine) and executing GHDL's testsuit produces errors due to dynamically loading `libgcc_s.1.1.dylib`. See also https://github.com/ghdl/ghdl/pull/2718#issuecomment-2289994493 for error details.

> dyld[3313]: Library not loaded: @rpath/libgcc_s.1.1.dylib

The idea is to provide more macOS builds (as well as other builds and more test combinations), so homebrew can be fixed as it downloads directly from GHDLs assets.

kms1212, 2024-09-22:
@Paebbels This is my `otool` output:
```
/opt/ghdl/bin/ghdl1-llvm:
	/opt/homebrew/opt/llvm/lib/libLLVM.dylib (compatibility version 1.0.0, current version 18.1.8)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.100.2)
	/usr/lib/libc++.1.dylib (compatibility version 1.0.0, current version 1700.255.0)
	/opt/homebrew/opt/llvm/lib/libunwind.1.dylib (compatibility version 1.0.0, current version 1.0.0)
/opt/ghdl/lib/libghdl-4_1_0.dylib:
	@rpath/libghdl-4_1_0.dylib (compatibility version 0.0.0, current version 0.0.0)
	@rpath/libgnat-12.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.100.2)
/opt/ghdl/lib/libghdlvpi.dylib:
	@rpath/libghdlvpi.dylib (compatibility version 0.0.0, current version 0.0.0)
	/usr/lib/libSystem.B.dylib (compatibility version 1.0.0, current version 1345.100.2)
```

As there's no libgcc dependency on my build, it seems okay to use artifacts resulting from following my instructions which links it as a static library.

Paebbels, 2024-12-02:
@kms1212 I solved it by copying `libgcc` besides the dylib missing that file.

Have you noticed that `libghdl` contains a reference to itself? Same for `libghdlvpi`. There are use testcases that fail, because there is a duplicate. See #2806.