ghdl/ghdl #1671: libghdl thoughts

m-kru, 2021-03-04:
I have been trying to use `libghdl.a`. Here are my thoughts and questions.

1. By default header files (right now only 2 `ghdlsynth_gates.h` and  `ghdlsynth.h`) are placed in `/usr/local/include`. I think it would be more clean way to create ghdl directory within `/usr/local/include`. The  `#include` paths would become `ghdl/synth_gates.h` and  `ghdl/synth.h` and there would be single `ghdl` directory within `/usr/local/include` containing all header files related with GHDL. Currently not a big issues as there are only 2, but might be useful in the future.
2. The header files are C++ header files. I guess this is because they were used only with the C++ code. However it is rather the C that is the lingua franca for such libraries API. Having C header files it would be more compatible with other languages by default. Of course if there are strong reasons for staying with C++ headers it is fine. However, it would be nice to document somewhere that `libghdl` has C++ interface solely.
3. `libghdl` is currently not usable in a way user expects such library to be. Namely, it is missing header files. There are 2 for synthesis only. There are also Python bindings, but their usage is limited to Python. I have skimmed through the Ada sources and it looks like writing these header files by hand is too much work. @tgingold do you think it would be relatively easy to auto generate proper header files from Ada sources?

tgingold, 2021-03-04:
1. Yes, that's something that shouldn't difficult to implement and I agree with that.

2. Currently, the main use of ghdlsynth.h is the Yosys plugin, which is written in C++.  This file could be translated to C.

3. Some headers could be automatically generated like for Python (some files are automatically generated).  For the other ones, that would be harder.

But what do you like to do with libghdl ?

m-kru, 2021-03-04:
> But what do you like to do with libghdl ?

Actually a lot of things, from code formatting to reset tree analysis. The code is already there in the `libghdl.a`, but the header files are missing.

> 2. Currently, the main use of ghdlsynth.h is the Yosys plugin, which is written in C++.  This file could be translated to C.

Are there any tests for this. I could change this, but I would not like to break you anything.

umarcor, 2021-03-04:
@m-kru the testing of ghdl-yosys-plugin and, therefore, of those headers is distributed between ghdl/ghdl, ghdl/docker and ghdl/ghdl-yosys-plugin. However, I can prepared a complete CI pipeline for you, so that you have it tested with a single script. Are you ok with using containers or do you want to execute everything on your host? In the latter, which is your host OS?

m-kru, 2021-03-04:
Would be really nice. If I had some test against which I can test, then I would be able to do 1 and 2. The 3 is a much bigger task. I do not see a need to use container in this case. Arch Linux.

m-kru, 2021-03-08:
@tgingold 1 is now completed. To implement 2 (C++ to C) I would like to run some tests for yosys so that I know I have not broken anything. Is there any way to change GHDL locally and run tests for ghdl-yosys-plugin locally? Well, I am sure it is possible, the question is rather how?

tgingold, 2021-03-08:
I usually test without installing.  But I create symbolic links for some headers.

Or you can install in a temporary place and set the GHDL environment variable before building the yosys plugin.