ghdl/ghdl #1675: Expose record fields in synthesis

antonblanchard, 2021-03-08:
One issue when using ghdl/yosys synthesis is that records get flattened into a vector. The wishbone bus in microwatt is one example:

```
    constant wishbone_addr_bits : integer := 32;
    constant wishbone_data_bits : integer := 64;
    constant wishbone_sel_bits : integer := wishbone_data_bits/8;

    subtype wishbone_addr_type is std_ulogic_vector(wishbone_addr_bits-1 downto 0);
    subtype wishbone_data_type is std_ulogic_vector(wishbone_data_bits-1 downto 0);
    subtype wishbone_sel_type  is std_ulogic_vector(wishbone_sel_bits-1  downto 0);

    type wishbone_master_out is record
        adr : wishbone_addr_type;
        dat : wishbone_data_type;
        sel : wishbone_sel_type;
        cyc : std_ulogic;
        stb : std_ulogic;
        we  : std_ulogic;
    end record;
```

When run through ghdl/yosys synthesis, and exported as verilog, this interface becomes:

```
  output [106:0] wishbone_data_out;
```

This makes debugging a bit more difficult, eg are any of my control bits X state?. We might also want to apply specific timing constraints to signals.

tgingold, 2021-03-08:
Yes I agree it makes debugging difficult.

Would it be ok if the record elements are exposed only on ports (and not for internal signals) ?

Is backward compatibility important (ie do we need to option to keep the current output) ?

antonblanchard, 2021-03-09:
> Would it be ok if the record elements are exposed only on ports (and not for internal signals)?

For synthesis I think ports would be enough, because internal signals would be completely lost in the yosys optimisation steps.

I can however see a use for internal signals in simulation, eg ghdl->yosys->verilator to do mixed language simulation. 

> Is backward compatibility important (ie do we need to option to keep the current output) ?

I can't speak for others, but backward compatibility for this is not a big deal for me.

closed 2021-03-27 by tgingold

antonblanchard, 2021-03-29:
Thanks @tgingold!

commit c6d3502d (2021-03-27): synth: expand ports for record. Fix #1675