ghdl/ghdl #1677: A process statement without a sensitivity list does not advance simulation time

nobodywasishere, 2021-03-08:
**Description**
A `process` statement that does not have a sensitivity list produces an infinite loop and simulation time does not advance. GHDL will just hang.

**Expected behaviour**
I expect to get a warning about simulation time not advancing, especially when there's a wait statement on said process. Currently, GHDL will just wait indefinitely.

**How to reproduce?**

```vhd :file: ent.vhd
library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

entity saturatingadd is
  port(
	  result : out unsigned(7 downto 0)
  );
end saturatingadd;

architecture synth of saturatingadd is

begin

    process is
    begin
        result <= "00000000";
    end process;

end;

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;
use std.textio.all;

entity ent is
end ent;

architecture test of ent is

    component saturatingadd is
      port(
    	  result : out unsigned(7 downto 0)
      );
    end component;

    signal result : unsigned(7 downto 0);

begin

  dut : saturatingadd port map(result);

  process is
  begin
    wait for 10 ns;
    write (output, "TEST PASSED." & LF);
    wait;
  end process;
end test;

```

```sh :image: ghdl/ghdl:buster-mcode
ghdl -a ent.vhd
ghdl -r ent
```

> NOTE: `:file:` and `:image:` identifiers are specific to [issue-runner](https://github.com/1138-4EB/issue-runner). We suggest to use these, since it allows continuous integration workflows to automatically test the MWE. Using `ghdl/ghdl:*` docker images to run the MWEs ensures that the latest available GHDL is used.

**Context**
Please, provide the following information:

- OS: Docker and Arch Linux
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [ ] Built from sources: `commit SHA`
  - [x] Docker: "hdlc/ghdl:yosys" (whichever the latest is)
  - [x]  Arch Linux `GHDL 2.0.0-dev (tarball) [Dunoon edition] open-tool-forge.nightly-20210301`

**Additional context**
Add any other context about the problem here. If applicable, add screenshots to help explain your problem.

nobodywasishere, 2021-03-08:
From eine on gitter about the issue:

> `dut : saturatingadd port map(a, b, result);` is an "infinite running concurrent statement". There is no wait inside.

tgingold on gitter for a possible solution:

> We can emit a warning if a non-sensitized process doesn't have a wait statement (and doesn't call a procedure).
Or/and we can limit the number of times a process loops.
I think that the warning will catch 99% of the mistakes.

Paebbels, 2021-03-09:
@tgingold please also see my old issue #41 which was similar but for recursions.

tgingold, 2021-03-09:
But recursion is a much more difficult problem - it is the halting problem.
It could be limited dynamically but with a cost.

Paebbels, 2021-03-09:
I think it's not much more difficult.

In a process you have a counter to count iterations. In a recursion to have a global variable or structure/point you pass along the execution path which gets incremented on a call and decremented on a return. In both cases you have the additional cost of a add/sub. In case of recursion, the bigger overhead comes from passing the thing you count up/down.

tgingold, 2021-03-09:
Well this is not option I choose.  I will simply emit a warning when there is a process statement without a wait (and without procedure calls).

We could add a counter for processes, but I think this is overkill.  I think that most processes will execute once or twice.  And the counter could be  implemented in the runtime.

For subprograms, a counter is not enough as the limitation is also the stack, whose size is difficult to compute.  We could check the stack size before every call, but that's also heavy.  And recursion is both usual and also possibly mutual.

commit 86e92483 (2021-03-09): Add a warning for processes without a wait statement.  Fix #1677
closed 2021-03-09 by tgingold