ghdl/ghdl #1678: Formatting style for Ada enumeration and array type declarations.

m-kru, 2021-03-11:
I am preparing a script for automating generation of C header files based on the Ada `.ads` files.
Current formatting of multi-lines enumeration and arrays declarations are not parser friendly.
```Ada
   type Language_Type is
     (
      Language_Unknown,
      Language_Vhdl,
      Language_Psl,
      Language_Verilog
     );

   type Table_Type is
     array (Table_Index_Type range <>) of Table_Component_Type;
```

Why is it not friendly? Based on the first line of the declaration one can not determine what is going to be declared.
This makes parsing line by line a bit harder.
I was wondering whether this is the formatting style done by some beautifier and whether we can change it?

Following style for multi-lines enumerations would be much easier to parse:
```Ada
   type Language_Type is (
      Language_Unknown,
      Language_Vhdl,
      Language_Psl,
      Language_Verilog
   );
```

In case of arrays there are 2 possibilities.
1. Make them always single line. This would result with lines length greater than 80 columns.
```Ada
   type Table_Type is array (Table_Index_Type range <>) of Table_Component_Type;
```
2.  At least move the array keyword to the first line:
```Ada
   type Table_Type is array
     (Table_Index_Type range <>) of Table_Component_Type;
```

tgingold, 2021-03-11:
Hum, I am a little bit afraid!

The enumeration types are already parsed by the python scripts to generate python binding and other files.

I am pretty sure that trying to parse arrays is not a good idea.  They don't really have equivalent types in C.

If you want to do an automatic header generation (which might not be straightforward), I would first deal only with scalar types.  All the other types should be ignored and any subprogram using them should also be ignored as there is no directly translation to C.

m-kru, 2021-03-11:
> The enumeration types are already parsed by the python scripts to generate python binding and other files.

We could leave enumerations untouched if the `array` keyword was moved to the first line.

> I am pretty sure that trying to parse arrays is not a good idea.  They don't really have equivalent types in C.

I do not want to parse arrays. I just want to know that this is an array type based on a single line.

If the parser encounters something like:
```Ada
type Language_Type is
```
it can't determine what is going to be declared. It needs to check next line.
This is of course doable, however the whole parsing process is not pure linear anymore.

To sum up. Moving `array` to be the last word in the first line, instead of first word in the second line would be enough.

m-kru, 2021-03-11:
I have checked, there are 18 cases where `array` appears as the first word in the second line of array type declaration.

tgingold, 2021-03-12:
If you scan `type A is` then you can determinate the nature of the type by parsing the second line.

m-kru, 2021-03-12:
@tgingold There are some types that can not be easily parsed, for example arrays and some types that I do not even know what they mean:
```Ada
   type Quantity_Dependency_Type (Nbr : Natural);

   type Size_Type is mod System.Memory_Size;
```

I was wondering how "handicapped" would be the functionality of GHDL lib accessed via C interface with limited number of types.
I really start thinking that learning Ada would be simply easier.

tgingold, 2021-03-13:
The first one is an incomplete type declaration for a type with a discriminant.  You can forget it as there is equivalent C type.

The second one is a modular type.  It corresponds to an unsigned C type.

m-kru, 2021-03-13:
> The first one is an incomplete type declaration for a type with a discriminant. You can forget it as there is equivalent C type.

There _is_ or there _is no_?
If there is, what is the equivalent type?
If there is no, how are such types used internally? Will this fact limit somehow functionalities of the C bindings?

tgingold, 2021-03-13:
Sorry, there is no direct equivalent C type.

In the case of Quantity_Dependency_Type, it is in src/vhdl/simulate which is used only by ghdl_simul, which is usually not built.  And it is used to create a list, so not even by subprograms.  In other words, you don't care about it.

Most of the widely used types are scalar types (in particular for the nodes of the AST).  So you can already do a lot with them.
Strings are not represented the same way in C and in Ada.  So to write the python bindings, we needed to add a few subprograms (to handle C strings).

m-kru, 2021-03-13:
Can I treat all `mod` types as `unsigned int`?

closed 2021-03-18 by m-kru