ghdl/ghdl #1682: unhandled attribute "loc" on port list for entity declaration

cr1901, 2021-03-12:
**Description**
Some yosys backends do not use separate user-constraint files (LPF) to map pins to ports. The "traditional" method of creating user constraints for these backends is to use an attribute called `LOC`, that is declared as part of the port list.

E.g. in Verilog, the follow code constrains a module port to the LED pin on TinyFPGA Ax:

```
module top (
  (* LOC="13" *)
  inout pin1
);
```

GHDL seems to recognize the equivalent attributes on entity declarations, but decides to discard the attribute instead of keeping it.

**Expected behaviour**
According to VHDL 2008 spec, section 3.2, an attribute is legal to put into an entity declaration. GHDL accepts it, but then discards the attribute:

```
yosys -qp 'ghdl --std=08 prims.vhd tinyfpga.vhd -e'
note: top entity is "top"
tinyfpga.vhd:20:3:warning: instance "internal_oscillator_inst" of component "osch" is not bound [-Wbinding]
tinyfpga.vhd:15:14:warning: (in default configuration of top(arch))
tinyfpga.vhd:12:13:warning: attribute "loc" for port "pin1" is not kept in the netlist
tinyfpga.vhd:12:13:warning: unhandled attribute "loc"
```

I would expect attributes that I declare on ports to survive for further processing by `yosys`. In particular, if `LOC` attributes are discarded for my backend (MachXO2), the pnr tool will have no location constraints at all; I haven't implemented LPF parsing yet.

**How to reproduce?**
Minimal Working Example (notice the `attribute LOC of pin1: signal is "13";`):

```vhd :file: tinyfpga.vhd
library ieee ;
context ieee.ieee_std_context;

use work.components.all;

entity top is
  port (
    pin1: out std_logic
  );

  attribute LOC: string;
  attribute LOC of pin1: signal is "13";
end;

architecture arch of top is
  signal clk: std_logic;
  signal led_timer: unsigned(23 downto 0) := (others=>'0');
begin

  internal_oscillator_inst: OSCH
  generic map (
    NOM_FREQ => "16.63"
  )
  port map (
    STDBY => '0',
    OSC => clk
  );

  process(clk)
  begin
    if rising_edge(clk) then
      led_timer <= led_timer + 1;
    end if;
  end process;

  pin1 <= led_timer(led_timer'left);

end;

```

```sh :image: ghdl/ghdl:buster-mcode
yosys -qp 'ghdl --std=08 prims.vhd tinyfpga.vhd -e'
```

**Context**
- OS:
  ```
  $ uname -a
  MINGW64_NT-10.0-19042 DESKTOP-H0PMN4M 3.1.7-340.x86_64 2020-11-08 12:32 UTC x86_64 Msys
  ```
- Origin:
  - [ ] Package manager: `version`
  - [ ] Released binaries: `tarball_url`
  - [x] Built from sources:
    ```
    $ ghdl -v
    GHDL 2.0.0-dev (0.37.0.r1440.ge6f397e0.dirty) [Dunoon edition]
    Compiled with GNAT Version: 10.2.0
    llvm code generator
    Written by Tristan Gingold.

    Copyright (C) 2003 - 2021 Tristan Gingold.
    GHDL is free software, covered by the GNU General Public License.  There is NO
    warranty; not even for MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.
    ```

**Additional context**
Being an absolute VHDL beginner, I don't know how to duplicate this with `ghdl` alone:

```
$ ghdl -e tinyfpga.vhd
C:\msys64\mingw64\bin\ghdl.exe: bad unit name 'tinyfpga.vhd'
C:\msys64\mingw64\bin\ghdl.exe: (a unit name is required instead of a filename)
```

The `:image:` identifier in the code snippet above might be incorrect for testing a MWE against `yosys`.

tgingold, 2021-03-13:
This requires some attention in the implementation.

In fact the attribute is passed to the backend but in lower case letters (I think - not checked).  So it just needs to be converted to upper case letters, which is simple.

But, this is a vendor specific operation.  It is well possible that another vendor is using lower case letters.  So I fear it is time to add an option to handle vendor specificities.  Right now I am thinking about `--attributes=lattice`.

The problem (of course) is that vendor names change, and there might be other specificities than attributes.  But that's a start point.

You can reproduce that with `ghdl` alone using `ghdl --synth tinyfpga.vhd -e`

cr1901, 2021-03-13:
This is a bit of bikeshedding, but:

While Lattice Diamond indeed supports `LOC` attributes in ports, `LOC` is not Lattice exclusive; at least one Xilinx CPLD and the Greenpak4 backend _in yosys_ use `LOC` attributes as well that downstream tools are expected to consume. But these `LOC` attributes aren't tied to tools provided by the vendor. I think it was just reusing what vendors have done before.

>In fact the attribute is passed to the backend but in lower case letters (I think - not checked).

Last I checked the output JSON when running the whole flow, the `LOC` attribute was gone. Indeed this may be a case thing, but I had something come up in the past hour so I can't immediately check this.

Paebbels, 2021-03-13:
Xilinx ISE tool chain was using `LOC` for pins and hard macros.

In Vivado everything is done via TCL commands in XDC files. Vivado actively ignores attributes, I haven't found out why they do it (because it's still documented...). Pins are now placed with `PACKAGE_PIN`, but `LOC` should also work.

In Intel (Altera) environments, timing  constraints can be specified in attributes and handed over to STA (static timing analysis).

closed 2021-03-17 by tgingold

tgingold, 2021-03-18:
I have made a first change to support the LOC attribute (both in ghdl and in the yosys plugin).
I am not sure it is still OK.  In particular the case and how it applies to any port direction.

How does it work when the port is a bus ?  Is this attribute documented somewhere ?

cr1901, 2021-04-06:
@tgingold Apologies for the delay in responding. I'm recompiling `yosys` right now among other
things and will report back to see how your changes worked. Re: your questions:

>How does it work when the port is a bus ?  Is this attribute documented somewhere ?

## Diamond
Diamond supports any case sensitivity for `LOC` attributes. Though there is
syntax to apply multiple `LOC` constraints to a single signal (end of page
1305), I've not tested it yet for pins; the example is for constraining a
multi-bit signal to SLICEs. I'm _guessing_ port direction is inherited from the signal
the `LOC` attribute is attached to.

```
Lattice Diamond 3.11 SP1 Help

Page 1304
LOC

Convention LOC= site_name (PIN | POS)

Description Specifies a site location for the component that is created when
this block is mapped. If attached to multiple blocks, indicates that these blocks
are to be mapped together in the specified site.

When the device is mapped, the Map Design process (map) first maps blocks
to PLCs without looking at LOC attributes. Then it tries to merge all PLCs with
the same LOC attribute into a single PLC, and then it assigns this PLC to the
site with the specified site name. If all of the blocks with the LOC attribute
cannot be mapped into a single component, MAP groups together as many as
possible.

When it encounters a LOC preference, MAP also writes a LOCATE
preference for the component into the preference file. The LOCATE
preference locks the component to the site. When you run PAR, the
component cannot be unplaced, moved, swapped, or deleted.

The LOC attribute can be attached to anything that will end up on an I/O cell,
and to clocks and internal flip-flops, but it should not be attached to
combinational logic that will end up on a logic cell; doing so could fail to
generate a locate preference. The LOC attribute overrides register ordering.

Device Support All
```

```
Page 1305
VHDL Syntax ATTRIBUTE LOC : string;
ATTRIBUTE LOC OF [pin_name]: SIGNAL IS "[site_name]";

VHDL Example Code ATTRIBUTE LOC : string;
ATTRIBUTE LOC OF output_vector : SIGNAL IS "H5";

Verilog Syntax – Precision //pragma attribute [pin_name] LOC [site_name]

Verilog Example Code – Precision // pragma attribute rst LOC H5

Verilog Syntax – Synplify PinType [pin_name] /* synthesis
LOC="[site_name]"*/;

Verilog Example Code – Synplify //input rst /* synthesis
LOC="H5" */ ;

The following examples show slice locations for packing registers:
VHDL Example Code for Register Packing ATTRIBUTE LOC : string;
ATTRIBUTE LOC OF inreg : SIGNAL IS "R2C9D R2C9D R2C8D R2C8D
R2C7D R2C7D R2C6D R2C6D R2C5D R2C5D R2C4D R2C4D R2C3D R2C3D
R2C2D R2C2D";
Verilog Example Code for Register Packing (Precision) // pragma
attribute inreg LOC R2C9D R2C9D R2C8D R2C8D R2C7D R2C7D R2C6D
R2C6D R2C5D R2C5D R2C4D R2C4D R2C3D R2C3D R2C2D R2C2D
```

## Yosys
I can't speak for the other backends in `yosys` using `LOC`, but this is what
I do for MachXO2:

AFAICT, `LOC` attributes are not specially handled by `yosys` and are passed
unmodified (unless you use [`attrmvcp`](https://github.com/YosysHQ/yosys/blob/master/passes/techmap/attrmvcp.cc) to move them, which is useful for
inferring I/O buf primitives w/ `LOC` constraints). It's up to the PNR backend
on how to handle them. One thing `yosys` does is split multi-bit signals into
single bit signals via [`iopadmap -bits`](https://github.com/YosysHQ/yosys/blob/master/passes/techmap/iopadmap.cc), which I have enabled for
`synth_machxo2` (`-bits` being an option).

In my case of MachXO2, at present I only support `LOC` attributes on single-bit
signals, upper-case only ([link](https://github.com/YosysHQ/nextpnr/blob/master/machxo2/pack.cc#L243-L267)). In practice, the `yosys` flow of MachXO2 right now
expects a user to manually split their signals at their top-level so `LOC`
attributes get passed down to the backend properly. Doing splitting via
`iopadmap -bits` is more "laying the groundwork" for extension for when I do
support multi-bit signals, ideally without needing to invasively modify the
PNR backend :).

That said I feel like `iopadmap -bits`-splitting would need to be multi-bit
`LOC`-aware to actually support `LOC` constraints on multi-bit signals.
I'm not sure this would be implemented in `yosys`. So I may remove the `-bits`
option in the future. I'm not sure yet.

umarcor, 2021-04-06:
> I'm recompiling `yosys` right now among other things and will report back to see how your changes worked. Re: your questions:

On MSYS2, you can pick these fresh artifacts: https://github.com/msys2/MINGW-packages/actions/runs/720446685

umarcor, 2021-04-06:
@cr1901 does splitting by bit imply that constraint files (when supported) will also need to specify pins for individual signal names and not for arrays?

https://github.com/hdl/constraints/blob/9583871f524d38fc841fa0113535f1a822c0e02c/board/IceStick/GPIO.LED.pcf#L12-L16

vs

https://github.com/hdl/constraints/blob/9583871f524d38fc841fa0113535f1a822c0e02c/board/OrangeCrab/constraints.lpf#L11-L13

cr1901, 2021-04-06:
@umarcor ICE40 PCFs also support arrays :P. And I don't imagine that individual signal names will need to be used, but I'm not sure right now. I'll figure it out when I actually implement LPF.

Anyways, the warning is gone, but I still can't get a working bitstream. I think emitting `loc` instead of `LOC` is the reason. Instead of having to test all 8 combinations of case-sensitivity for `LOC` in `attrmvcp`, is it possible to emit `LOC` by default?

tgingold, 2021-04-09:
I am wondering whether it should be done in the yosys plugin or during synthesis.

cr1901, 2021-04-13:
@tgingold The case problem might not matter after all:

```
William@DESKTOP-H0PMN4M MINGW64 ~/Projects/FPGA/nextpnr/machxo2/examples/vhdl
$ yosys -Qh attrmap

-- Running command `help attrmap' --

    attrmap [options] [selection]

This command renames attributes and/or maps key/value pairs to
other key/value pairs.

    -tocase <name>
        Match attribute names case-insensitively and set it to the specified
        name.

    -rename <old_name> <new_name>
        Rename attributes as specified

    -map <old_name>=<old_value> <new_name>=<new_value>
        Map key/value pairs as indicated.

    -imap <old_name>=<old_value> <new_name>=<new_value>
        Like -map, but use case-insensitive match for <old_value> when
        it is a string value.

    -remove <name>=<value>
        Remove attributes matching this pattern.

    -modattr
        Operate on module attributes instead of attributes on wires and cells.

For example, mapping Xilinx-style "keep" attributes to Yosys-style:

    attrmap -tocase keep -imap keep="true" keep=1 \
            -imap keep="false" keep=0 -remove keep=0

End of script. Logfile hash: 4cc7d50980
Yosys 0.9+4052 (git sha1 0ccc7229, sccache x86_64-w64-mingw32-g++  -Os)
Time spent: 100% 1x help (0 sec)
```

Using the following command, I was able to get a working VHDL blinky bitstream for TinyFPGA Ax:

```
yosys -p "ghdl --std=08 prims.vhd tinyfpga.vhd -e; attrmap -tocase LOC; synth_machxo2 -json tinyfpga-vhdl.json"
```

This way, `nextpnr-machox2` doesn't need to check for all possible combinations of `LOC` to place constraints- it assumes the input JSON is appropriately preprocessed anyway.

tgingold, 2021-04-14:
It might be an easier way as it avoids GHDL to deal with family specific features.

Feel free to close this issue if you think it's OK.

commit b38003fe (2021-03-17): synth: handle loc attribute (for ports). Fix #1682